EverythingSmartHome/everything-presence-one #97: [BETA] - Zones Issue (Ignoring segments)

ee02217, 2023-08-03:
In the new beta, it seems the segments are being ignored.
Please see below config:
[image]

Although the zone 2 only starts at 225cm (segment 15), it still reports as Active at 200cm.

EverythingSmartHome, 2023-08-03:
Zones do not equal targets, you can have 4 targets in the same zone for example. It is not one target per zone if that makes sense.

You have zone 1 set from 0cm to 210cm (14 x 15cm = 210) which is what the sensor is reporting as target 2. Seems expected to me?

Restart doesn't actually reset the zones, the mmWave sensor is UART so the esp32 sends the commands to the sensor and it sets the distance. Then when the esp32 restarts, it looses memory of what was set, but the mmWave sensor does retain the settings, it's just the UI that looks out of sync. This is normal/expected for now due to the nature of UART.

ee02217, 2023-08-03:
Ok, understood what you are saying regarding targets. I think I misunderstood the functionality.
So, the zones are only used to exclude certain areas (distances) from registering presence.

Would be nice to have a sensor per area, to know which area is registering presence. This would allow for some nice automations to quickly identify if someone is entering or leaving (moved from A to B).

EverythingSmartHome, 2023-08-03:
It would require a different sensor unfortunately, it is beyond the capabilities of this sensor

ee02217, 2023-08-03:
Understood. Thanks!

closed 2023-08-03 by ee02217