EverythingSmartHome/everything-presence-one #99: Breaking change between 2023.4.2 (1.1.3) and 2023.7.1 (1.1.6)

RobinBol, 2023-08-09:
It seems that at least the following uniqueIds have changed in the `ListEntitiesEntityResponse` between 2023.4.2 (1.1.3) and 2023.7.1 (1.1.6):

Before
`    uniqueId: 'everything-presence-one-7083ccbinary_sensor_mmwave',`
`    uniqueId: 'everything-presence-one-7083ccbinary_sensor_occupancy',`

After
`    uniqueId: 'everything-presence-one-7083ccbinary_sensormmwave',`
`    uniqueId: 'everything-presence-one-7083ccbinary_sensoroccupancy',`

This obviously broke the mmWave and occupancy functionality for users of my app. Can you confirm this change was made, and it was intentional?

EverythingSmartHome, 2023-08-10:
This was an unintended change as reported a while back here: https://github.com/EverythingSmartHome/everything-presence-one/issues/80

I'm not sure if this matters or you factored this in but the uniqueID you are using ```everything-presence-one-7083ccbinary_sensor_mmwave``` contains the mac address which is different for each device, the section that says ```7083cc``` - this will change on every sensor to a random 6 character string. Like I say not sure if you factored this in or not but just from your example above I thought I should mention it.

RobinBol, 2023-08-10:
> This was an unintended change as reported a while back here: #80

I see, apologies, didn't turn up in my search. Then at least I know I have to take both with and without `_` into account.

> I'm not sure if this matters or you factored this in but the uniqueID you are using `everything-presence-one-7083ccbinary_sensor_mmwave` contains the mac address which is different for each device, [...]

Thanks for pointing that out, the string is parsed correctly, but I was checking if the string includes "binary_sensor_mmwave", so the mac address is disregarded.

closed 2023-08-10 by RobinBol