EverythingSmartHome/everything-presence-one #100: Beta - Target Active \ Distance Showing 'unknown' in Home Assistant

a-bianucci, 2023-08-16:
Currently running the latest build of beta which was configured this morning.

I have the following configuration:
mmWave Zone 1 Start = 0
mmWave Zone 1 End = 90
mmWave Zone 2 Start = 105
mmWave Zone 2 End = 315
** Leveraged the Set Distance Button which shows value of: _Successfully sent to sensor! 0 7 8 21_

Target 1 Active is enabled but = unknown
Target 1 Distance is enabled but = unknown
Target 2 Active is enabled but = unknown
Target 2 Distance is enabled but = unknown

How do I get the Target x Active\Distance to return something.   Below is my configuration

```
`substitutions:
  name: office
  room: "Office"
  friendly_name: "EP1 Office"
  project_name: "Everything Smart Technology.Everything Presence One"
  project_version: "1.2.3b"
  temperature_offset: "-5"
  humidity_offset: "5"
  temperature_update_interval: "60s"
  illuminance_update_interval: "30s"
  pir_delay_off: "10s"
  occupancy_delay_off: "15s"
  hidden_ssid: "false"
  factory_reset_disabled: "true"
  uart_target_output_disabled: "false"
  uart_presence_output_disabled: "false"

esphome:
  name: ${name}
  friendly_name: ${friendly_name}
  name_add_mac_suffix: false
  project:
    name: "${project_name}"
    version: "${project_version}"
  includes:
    - SEN0395_distance.h

esp32:
  board: esp32dev

esp32_ble_tracker:
  scan_parameters:
    interval: 1100ms
    window: 1100ms

bluetooth_proxy:
  active: true

logger:

api:
  encryption:
    key: <removed>

ota:

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

improv_serial:

dashboard_import:
[... 514 lines cut ...]
          if(cmd != "detRangeCfg -1") {
            std::vector<unsigned char> cmd_vec(cmd.begin(), cmd.end());
            ESP_LOGI("set_mmwave_distance", "Sending command to sensor %s", cmd.c_str());
            id(distance_zone_status).publish_state("Successfully sent to sensor! " + cmd.substr(14));
            return cmd_vec;
          } else {
            ESP_LOGE("set_mmwave_distance", "No valid segments configured. Please adjust the sliders.");
            id(distance_zone_status).publish_state("No valid segments configured. Please adjust the sliders.");
            return std::vector<unsigned char>();
          }
      - delay: 1s
      - uart.write: "saveCfg 0x45670123 0xCDEF89AB 0x956128C6 0xDF54AC89"
      - delay: 1s
      - switch.turn_on: mmwave_sensor

text_sensor:
  - platform: template
    name: "Distance Zones Status"
    id: "distance_zone_status"

custom_component:
- lambda: |-
    auto my_custom = new Sen0395Distance(id(uart_bus));
    return {my_custom};`
```

EverythingSmartHome, 2023-08-16:
Try and toggle the "UART Target Output" switch off, wait 10s then back on and see if that helps

a-bianucci, 2023-08-17:
> Try and toggle the "UART Target Output" switch off, wait 10s then back on and see if that helps

That did it! Thank you!

For what it's worth every time I update the config I need to toggle that.

closed 2023-08-17 by a-bianucci