EverythingSmartHome/everything-presence-one #101: Can't join wifi after re-install

Nuuki9, 2023-09-11:
I've been running my Presence One for a little while and wanted to update the firmware.

I used the link on the site, and took it through a full installation - that may have been wrong, but it seemed like the default. All was fine, but I now can't re-connect it to Wi-Fi. Every time I try I get "Unable to connect". My IOT SSID uses RADIUS authentication for auto VLAN assignment, but as the MAC address hasn't changed that shouldn't be affected, and I tried another SSID just in case.

I did download the logs but not seeing any messages that relate to attempts to connect to wifi.

So now I'm a bit stuck. If I can edit the YAML config I can't drop the network info in there, but not seeing how I can do that. I generally use ESPHome for other ESP devices (for OTA etc) but seemed like the direct connect was the best option here for now.

EverythingSmartHome, 2023-09-11:
Hello!

If you reflashed from the user guide, then you will need to re-enter your WiFi details from that page which it sounds like you perhaps didn't do if you aren't seeing any WiFi details in the logs.

You can do that by going back to the user guide, pressing connect, choosing the serial port and hitting the option to connect to WiFi.

If that's still not working then please attach full start-up logs captured using serial and I can take a look!

Nuuki9, 2023-09-12:
Sorry for not being clearer. I did ideed reflash the device, and then try to join Wifi. Each time I try I get the following:

[image]

As noted I did try more than one SSID, but no luck. I let it run for a minute and then downloaded the logs, attached.

[esp-web-tools-logs.txt](https://github.com/EverythingSmartHome/everything-presence-one/files/12583997/esp-web-tools-logs.txt)

EverythingSmartHome, 2023-09-12:
Can you try the method of connecting to the EP1's hotspot from a phone or laptop, loading the webpage 192.168.4.1 and try to connect that way?

Nuuki9, 2023-09-12:
Ah yes - I forgot about that.

I've now used the captive portal and again tried connecting to two different SSIDs, but again both fail.

I can see the connection attempt in the logs, which include the passwords (redacted in the logs). These were copied straight from a password manager and are definitely correct. I have dozens of other devices on those SSIDs, and the EP1 itself was previously on there fine.

I'm stumped...

[esp-web-tools-logs.txt](https://github.com/EverythingSmartHome/everything-presence-one/files/12586592/esp-web-tools-logs.txt)

EverythingSmartHome, 2023-09-13:
Indeed strange, it seems more like a block or ban on the AP side, as the EP1 is certainly finding it correctly during the scan and attempting to connect to it. Is there anyway it could have been blacklisted or blocked from joining? For example on Unifi there is the block device button (which I've accidently hit once before without realising it)

NicoLouis-be, 2023-09-13:
I had the same problem with Unifi, had to add a manual IP/ssid/... in the code and that fixed it for now. But i couldn't get it connected after the update of code

Nuuki9, 2023-09-13:
> Indeed strange, it seems more like a block or ban on the AP side, as the EP1 is certainly finding it correctly during the scan and attempting to connect to it. Is there anyway it could have been [...]

I am indeed running a Unifi network. Just double checked, and its not Blocked. Everything else continues to work fine and as I've tried several SSIDs I'm not sure there's much else I Can do from the AP config side (though see below). I have recently upgraded to the latest Network version (7.5.174), though that didn't precisely coincide with the issue anyway so I don't think that's related.

Sounds like @NicoLouis-be has seen similar issues. If you can remind me how I can edit the code I'm happy to try that - do I just grab the [everything-presence-one.yaml](https://github.com/EverythingSmartHome/everything-presence-one/blob/main/everything-presence-one.yaml) file, make the edits, and deploy it using ESPHome?

NicoLouis-be, 2023-09-13:
Added this, where i have my ssid and password in the secret file so i can copy paste it to all my esphome devices
```
wifi:
  fast_connect: ${hidden_ssid}
  ap: {}
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  manual_ip:
    static_ip: 
    gateway: 
    subnet: 
```

Nuuki9, 2023-09-13:
Thanks - I have a wifi block from another device I can use, but you're right that I should be using !secrets. I'm struggling with remembering how to deploy the YAML file - if I use https://web.esphome.io/ its not giving me an option to deploy YAML. My local ESPHome instance runs, but as its over HTTP I can't use it for the initial deployment I don't think, only subsequent management.

Am I getting somethingh wrong there? I can look to get ESPHome running on HTTPS via my reverse proxy but its a slight pain that I'd rather avoid if I don't have to do that right now :-)

Nuuki9, 2023-09-13:
I just watched one of Lewis' videos and reminded myself what I was doing. Am comiling the firmware with my changes now and will then give it a try. Will report back shortly )hopefully)...

Nuuki9, 2023-09-13:
I'm now starting to suspect a hardware issue...

I was able to compile and deploy the YAML with my own wifi settings - tried multiple SSIDs again but no luck - I'm still seeing "auth expired" errors. I double checked another ESP device I have and that was fine.

Doing some searching around, I found multiple threads that referenced a possible power issue. I then realised that the status LED is flashing constantly - the one on the board, not the mmWave sensor. I've tried switching to an external power supply (rather than from the laptop I was using to flash it) and same issue - I'm using a 60W GaN charger, which I know works and obviously has no issue with the necessary power.

Does the flashing LED indicate an issue? If it does then that is presumably related to the root cause, and the wifi problems are likely a symptom.

EverythingSmartHome, 2023-09-13:
No, the LED is a status LED, it will flash when either WiFi is not connected or the Home Assistant API is not connected, that's it's purpose.

I don't believe it to be a hardware thing personally, I've seen similar problems on my own Unifi APs when using esp development boards with the same error message.

Can you try and create a hotspot on your phone or laptop and try to get it connected to that? As you say, it worked previously so it's unlikely for hardware to suddenly go bad and stop connect, especially when it's able to scan for APs fine and bring up its own AP too without issue. Not impossible but seems unlikely at this point

NicoLouis-be, 2023-09-13:
Is the device still in your unifi list of devices? Maybe a force delete to clear it's history in unifi? How long is your dhcp release set to?

NicoLouis-be, 2023-09-13:
Glad to hear that fixed it

If you create a random new device in esphome the api tag will also give you a key. I also kept that one in the code, not sure if its needed

Nuuki9, 2023-09-13:
Thanks, I'll take a look at that - either way the original issue is resolved so I'll close this ticket.

closed 2023-09-13 by Nuuki9