EverythingSmartHome/everything-presence-one #103: Lux Value Scaling/Compensation for Case Cover Needed

jrstubbington, 2023-09-25:
## Expected Behavior
Lux value can be calibrated to compensate for front cover of EP1

## Actual Behavior
Lux can only be adjusted via static offset, leading to "inaccurate" measurements compared to the front cover being off.

## Background
Was comparing the EP1 and the FP1 and noticed the lux values being drastically different - 45lux reported on the Aqara FP1 and the ESH EP1 was reporting 7lux with the front cover on. Grabbed my Sekonic L478D light meter and recorded a value of ~43lux. Took the front cover off the EP1 and it reported 43.7lux, so significantly more accurate. Ensured that the BH1750 had a clear line through the case opening to the light source (an open window) but it didn't affect the results.

If the EP1 went through all the trouble to use a scientifically accurate way of measuring the light value I would hope there's a way to compensate for the covering. Changing the offset to 40lux in the ESP Home config led to a value of 40lux in a pitch black room. 

For reference, the Aqara FP1 _seems_ to be performing this compensation in some manner, as I have it situated in a very similar spot and it's pretty much spot-on with its measurements (but it's sensitivity is much lower than the EP1 in my tests).

## Ask
Could some form of non-offset compensation be added to the EP1?'s lux reporting sensor?

fhold, 2023-11-02:
I'm having the same behaviour the lux value is way off with the standard case.

jrstubbington, 2023-11-03:
So over the past day or so, I've been playing around with adding a helper in homeassistant to scale the EP1's lux value to deal with the case with a reasonable amount of success.

[image]

Dark Blue: Aqura FP1 Illuminance 
Purple: EP1 Illuminance 
Light Blue: "Scaled" EP1 Illuminance

Case Details:
* Printed at home from the source files
* Oveture PETG - Signal White

Find the scaling equation below - note this is not very accurate right now as I've mostly been doing guess and check work then finding constants to prevent boundary conditions. I'm sure there's an easier way to derive this equation and it's constants but it's a decent starting point if anyone wants to play with it.

```jinja2
{{ states('sensor.everything_presence_one_7e9e1c_illuminance') | float * 2 ** 2.6 + 0.02715248724 | log + 3.60629 | round(2) }}
```

```
Log[(x 2)^2.6 + 0.02715248724] + 3.60629
```

I'll continue playing with this and see if there's be a way of natively adding a lux compensation equation to the code directly and add any updates to my equation here.

Dc1975, 2023-11-23:
I have the same problem. The cover of the sensor reduces the lux value by approx. 10 units.
Unfortunately, changing the offset is not a workaround, as the set offset is then displayed as the current value in a dark room.

NicoLouis-be, 2023-12-05:
> So over the past day or so, I've been playing around with adding a helper in homeassistant to scale the EP1's lux value to deal with the case with a reasonable amount of success.
> 
> ![image](https://private-user-images.githubusercontent.com/10409281/280173329-318b288d-5abf-4d59-aeb1-effde36355bf.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTEiLCJleHAiOjE3MDE3ODU2ODUsIm5iZiI6MTcwMTc4NTM4NSwicGF0aCI6Ii8xMDQwOTI4MS8yODAxNzMzMjktMzE4YjI4OGQtNWFiZi00ZDU5LWFlYjEtZWZmZGUzNjM1NWJmLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFJV05KWUFYNENTVkVINTNBJTJGMjAyMzEyMDUlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjMxMjA1VDE0MDk0NVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWE1YzcwYTIzODI0OTY2ZWJjOWVhMzRiYjBjNmU0MmQ5NzQxOTI2NjkyNzIwYzJjNTQ4YmQ2ZjgwOWMwNDEzMDUmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.uoWGHoqQJhPas2JkeHnqllyGO-4pctRqPMKSLJ3dEWI)
> 
> Dark Blue: Aqura FP1 Illuminance Purple: EP1 Illuminance Light Blue: "Scaled" EP1 Illuminance
> 
> Case Details:
> 
> * Printed at home from the source files
> * Oveture PETG - Signal White
> 
> Find the scaling equation below - note this is not very accurate right now as I've mostly been doing guess and check work then finding constants to prevent boundary conditions. I'm sure there's an easier way to derive this equation and it's constants but it's a decent starting point if anyone wants to play with it.
> 
> ```
> {{ states('sensor.everything_presence_one_7e9e1c_illuminance') | float * 2 ** 2.6 + 0.02715248724 | log + 3.60629 | round(2) }}
> ```
> 
> ```
> Log[(x 2)^2.6 + 0.02715248724] + 3.60629
> ```
> 
> I'll continue playing with this and see if there's be a way of natively adding a lux compensation equation to the code directly and add any updates to my equation here.

some parentheses needed to make the round(2) to work, 
just installed this as a template sensor, will also test this

`{{ (states('sensor.everything_presence_one_f0f104_illuminance') | float * 2 ** 2.6 + 0.02715248724 | log + 3.60629) | round(2) }}`

EverythingSmartHome, 2023-12-24:
@NicoLouis-be @jrstubbington Sorry for the delay here, that looks good. How are you finding it now?

Have either of you noticed the calibrate_linear filter that might also work here and might be easier to use?

https://esphome.io/components/sensor/index.html#calibrate-linear

EverythingSmartHome, 2024-01-11:
If anyone is still tracking this, would be great if you can try out the latest firmware which uses calibrated values and should drastically change the lux values. Feedback appreciated

NicoLouis-be, 2024-01-11:
I had installed that piece of code like a month ago, and it did work better for me. I did compare it with some other sensor but haven't saved the result. 
i was happy with it and never thought of it afterwards.

Did you add this code in the new firmware or came up with a different solution?

EverythingSmartHome, 2024-01-11:
That makes sense thanks. I added a filter using the calibrate linear function and added a bunch of values with and without the case on with light sources at all varying angles and intensity which has helped a lot

dobromir-hristov, 2024-01-14:
I bought the Shelly BTU door sensor which has a lux sensor. It was reporting values of 1k lux compared to the EP1 80 lux on a sunny day. I thought surely the door sensor was off, the EP1 is right. A week later I removed the EP1 cover and they reported the same. 😳

Glad to see an update to fix the issue. I will try it and report soon.

EverythingSmartHome, 2024-01-15:
> @EverythingSmartHome I got -0.4 lux when it's dark with your solution in the last release

Just need to add a simple clamp to resolve. It's on my list. What about the actual light level in normal conditions?

EverythingSmartHome, 2024-01-16:
@NicoLouis-be Can you try the latest version, 1.1.9 if you have time? Should have the fix for negative values.

NicoLouis-be, 2024-01-16:
First impression looks good, value jumped from -0.4 to 0.
I will see tomorow morning how it went during the night

[image]

EverythingSmartHome, 2024-01-16:
Appreciate you testing, thanks for the feedback.

How do you find the values now compared to previously during the day? Do you notice a significant uplift?

closed 2024-07-19 by EverythingSmartHome