EverythingSmartHome/everything-presence-one #106: Connect to WiFi network without SSID broadcast

kisst, 2023-10-20:
When the SSID isn't broadcasted ( aka it's hidden) the Everything Presence One does not connect to the WiFi network, all other WiFi device I have had no trouble with it and now I had to change the WiFi only for this, it would be nice if I can get back to "normal"

EverythingSmartHome, 2023-10-20:
This is a standard ESPHome featured as described here under "fast_connect": https://esphome.io/components/wifi.html#configuration-variables

The EP1 has a substitution built in to allow you to modify this behaviour easily, you just need to add the hidden_ssid parameter to your substitutions section:

```
substitutions:
  name: "everything-presence-one"
  friendly_name: "Everything Presence One"
  ...
  hidden_ssid: "true"
  ```

kisst, 2023-10-20:
I am a rookie to the whole ESP ecosystem, so I don't even know where to start, where to stick this config, I had a great difficulty to get even to this point, so some hit for where to read up next would be great.

Of course at least some of the pain is my own doing, like I use the whole setup in an air-gapped offline setup, so even to follow the instructions listed on the box's QR code, I had to start with a server build to host a clone of your update page, then install a new browser which does support web-console, then temp whitelisting china domains and then IP ranges to download the driver, I don't have a single windows or mac device in my whole network, I run Hass in Kube so no USB, no Add-ons, etc etc 

From UX point of view, when I seen that out of the box it broadcasts a WiFi in AP mode, and runs a DHCP too, I expected to be able to just connect to it via a browser, setup my own network, and be on my way, and the reality was ... painfully over-complicated , but the least would be that the WiFi setting page would have a tick-box for hidden SSID.

So while so far I am happy with the HW itself, the User Experience is bumpy to say the least.

EverythingSmartHome, 2023-10-20:
The post is kind of contradictory, at least to me. You mention the setup is painfully over-complicated, then go onto mention how you run the entire setup in an air-gapped offline fashion. I'd say that the over-complicatedness is entirely a result of your own doing - which is totally fine, I get why you do it that way, just pointing out there the true issue lies.

If you are a regular user, the setup process takes like 2-3 minutes realistically and you don't need to jump through any of those hoops. CH340 drivers are included with most most Linux OS's too so not sure why you had an issue, unless you meant that you have no Windows, Mac or Linux systems at all in which case, yeah it's going to be tricky without access to some for of computer.

The web browser was removed in a previous update because it uses up a lot more memory and has the potential to cause a lot more bugs - user feedback is that this method of using the browser is much better and I personally think its nicer and quicker too.

EverythingSmartHome, 2023-10-20:
As for the original issue, did you manage to get it added to your own ESPHome dashboard yet? You mention you don't use add-ons in Home Assistant so to get it to be auto discovered in the ESPHome docker container, your going to need to make sure mDNS is working. Otherwise you can just create an entirely new config based off the original which is here and make the changed needed.

https://github.com/EverythingSmartHome/everything-presence-one/blob/main/everything-presence-one.yaml

kisst, 2023-10-21:
You are absolutely right that most of the complication comes from my setup, maybe even the expectations misalignment is on me, but let me share what I expected and you can tell me if this is even possible.

I get the device, I plug it in, it strat to broadcast a temp setup WiFi, I connect to it, open a browser and complete the setup, check it's MAC so I can whitelist it for the WiFi and DHCP, then I configure my WiFi on it, reboot it and done.

This kind of "if not set up" then "go into setup mode" is fairly typical, and shouldn't impact performance as only behave like so until it's set up. Unless you are running low on storage as you would need at least a lightweight web server even if it is just a python execution environment with a small flask script.

My point is that if the device is at least semi functional out of the box and has a ready TCP stack, then I wouldn't expect that I will need to plug it into a PC to set it up. I fully understand why you would need to do that for upgrade and firmware switch between Hass and Smartthings but when large percentage of the users use Hass anyway, and only want to change 1-2 parameters, I would think that I should be able to do that across the existing TCP/IP stack rather than USB serial and a public website combined with a selected few webbrowser.

Come to think of it I might had sanboxing issue with my browser as the reason I went down the rabbit whole soo deep down is that in my default Linux mashine the serial connections never shown up even after temporary whitelisting the device in usbguard.

But back to the original issue I tried to raise, during the update process, after flushed with the Hass flavoured variant, there is a screen for WiFi setup. which first does s scan and you can give credentials for the discovered networks, OR you can provide the name and the credentials for non discovered networks. On this page a check box for SSID hidden would be already a big help, and if it even tells me the device MAC even better, so I don't have to hunt that from the console log.

Lastly to answer your question, no I haven't managed to get an ESPHome Dashboard running yet, as a quick startup of the docker version also suggested that to me on the website interface that to add a device I will need to connect it via USB serial first, but knowing that based on mDNS it should discover the already set up device, I can give it a try with different networking setup, I will be happy to report back on progress once I have any. While we went pretty off topic from the original report I still think that setup over WiFi should be doable and would make a huge difference in user experience. 

btw. Thanks for the support and thanks for the development of the HW, I am currently working on a development for a case design which will enable the sensor to be mounted into spot light socket, and once ready will be more than happy to offer it as my contribution back to the project.

kisst, 2023-10-22:
Update: Had no luck with the mDNS based discovery, but I managed to connect via `/etc/hosts` fixed entry, not ideal but with this way at least I managed to upload a new config, so enabled the BLE proxy as well as set up the WiFi with fast connect and hidden SSID.

Right now I am set and problem free, in the meantime found also a 3D model for [spot light mount](https://cults3d.com/en/3d-model/gadget/everything-presence-one-in-ceiling-wall-and-surface-mount) too, I  will try and if works will just leave my own design.

However as this was just my first sensor , to try it out, and apart from setup pain, I am happy with the sensor data, I am considering to order quite a few more, and it would be great if the setup process would be updated ideally to the point of no USB serial needed, but at least if it is needed then the [update page](https://everythingsmarthome.github.io/everything-presence-one/updating.html) after the latest version upload, as part of the WiFi setup, it would ask few more question, like fast connect, hidden-ssid, etc.

kisst, 2023-10-24:
So I took a deep dive to unknown, and this is what I found.

This setup page use [esp-web-tool](https://github.com/EverythingSmartHome/everything-presence-one/blob/main/static/updating.md?plain=1#L77)  which use [improv-wifi](https://github.com/esphome/esp-web-tools/blob/main/index.html#L282) to configure the WiFi based on the [RPC command](https://github.com/improv-wifi/website/blob/main/src/serial.md?plain=1#L93)  which does not have a parameter for hidden networks, it only takes SSID / Password, however reading the specs I discovered that the same config is available via [BLE too](https://github.com/improv-wifi/website/blob/main/src/ble.md?plain=1#L88) which means in practice that the provisioning experience of the device could be made much smoother, and once on a WiFi then a config update can be pushed out wirelessly later.

@EverythingSmartHome what's your take ?

EverythingSmartHome, 2023-10-24:
Sorry for the delay - yes that's what I was going to say to your earlier messages, that there is no option for adding a hidden network through esp-webtools as far as I know.

I'm not sure how what you described would work in practice sorry, I'm not a developer and I just use the standard esp-web-tools that is provided as it would be far beyond my coding ability to make it work any other way. It's the same way that big projects like [ESPhome](https://web.esphome.io/), [Tasmota](https://tasmota.github.io/install/), [WLED](https://install.wled.me/) and others all use.

kisst, 2024-06-20:
@EverythingSmartHome 
I don't have a Google account hence can't comment under the YT video but wanted to say it big Thank you! https://www.youtube.com/watch?v=xpa5pCjljy0&t=173
The factory firmware and HA discovery is awesome, now I am only waiting for the PoE version ;)
Can't test it myself yet but you might be able to just close this issue as resolved. ( hopefully )

closed 2024-07-19 by EverythingSmartHome