EverythingSmartHome/everything-presence-one #119: $JYRPO declaration issue?

svente08, 2023-11-08:
Hi all,

Trying to get 1.3.0b running. Installation works but multi target distance not really.

Could declaration be the problem?

[logs_esphome-web-8bd630_run.txt](https://github.com/EverythingSmartHome/everything-presence-one/files/13301340/logs_esphome-web-8bd630_run.txt)

Thx in advance

EverythingSmartHome, 2023-11-09:
Did you try toggling UART target off, waiting 10s then turning back on again?

svente08, 2023-11-11:
No success. As soon as I activate a second zone, only inconsistent values are displayed.

EverythingSmartHome, 2023-11-12:
Can you post logs and detail what the issue your seeing is?

svente08, 2023-11-13:
Attached logfiles and  HA screenshot. (moving target more than 2 meters away)

[Archive.zip](https://github.com/EverythingSmartHome/everything-presence-one/files/13334492/Archive.zip)

(no bh1750/shtcx sensor connected)

madninjaskillz, 2023-11-29:
Simple work around for that; change
`if (line.substr(0, 6) == "$JYRPO") {`
to
`if (line.substr(1, 6) == "JYRPO") {`

I'm not sure if the substr here is end or number of characters - if its the latter, you'll need to change the 6 to a 5 for the functionality to still work, but at the least, this change DOES get rid of the warning.

Djelibeybi, 2023-12-17:
It needs to be `if (line.substr(1, 5) == "JYRPO")` which at least gets me some values, but I'm not entirely sure if they're correct.

svente08, 2023-12-30:
I got rid of the error message - thx for that. But the values still not correct. Target 1 ist always active and shows the end value of target.

@Djelibeybi  What do you mean by configure correctly... Talking about start and end points of each beginning with target 1 and so on? Or Is there anything else to consider?