EverythingSmartHome/everything-presence-one #124: Adopt EP1 Fails

tibmeister, 2023-12-08:
When trying to adopt EP1 into ESPHome, I receive a lot of errors about the yaml file formatting.

```
INFO Reading configuration /config/esphome/everything-presence-one-d94ad4.yaml...
INFO Updating https://github.com/everythingsmarthome/everything-presence-one.git@main
WARNING GPIO15 is a Strapping PIN and should be avoided.
Attaching external pullup/down resistors to strapping pins can cause unexpected failures.
See https://esphome.io/guides/faq.html#why-am-i-getting-a-warning-about-strapping-pins
Failed config

esphome: None
  name: everything-presence-one-d94ad4
  
  [friendly_name] is an invalid option for [esphome]. Please check the indentation.
  friendly_name: Everything Presence One None
  name_add_mac_suffix: False
  project: 
    name: Everything Smart Technology.Everything Presence One
    version: 1.1.7
  build_path: .esphome/build/everything-presence-one-d94ad4
dashboard_import: [source /config/esphome/.esphome/packages/405587a8/everything-presence-one.yaml:44]
  package_import_url: |-
    github://everythingsmarthome/everything-presence-one/everything-presence-one.yaml@main
  
  [import_full_config] is an invalid option for [dashboard_import]. Please check the indentation.
  import_full_config: False [source /config/esphome/.esphome/packages/405587a8/everything-presence-one.yaml:45]
binary_sensor.gpio: [source /config/esphome/.esphome/packages/405587a8/everything-presence-one.yaml:117]
  platform: gpio
  pin: 
    number: 33
    mode: INPUT_PULLDOWN
  name: PIR
  id: pir_motion_sensor
  device_class: motion
  filters:  [source /config/esphome/.esphome/packages/405587a8/everything-presence-one.yaml:125]
    - [source /config/esphome/.esphome/packages/405587a8/everything-presence-one.yaml:125]
      
      This option is not templatable!.
      delayed_off: !lambda |-
        return id(pir_off_latency).state * 1000.0; [source /config/esphome/.esphome/packages/405587a8/everything-presence-one.yaml:125]
binary_sensor.template: [source /config/esphome/.esphome/packages/405587a8/everything-presence-one.yaml:126]
  platform: template
  name: Occupancy
  id: occupancy
  device_class: occupancy
  filters:  [source /config/esphome/.esphome/packages/405587a8/everything-presence-one.yaml:131]
    - [source /config/esphome/.esphome/packages/405587a8/everything-presence-one.yaml:131]
      
      This option is not templatable!.
      delayed_off: !lambda |-
        return id(occupancy_off_latency).state * 1000.0; [source /config/esphome/.esphome/packages/405587a8/everything-presence-one.yaml:131]
  on_state: 
    - lambda: id(illuminance_sensor).update();
  lambda: |-
    if ( id(mmwave).state or id(pir_motion_sensor).state) {
      return true;
    } 
    else if (id(mmwave).state == 0 and id(pir_motion_sensor).state == 0) {
      return false;
    } 
    else {
      return id(occupancy).state;
    }
switch.template: [source /config/esphome/.esphome/packages/405587a8/everything-presence-one.yaml:158]
  platform: template
  name: mmWave sensor
  id: mmwave_sensor
  disabled_by_default: True
  entity_category: config
  optimistic: True
  
  [restore_mode] is an invalid option for [switch.template]. Did you mean [restore_state]?
  restore_mode: RESTORE_DEFAULT_ON [source /config/esphome/.esphome/packages/405587a8/everything-presence-one.yaml:164]
  turn_on_action: 
    - uart.write: sensorStart
    - delay: 1s
  turn_off_action: 
    - uart.write: sensorStop
    - delay: 1s
```

EverythingSmartHome, 2023-12-09:
Hello - seems like you are using quite an old version of the ESPHome add-on, please update to the latest and try again

tibmeister, 2023-12-11:
Not sure how I missed that one!  It's working as expected now.

closed 2023-12-11 by tibmeister