EverythingSmartHome/everything-presence-one #130: EP1 is over priced for the defects it has

idkpmiller, 2024-01-08:
I have 3 x EP1 and 4 x EPL unitas, I love my EPL and hate my EP1's with a passion.
The EP1 lux is a joke and is not usable.
The distance is inaccurate compared to the cheaper EPL.
Every time I get an update it takes mulriple attempts to get each EP1 unit to upgrade and the EPL just work first time.
I have seen people asking for the lux rto have a better offset. Ido not think this is a suitable solution, this is a design flaw with the caser provided and needs to be fixed by a new case or front being provided to all those that purchased one that is not fit for purpose.
I also recall that we were expecting that the bluetooth proxy wiould be coming real soon, it is available on the EPL but I still do not see it on the more expensive EP1.

A really upset customer.

EverythingSmartHome, 2024-01-08:
Hello,

Sorry to hear you aren't happy with them, I wish you had contacted us via support to ask about the queries rather than feeling frustrated about them. We are here to help!

Can you let me know which version you on? I'm not clear as you mention distance being inaccurate (which is only in the beta) but then say you struggle to update which doesn't happen in the beta making me think your on the stable version. If you can confirm the version that would be great.

Regarding Lux, I actually have an update in the works currently I was working on over the holidays which I'm hopeful will improve things a good bit. I'll try to get it out soon.

As for distance tracking,firstly it was never an advertised feature of the EP1 to track distance, it was only ever added much later in Beta (and still was never a listed feature) and is still in beta only, so I'm not sure it's reasonable to be disappointed by an additional non advertised feature that is in beta. That said, I personally find the distance to be pretty accurate (and also did well in comparison videos from other creators which included other sensors and the Lite). Unfortunately there isn't anything I can change about the way distance tracking works as it's baked into the mmWave sensor firmware which we don't have access to.

Bluetooth proxy can be quite easily added via your own config as always with ESPHome. It will be coming by default to the Beta firmware, however I'm still waiting on some feedback with beta and don't want to change too many things at once. I can perhaps create an additional firmware to include Bluetooth proxy, but be aware it can cause intermittent WiFi issues (that goes for all ESPHome devices and not exclusive to the EP1/Lite) when both Bluetooth and WiFi are active.

I'm sorry if you are unhappy, I'll try to make those changes where I can for you

EverythingSmartHome, 2024-01-09:
FYI I just added an update that should improve the light level readings when the cover is on, I would love if you gave it a try. I haven't finished adding all the calibration data just yet but it should drastically improve the values your seeing as is.

idkpmiller, 2024-01-09:
Great to hear that the EP1 is getting fixes. Just so we are clear. The EPL is awesome and worth irs mony. It is hard to see that the two devices are from the same stable the difference is that big. I have been contemplating changing out the EP1 units for EPLs.
I will try the new firmware when it comes out to see if lux is more accurate. If it is a static offset applied then I will stand by what I said its a case design flaw.

EverythingSmartHome, 2024-01-09:
The update is out and you can give it a try now. It's not a static offset and is applied out the box so you should see a noticeable change in the before and after

idkpmiller, 2024-01-10:
Not seen any recent update in HA for them. They are currently running 2023.12.5 26th december.

Sent from a mobile device, as a result message may be concise

EverythingSmartHome, 2024-01-10:
That's the ESPHome update but not the EP1 config. You need to go into the ESPHome update and install the config over WiFi or cable at which point it should update. Co fig updates don't show in the home assistant update panel unfortunately

lanyar, 2024-04-08:
> Great to hear that the EP1 is getting fixes. Just so we are clear. The EPL is awesome and worth irs mony. It is hard to see that the two devices are from the same stable the difference is that big. I have been contemplating changing out the EP1 units for EPLs. 

What is an EPL?

markmghali, 2024-04-08:
> > Great to hear that the EP1 is getting fixes. Just so we are clear. The EPL is awesome and worth irs mony. It is hard to see that the two devices are from the same stable the difference is that big. I have been contemplating changing out the EP1 units for EPLs. 
> 
> What is an EPL?

Everything presence lite! Go see it at the shop

sidlos, 2024-04-27:
I am frustrated also with ESP1. Can you please tell me where should I report not functioning ESP1? Only email I could find is orders@everythingsmart.io but I have not received any response in week.

EverythingSmartHome, 2024-04-27:
I don't believe we have any emails from you, please email again and I will response ASAP - perhaps I accidentally missed it.

closed 2024-07-19 by EverythingSmartHome