EverythingSmartHome/everything-presence-one #133: Updated EP1 won't now work

BenHornerTech, 2024-01-11:
I updated my EP1 via USB connection to my laptop and the website.  Following the update it reconnected to my Wifi, but doesn't seem to reconnect to Home Assistant.  When I try to connect it to home assistant ESPHome says it's already set up and the logs show this (IP redacted on purpsose):

`WARNING everything-presence-one-272784 @ xxx.xxx.xxx.xxx: Connection error occurred: everything-presence-one-272784 @ xxx.xxx.xxx.xxx: The connection dropped immediately after encrypted hello; Try enabling encryption on the device or turning off encryption on the client (ESPHome Logs 2023.12.5).
WARNING Can't connect to ESPHome API for everything-presence-one-272784 @ xxx.xxx.xxx.xxx: Error while finishing connection: Finishing connection cancelled due to fatal exception: everything-presence-one-272784 @ xxx.xxx.xxx.xxx: The connection dropped immediately after encrypted hello; Try enabling encryption on the device or turning off encryption on the client (ESPHome Logs 2023.12.5). (HandshakeAPIError)`

BenHornerTech, 2024-01-11:
Also see this in the startup logs:

`D][switch:012]: 'mmWave sensor' Turning ON.
[D][switch:055]: 'mmWave sensor': Sending state ON
[D][switch:012]: 'mmWave sensor' Turning ON.
[D][switch:012]: 'mmWave sensor' Turning ON.
[D][uart_debug:158]: >>> "sensorStartsensorStartsensorStart"
[D][uart_debug:158]: <<< "sensorStart\r\n"
[D][uart_debug:158]: <<< "new parameter isn\'t save, can\'t startSensor\r\n"
[D][uart_debug:158]: <<< "Error\r\n"
[D][uart_debug:158]: <<< "leapMMW:/>sensorStartsensorStart\r\n"
[D][uart_debug:158]: <<< "sensorStartsensorStart is not recognized as a CLI command\r\n"
[D][uart_debug:158]: <<< "Error\r\n"
[D][api:102]: Accepted xxx.xxx.xxx.xxx
[W][api.connection:107]: : Reading failed: BAD_INDICATOR errno=11
[D][status_led:030]: Restoring light state OFF
[D][uart_debug:158]: <<< "leapMMW:/>"
[D][api:102]: Accepted xxx.xxx.xxx.xxx
[W][api.connection:107]: : Reading failed: BAD_INDICATOR errno=11`

EverythingSmartHome, 2024-01-11:
It looks like you've had it setup in the past in Home Assistant with an encryption key added, and then when you've done the update it no longer has the encryption key (doing it via the user guide doesn't have an encryption key), but Home Assistant is still trying to connect using the encryption key and thus failing.

Try removing from HA and re-adding, or do the same as you did before and flash using the encryption key in the ESPHome add-on.

BenHornerTech, 2024-01-11:
Thanks for the quick reply.  Re-adding to HA sounds like a fair bit of work, is there a guide for re-flashing with the API key from ESPHome?

spmaid, 2024-01-12:
Seeing the exact same issue, and I think same cause, wanting to update the firmware, guessing I've gone about the update the wrong way?

EverythingSmartHome, 2024-01-12:
This is not an EP1 firmware thing and the above link can and is used by many to update the firmware to a device that has already been added.

I'll try to explain why it's happened a little better - it occurs when you've flashed the EP1 with the stock firmware, you then add it to home assistant (all fine till this point), then you add it to the ESPHome addon. At this point, the ESPHome addon asks you if you want to an an encryption key, which most users don't read and just hit "next" without realising it and end up flashing the ep1 again with an encryption key. Then when you update the firmware with the above link, it ends up overwriting your encryption key (that you set) and home assistant can no longer connect as it's trying to use that key.

If you stick with the stock firmware and add it to ESPHome without adding an encryption key, you won't have the issue. Or if you add it to ESPHome, add an encryption key, then only update using your ESPHome config then you won't have the issue.

Hopefully that makes sense

BenHornerTech, 2024-01-12:
Many thanks for the detailed response.  If this comes up regularly it may be worth adding to the update page or the troubleshooting page, as it wasn't entirely clear to me :-)