EverythingSmartHome/everything-presence-one #137: mmWave LED sometimes on after Reboot/Power-cycle/Firmware upgrade

bakerkj, 2024-01-15:
Physical mmWave LED sometimes on after Reboot/Power-cycle/Firmware upgrade even though home assistant switch indicates that the LED is off.

jokerigno, 2024-01-15:
+1 on this.

I found a workaround turn on esp 32 led wait few sec and then turn them off

TB-SE, 2024-01-17:
+1 same issue

TNTPro, 2024-01-19:
I'm having the same issue. Also, when the mmWave light is on, the sensitivity of the mmWave seems to be bad, as my room lights keeps turning off, like I'm not there.

IneedAnewNick, 2024-01-19:
Yep same issue here too

marcinwasilewski, 2024-01-21:
Isn't the LED on until the config (sensor sensitivity settings etc) is written to flash ?

IneedAnewNick, 2024-01-21:
This only happens on one of my three changing a setting does turn it off but as I say only one out of three does this.

EverythingSmartHome, 2024-02-05:
Can you please try the latest version? I added a quick couple of lines which I hope will help to ensure that the sensor is started after boot. The new version is v1.1.10, you may need to flash over USB using the user guide if you still have the old cached

bakerkj, 2024-02-06:
@EverythingSmartHome I'll give it a try. Thanks!!

TB-SE, 2024-02-06:
> The new version is v1.1.10, you may need to flash over USB using the user guide if you still have the old cached

I can't see the new version on Github. The latest I see is 1.1.9

EverythingSmartHome, 2024-02-07:
> > The new version is v1.1.10, you may need to flash over USB using the user guide if you still have the old cached
> 
> I can't see the new version on Github. The latest I see is 1.1.9

It's not listed as a Github release, but it is there. Just update via the ESPHome add-on or the user guide

bakerkj, 2024-02-11:
@EverythingSmartHome this seems to continue to occur even with the latest changes.

EverythingSmartHome, 2024-02-11:
> @EverythingSmartHome this seems to continue to occur even with the latest changes.

I would try again if you haven't updated today, I added some further changes that most definitely resolve this from my own testing.

bakerkj, 2024-02-12:
@EverythingSmartHome I didn't see that you'd pushed more changes. I'll give it another go. 

Thanks!

jokerigno, 2024-02-20:
USB is the only way to update latest fw?

danielsvane, 2024-02-23:
Im also unsure how to best update. Do I have to connect it to a PC, or can I do it through esphome integration? This is what it looks like from my esphome tab. Im scared to click the button, since its working fine (the red LED is just constantly on).

[image]

olympia, 2024-03-26:
I would be interested in the best update approach too (also in order to keep the settings).

emdekext, 2024-05-09:
Same problem here, everytime after an esphome firmware flash (update for instance) the mmwave LED stays on, but is reported as being switched of, and the mmwave sensor is reported as switched on but does not report any presence/movement. Only after writing settings to the mmwave sensor flash it starts to function again.

closed 2024-07-19 by EverythingSmartHome