EverythingSmartHome/everything-presence-one #138: Crashes running on ESPHome 2023.12.6

alexolivier, 2024-01-16:
Firsly have to say i love these sensors and the YouTube channel - very happy to support you however possible.

This is likely an ESPHome issue rather than EP1 specific, but since upgrading to ESPHome 2023.12.6 the PIR + mmWave sensors fail to register and ESP32 crashes:

```
[12:54:16][C][improv_serial:032]: Improv Serial:
[12:54:16][D][uart_debug:158]: >>> "sensorStartsensorStart"
[12:54:16][D][uart_debug:158]: <<< "sensorStart\r\n"
[12:54:16][D][uart_debug:158]: <<< "Done\r\n"
[12:54:16][D][switch:012]: 'mmWave sensor' Turning ON.
[12:54:16][D][uart_debug:158]: <<< "leapMMW:/>"
[12:54:16][D][uart_debug:158]: >>> "sensorStart"
[12:54:16][D][uart_debug:158]: <<< "sensorStart\r\n"
[12:54:16][D][uart_debug:158]: <<< "sensor started already\r\n"
[12:54:16][D][uart_debug:158]: <<< "Error\r\n"
[12:54:16][W][component:214]: Component uart took a long time for an operation (0.05 s).
[12:54:16][W][component:215]: Components should block for at most 20-30ms.
[12:54:16][D][uart_debug:158]: <<< "leapMMW:/>"
WARNING ep1-office @ 192.168.30.21: Connection error occurred: [Errno 104] Connection reset by peer
INFO Processing unexpected disconnect from ESPHome API for ep1-office @ 192.168.30.21
WARNING Disconnected from API
INFO Successfully connected to ep1-office @ 192.168.30.21 in 0.008s
INFO Successful handshake with ep1-office @ 192.168.30.21 in 0.186s
[12:55:04][D][uart_debug:158]: <<< "sensorStart\r\n"
[12:55:04][D][uart_debug:158]: <<< "Done\r\n"
[12:55:04][D][uart_debug:158]: <<< "leapMMW:/>"
[12:55:05][D][binary_sensor:034]: 'PIR': Sending initial state OFF
```

Full log
```
INFO ESPHome 2023.12.6
INFO Reading configuration /config/esphome/ep1-office.yaml...
WARNING GPIO15 is a strapping PIN and should only be used for I/O with care.
Attaching external pullup/down resistors to strapping pins can cause unexpected failures.
See https://esphome.io/guides/faq.html#why-am-i-getting-a-warning-about-strapping-pins
INFO Generating C++ source...
INFO Compiling app...
Processing ep1-office (board: esp32dev; framework: arduino; platform: platformio/espressif32@5.4.0)
--------------------------------------------------------------------------------
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch5
Dependency Graph
|-- AsyncTCP-esphome @ 2.0.1
|-- WiFi @ 2.0.0
|-- FS @ 2.0.0
|-- Update @ 2.0.0
|-- ESPAsyncWebServer-esphome @ 3.1.0
|-- DNSServer @ 2.0.0
|-- ESPmDNS @ 2.0.0
|-- noise-c @ 0.1.4
|-- Wire @ 2.0.0
|-- Improv @ 1.2.3
Compiling .pioenvs/ep1-office/src/main.cpp.o
Linking .pioenvs/ep1-office/firmware.elf
RAM:   [==        ]  16.5% (used 53944 bytes from 327680 bytes)
Flash: [==========]  95.6% (used 1754853 bytes from 1835008 bytes)
Building .pioenvs/ep1-office/firmware.bin
Creating esp32 image...
Successfully created esp32 image.
esp32_create_combined_bin([".pioenvs/ep1-office/firmware.bin"], [".pioenvs/ep1-office/firmware.elf"])
Wrote 0x1bdd70 bytes to file /data/build/ep1-office/.pioenvs/ep1-office/firmware-factory.bin, ready to flash to offset 0x0
========================= [SUCCESS] Took 24.37 seconds =========================
INFO Successfully compiled program.
INFO Connecting to 192.168.30.21
INFO Uploading /data/build/ep1-office/.pioenvs/ep1-office/firmware.bin (1760624 bytes)
Uploading: [============================================================] 100% Done...

INFO Upload took 14.82 seconds, waiting for result...
INFO OTA successful
[... 266 lines cut ...]
[12:54:16][C][api:140]:   Address: ep1-office.local:6053
[12:54:16][C][api:142]:   Using noise encryption: YES
[12:54:16][C][improv_serial:032]: Improv Serial:
[12:54:16][D][uart_debug:158]: >>> "sensorStartsensorStart"
[12:54:16][D][uart_debug:158]: <<< "sensorStart\r\n"
[12:54:16][D][uart_debug:158]: <<< "Done\r\n"
[12:54:16][D][switch:012]: 'mmWave sensor' Turning ON.
[12:54:16][D][uart_debug:158]: <<< "leapMMW:/>"
[12:54:16][D][uart_debug:158]: >>> "sensorStart"
[12:54:16][D][uart_debug:158]: <<< "sensorStart\r\n"
[12:54:16][D][uart_debug:158]: <<< "sensor started already\r\n"
[12:54:16][D][uart_debug:158]: <<< "Error\r\n"
[12:54:16][W][component:214]: Component uart took a long time for an operation (0.05 s).
[12:54:16][W][component:215]: Components should block for at most 20-30ms.
[12:54:16][D][uart_debug:158]: <<< "leapMMW:/>"
WARNING ep1-office @ 192.168.30.21: Connection error occurred: [Errno 104] Connection reset by peer
INFO Processing unexpected disconnect from ESPHome API for ep1-office @ 192.168.30.21
WARNING Disconnected from API
INFO Successfully connected to ep1-office @ 192.168.30.21 in 0.008s
INFO Successful handshake with ep1-office @ 192.168.30.21 in 0.186s
[12:55:04][D][uart_debug:158]: <<< "sensorStart\r\n"
[12:55:04][D][uart_debug:158]: <<< "Done\r\n"
[12:55:04][D][uart_debug:158]: <<< "leapMMW:/>"
[12:55:05][D][binary_sensor:034]: 'PIR': Sending initial state OFF
```

EverythingSmartHome, 2024-01-16:
Hello! Can you attach your config? It looks like you have Bluetooth proxy and/or ble tracking enabled which can cause disconnects, I would try removing if you have those enabled.

EverythingSmartHome, 2024-01-16:
You can try switching the framework to esp-idf which is the recommended for anything Bluetooth which may help

alexolivier, 2024-01-16:
Thanks will give that a go - and thanks for the prompt replies. All working as expected without BT so will close. Keep up the great work!

closed 2024-01-16 by alexolivier