EverythingSmartHome/everything-presence-one #142: Red LED stays on after firmware upgrade

Phadeguy, 2024-01-22:
Hey!

I haven't used my EP1 for a few months, but now I need it for a project.  I plugged it in and updated it to the latest and greatest [Firmware: 2023.12.8 (Jan 22 2024, 12:26:11), Hardware: 1.1.9], but now the center red LED won't turn off.  The "ESP32 status LED" is off and I've power-cycled it a few times just to be sure.  I can put a piece of tape over the light to "fix" it myself, but I figured I'd put in a bug report in case someone has the same issue (or a software fix).  Otherwise, the device still works.

Thank you for being awesome!

EverythingSmartHome, 2024-01-23:
It's a known issue and started with a recent release of ESPHome (I think?), the trick is to turn the mmwave LED on, wait 10s, then turn off again) - this seems to happen after a reboot or power cut I think

Phadeguy, 2024-01-23:
Ahh, gotcha.  It is an odd fix, but it's a fix, and it worked!  (Now I must think of a way to automate turning off the light.)  

Much appreciated, kind sir!!

closed 2024-01-23 by Phadeguy