EverythingSmartHome/everything-presence-one #143: mmWave sensor does not functio after update or reboot

yme207, 2024-01-28:
mmWave sensor does not work after firmware update. Also happens after restarting the device.  Have resorted to pulling power to device and then toggling mmLED mmWave sensor on/off switches multiple times. After multiple restarts it seems to come to life. Not clear what is solving the issue.  Running Firmware: 2023.12.9

cecat, 2024-02-05:
I have the same problem after the last update.  Red LED solid on, motion sensor working, mmWave not working. Power cycle does not fix.  I am also running 2023.12.9.

EverythingSmartHome, 2024-02-05:
Is anyone able to provide some startup logs? I cannot easily re-produce this issue myself unfortunately but I've had enough people report it that it does seem to be an issue, I just can't seem to make it happen myself. Some startup logs would be great if possible.

EverythingSmartHome, 2024-02-05:
Can you please try the latest version? I added a quick couple of lines which I hope will help to ensure that the sensor is started after boot. The new version is v1.1.10, you may need to flash over USB using the user guide if you still have the old cached

cecat, 2024-02-06:
I updated the firmware from 1.1.9 to 1.1.10 and it's back in business - sorry, should have checked that first.  I had to manually turn on the mmwave sensor, though.  I've got a second unit I have not yet updated - will see if that one also needs manual enabling of mmwave and report back tomorrow-

cecat, 2024-02-09:
Actually I misread - 1.1.10 is apparently the hardware rev level.  The first unit (my last comment) shows 1.1.10 but the troublesome unit (see below) shows:
Firmware: 2023.12.9 (Jan 24 2024, 00:07:17)
Hardware: 1.1.9

I did the firmware update (to 2023.12.9) on the second EP1 unit.  This one did not go well.  Did the update, selected "add to HA" and entered the unit's IP address.  Message "already added" (or something to that effect).  So I got a blinking red LED (this one down by the motion sensor, not up on the mmwave board as pre-update).  None of the controls are operative on the device page and all sensors show "unavailable."  Logs are below.  I am running HA OS v11.5, HA Core v2024.2.1, and Supervisor 2024.01.1 (all the latest versions).  

Log for the misbehaving EPO is [attached.]
[logs_everything-presence-one-2726b8_logs.txt](https://github.com/EverythingSmartHome/everything-presence-one/files/14227009/logs_everything-presence-one-2726b8_logs.txt)

So this one is dead at the moment. 
(sorry if this is the wrong place to report this- please do redirect me if so!)

cecat, 2024-02-12:
I notice my other EP1 does not have encryption turned on. I took out the encryption (api section) and things are working.  Not sure why this appeared in the code (I didn't add it!)

closed 2024-02-20 by EverythingSmartHome