EverythingSmartHome/everything-presence-one #144: Transition to new ESPHome DFRobot SEN0395 integration

bakerkj, 2024-02-05:
Can this custom integration with the DFRobot SEN0395 be transitioned to use the [DFRobot SEN0395 integration](https://esphome.io/components/dfrobot_sen0395) now shipped as part of ESPHome as part of [2023.11.0](https://esphome.io/changelog/2023.11.0)?

Snypes88, 2024-05-03:
@bakerkj this is my implementation using a SEN0395 custom file and then a Base custom file which includes all the necessary YAML code for the EPOne. @EverythingSmartHome are you considering using the oficial implementation anytime soon? I think there are some important bits of information that us users of the SEN0395 should be made aware of, like the fact that changing settings on the mmwave sensor writes to EEPROM and thus has a writecycle limit (kind of important to know imho).

`uart:
  id: uart_bus
  tx_pin: GPIO13
  rx_pin: GPIO14
  baud_rate: 115200
  debug:
    direction: BOTH
    dummy_receiver: true
    after:
      delimiter: "\n"
    sequence:
      - lambda: UARTDebug::log_string(direction, bytes);

dfrobot_sen0395:  
  id: mmWave_1
  uart_id: uart_bus 

binary_sensor:
  - platform: gpio
    name: Presence mmWave
    id: mmwave
    device_class: occupancy
    pin:
      number: GPIO15
      mode: INPUT_PULLDOWN
      

## Sen0395 common
button:
  - platform: template
    name: Restart mmW
    id: restart_mmwave
    entity_category: config
    internal: true
    on_press:
      - dfrobot_sen0395.reset:
    # on_press:  
    #   - dfrobot_sen0395.reset: {}  

  - platform: template
    id: restart_all
    name: Restart All
    icon: mdi:restart
    entity_category: config
    disabled_by_default: False
    internal: False
    on_press:
      - button.press: restart_mmwave
      - button.press: restart_internal
      
  - platform: template
    name: Restart Factory mmW 
    icon: mdi:cog-counterclockwise
    id: factory_reset_mmwave
    internal: ${factory_reset_disabled}
    entity_category: config
    on_press:
      - switch.turn_off: mmwave_sensor
      - delay: 1s
      - dfrobot_sen0395.settings:
          factory_reset: true
      - delay: 3s
      - switch.turn_on: mmwave_sensor

  - platform: template  
    id: trigger_mmwave_settings  
    # name: Trigger mmWave Settings  
    internal: True
[... 144 lines cut ...]

  - platform: dfrobot_sen0395
    # dfrobot_sen0395_id: mmWave_1
    name: LED mmWave
    id: mmwave_led
    type: turn_on_led
    disabled_by_default: False
    # entity_category: config
    # inverted: True

  - platform: dfrobot_sen0395
    # dfrobot_sen0395_id: mmWave_1
    name: mmW UART Presence
    id: uart_presence_output
    type: presence_via_uart
    disabled_by_default: False
    # entity_category: config
    # optimistic: true
    # restore_mode: DISABLED
    # turn_on_action:
    #   - uart.write: "sensorStart"
    #   - delay: 1s
    # turn_off_action:
    #   - uart.write: "sensorStop"
    #   - delay: 1s`

bakerkj, 2024-05-04:
@Snypes88, thank you for sharing your configuration! I had started down a similar path. 

I had troubles with the `uart` being in `dummy_receiver` mode. WIth this configuration enabled the `dfrobot_sen0395` c++ code did not see the responses from the device itself.

-- Ken

EverythingSmartHome, 2024-07-19:
Sorry for the delay! Not currently as I don't believe (please correct if wrong) it supports the target tracking sensors which a lot of people are using currently. But definitely willing to revisit it it becomes feature equal.

closed 2024-07-19 by EverythingSmartHome