EverythingSmartHome/everything-presence-one #157: Proximity Zones Not Updating After Firmware Update ('JYRPO' Not Declared Error)

ds-sebastian, 2024-02-19:
After updating the firmware on my Everything Presence One, to the latest Bluetooth proxy enabled beta firmware, I am experiencing issues with the proximity zones not updating. After using `mmWave Set Distance` it seemed to change from unavailable to off, but never updates beyond that despite the rest of the sensors working correctly (i.e remains `off`)

`Living Room Multisensor mmWave Zone 1 Start` is set to `15cm`
`Living Room Multisensor mmWave Zone 1 End` is set to `795cm`

Firmware: `2023.12.9 (Feb 12 2024, 15:54:35)`
Hardware: `1.3.4b`

I also receive warnings in the ESPHome logs regarding 'JYRPO' not being declared:
```
WARNING Found 'UARTDebug::log_string(direction, bytes)...' which looks like a substitution, but 'JYRPO' was not declared
...
INFO Successfully connected to everything-presence-one-26c304 @ 192.168.1.130...
```

I came across a related issue on GitHub where the user had a similar error involving _'JYRPO' not being declared_ where it was suggested toggling `UART target` off, waiting 10 seconds, and then turning it back on. However, I am unable to find the option to toggle UART target as a switch in Home Assistant.

First, are zones supported on the EP One? If, so, any suggestions to resolve?

EverythingSmartHome, 2024-02-20:
Do you get the same issue on the non bluetooth proxy version?

Distance zones are supported on the EP1, not XY coordinate zones. This is described in the user guide: https://everythingsmarthome.github.io/everything-presence-one/Home%20Assistant/home-assistant-entities.html#distance-beta

ds-sebastian, 2024-03-08:
@EverythingSmartHome 

I finally got around to testing the _non_ bluetooth proxy beta version and it appears to have the same issue where the Target 1 zone will not update (even after setting distance). Same warnings as above as well.

Firmware: `2024.2.2 (Mar 8 2024, 16:55:40)`
Hardware: `1.3.5b`

I also noticed `mmwave_zone_1_end` and `mmwave_zone_1_start` aren't persistent across rebooting the device, I'm not sure that's related or intentional

ds-sebastian, 2024-03-08:
I am not sure what I'm doing with ESPHome, but I tried to bring back the UART Target Output switch by adding the following to the ESPHome yaml:
```yaml
substitutions:
  name: everything-presence-one-26c304
  friendly_name: Everything Presence One 26c304
packages:
  Everything_Smart_Technology.Everything_Presence_One: github://everythingsmarthome/everything-presence-one/everything-presence-one-beta-ble.yaml@main
esphome:
  name: ${name}
  name_add_mac_suffix: false
  friendly_name: ${friendly_name}

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
# -------ADDED---------
switch:  
  - platform: template
    name: UART target output 2
    id: uart_target_output_2
    entity_category: config
    internal: ${uart_target_output_disabled}
    optimistic: true
    assumed_state: false
    restore_mode: ALWAYS_ON
    turn_on_action:
      - switch.turn_off: mmwave_sensor
      - delay: 1s
      - uart.write: "setUartOutput 2 1 1 1"
      - delay: 1s
      - uart.write: "saveConfig"
      - delay: 3s
      - switch.turn_on: mmwave_sensor
    turn_off_action:
      - switch.turn_off: mmwave_sensor
      - delay: 1s
      - uart.write: "setUartOutput 2 0"
      - delay: 1s
      - uart.write: "saveConfig"
      - delay: 3s
      - switch.turn_on: mmwave_sensor
```

The zone sensors began to work after hitting install! I removed my added switch (even though it never showed up 🤷) and they stayed working, so maybe just needed to be reinstalled? 

What's strange is I did an ESPHome firmware update (clicking `Update` in ESPHome) didn't fix the issue. Only after clicking `Install` within `Edit` yaml. Also flashing directly from [Everything Smart Home webite](https://everythingsmarthome.github.io/) didn't seem to work either with the BLE proxy or non-BLE proxy version.

closed 2024-03-08 by ds-sebastian