EverythingSmartHome/everything-presence-one #161: EP1 encryption key verification issue

idkpmiller, 2024-02-21:
Hi,
Two of my EP1 devices are playing up.
I have managed to move 4 EPL devices and 1 EP1 device from my RPi instance of home assistant to my VMWare version on x86 platform.

The two causing issues are kitchen and hallway.
The issues are as follows
1. When in ESPhome using the three dots for an EP1 device and 'Show API key' results in a message saying it cannot find the key in the configuration please check. As this issue is also present on the one EP1 that is functioning then I think this is a separte issue. all EPL devices show the API key specified in the configuration yaml file.
2. I have a repair notifgication for the two troublesome EP1 devices that reads "The ESPHome device Kitchen EP1 enabled transport encryption or changed the encryption key. Please enter the updated key. You can find it in the ESPHome Dashboard or in your device configuration." does not matter if I try and put the encryption key in it always says its wrong. deleting the devices and reinstalling with and without bluethooth made no difference the repair notification is never successful and always present.
3. flashing the hallway EP1 from the EP1 website tool results in a second device showing in ESPHome. It is like something is cached somewhere and refusing to update its contents.

Thanks for any guidance on a solution to rectify my EP1 dilema.
Paul

EverythingSmartHome, 2024-02-22:
Hello,

This isn't the devices playing up but rather the process used to migrate them.

The quickest and easiest way to resolve is probably to delete both from Home Assistant and ESPHome and then re-add them - if you keep the entity/device name the same then you won't need to change any automations

idkpmiller, 2024-02-22:
I believe that is what I have already tried but will try again. and make notes on my journey as I go if it is not as expected.

idkpmiller, 2024-02-22:
I have powered off the two devices, kitchen and hallway.
I have deleted them from esphome and home assistament.
I reboot the platform and when it comes up I see.
1. two 'discovered devices in the settings > Device section which are labelled hallway-EP1 and Kitchen-EP1.
2. two repair notifications asking me to enter the correct encryption key for the same two devices.
3. esphome ui is not showing the devices.

I think HA has cached information somewhere and not sure how to flush it out. As I said of the 7 devices I own these are the only two that behaved like this.
Paul

OziOwned, 2024-07-03:
Hi Paul, 

Did you find the solution for this? I am facing the same issue also.

Ben

chipmaestro, 2024-10-03:
Likewise; any help would be greatly appreciated.

Thanks,

Peter

EverythingSmartHome, 2024-10-03:
Can you provide details of the issue you are running into?

cmd-sam, 2024-11-12:
Same issue Herr