EverythingSmartHome/everything-presence-one #163: ESPHome Updates

jasonmadigan, 2024-03-04:
After performing an ESPHome update (after an HA prompt) for the EP1, the device goes into a strange/broken state:

- Sensors go unavailable
- I get a `Integration requires reconfiguration` notification
- Clicking `Reconfigure` for the device gives an error:
- - `The ESPHome device Everything Presence One 283acc enabled transport encryption or changed the encryption key. Please enter the updated key. You can find it in the ESPHome Dashboard or in your device configuration.`
- - I do not know what the key should be.

[image]

The only way to get things back and working again is to re-flash.

jasonmadigan, 2024-03-04:
Odd this isn't that graceful.

I found the key via ESPHome's UI. Edit the device, you'll find it in this block:

```yaml
api:
  encryption:
    key: xxx
```

EverythingSmartHome, 2024-03-04:
This is an ESPHome configuration thing, most likely when you have pressed adopt to the config in the ESPHome dashboard, it adds an encryption key (and tells you) which Home Assistant isn't aware of (if adopted after it was added to HA) - go into the ESPHome dashboard, edit the config and you should see the key listed

closed 2024-03-04 by EverythingSmartHome