EverythingSmartHome/everything-presence-one #164: MMWave sensitivity low but many false positives

erd82, 2024-03-25:
Hello,

Since weeks I have the problem that two of my Everything Presence One is too sensitive.
The mmwave sensor is showing this:
[image]

Instead the PIR sensor shows this:
[image]

I already configured the minimum sensitivity:
[image]

But it still detects movement every minute.

If I put the sensor upside down in the corner, the sensor should actually detect nothing. But it does.
Please don't recommend to check the plants or similar around. There is actually nothing around that could trigger the sensor.

The sensor(s) perfectly worked for months. But a few weeks / months (I don't exactly notice which ESPHome update caused it) the problem exists.

EverythingSmartHome, 2024-03-25:
Looks like you have on latency set to 0? I wouldn't recommend this, set to 0.5s first and observe the results which should drastically improve.

Btw, pointing it into a corner won't work, mmwave can see through walls and also the signal can "bounce" according to the distance set.

erd82, 2024-03-25:
Hi Lewis, you seem to have hit the nail on the head! Seems that I have not noticed the sensor "on latency" during the updates.

These pages in the documentation help me to understand:
- https://everythingsmarthome.github.io/everything-presence-one/tuning.html
- https://everythingsmarthome.github.io/everything-presence-one/tuning.html

Do you have any default values that you would recommend for MMWave sensor, like on latency = 0.5s and sensitivity = 4? I know this is finally dependent on the situation. Bit it's good to have suitable common start values. 

Not sure and I did not notice if your recommendations are already set as default after flashing the device. But if not, can you define such defaults?

EverythingSmartHome, 2024-03-25:
You can't really set defaults for the device since its done via serial and the SEN0395 already has defaults set that are stored on it's own chip, which is what the ESP32 defaults to. If we changed the defaults on the ESP32 to something else, it would mean that they were "out of sync" with the defaults stored on the SEN0395.

There isn't really ideal values other than what is described on the tuning page, you essentially adjust the values until you have no false positives and no false negatives either.

Hope that makes sense

closed 2024-03-28 by EverythingSmartHome

rsmanning, 2025-01-27:
I've tuned mine up to a full second of mm latency on, trigger sensitivity down to 4 and I'm still getting tons of false detection's.  I thought maybe mice in the garage could be triggering it but I set like six traps out there and nothing tripped or food taken in weeks now. I've watched cams out there, looked for water or ice coming off cars, nothing... Just does what it wants.  I'm back to requiring PIR and mm-wave before it turns on lights which is slower reaction.

EverythingSmartHome, 2025-01-27:
Can you share full settings? Remember that mmWaves can detect things through walls or objects which is where people mistakenly think nothing is moving, but its actually seeing something in the next room