EverythingSmartHome/everything-presence-one #165: Assembly Instructions do not match newest version of EP1

Kummallinen, 2024-04-03:
I just received a new EP1 & the mmWave sensor connection does not match the online assembly guide.

There is a single 8 pin wide header at the top of the board to connect the 5 pin sensor to. The silk screening on the board says to see the user guide for positioning but there is nothing in the guide I can see which matches.

I'm guessing it needs to be at the far right of the header to line up 5v on the board with VIN on the sensor?

EverythingSmartHome, 2024-04-03:
That is correct, I'm in the process of adding an update guide and firmware today. The new sensor goes to the far right of the 8 pin socket. You will need a new firmware too, let me build that now and add it.

Kummallinen, 2024-04-03:
Thank you for the quick response. I'll wait for the updated firmware before I go any further.

EverythingSmartHome, 2024-04-03:
Are you comfortable with flashing ESPHome firmware binaries yourself? If so, I can upload a bin for you to flash

Kummallinen, 2024-04-03:
I'm happy to give that a go if there are some steps to follow, but I'm also happy to wait if it is any more work for you to provide me the binaries early

EverythingSmartHome, 2024-04-03:
Firmware should be available to select here now: https://everythingsmarthome.github.io/everything-presence-one/updating.html

Kummallinen, 2024-04-03:
I've been able to install the firmware and connect to Home Assistant, thank you. Just to confirm there is supposed to be no Beta firmware for the new sensor, only "stable" so no target tracking & distance yet?

EverythingSmartHome, 2024-04-03:
Yes that's correct, I'm not sure if target tracking and distance will get added as a beta or not just yet

Kummallinen, 2024-04-03:
Great, just wanted to confirm as a selected "Beta" but it installed "Stable"

I've got mine all set up now but I'll leave this ticket open as the assembly instructions still need updating for the new board layout & sensor.

EverythingSmartHome, 2024-04-03:
Yeah that's just a placeholder incase we do a beta in the future but right now it installs the latest stable.

Sure no problem, will get updated soon

azielman, 2024-04-10:
Maybe just re-use the images from your shop? https://shop.everythingsmart.io/en-us/products/everything-presence-one-kit I was also concerned about the pin placement until I found the shop.

EverythingSmartHome, 2024-04-11:
Thanks - done now and completed in https://github.com/EverythingSmartHome/everything-presence-one/pull/172

Feedback welcome!

closed 2024-04-11 by EverythingSmartHome