EverythingSmartHome/everything-presence-one #171: Presence One for room based tracking.

Kelerion-UK, 2024-04-10:
I just received my new presence one's and they are working great out of the box - so thankyou!

I got them to replace my two espresence trackers to give more functionality. 

I was using the espresence trackers bluetooth for room based tracking (tracking my watch for example). I had this working and they detected when I switched from one room to another. I could distinguish them because espresence has a webui that allows you to set the name.

I can't figure out how to do the same with the presence one's? I have them individually names but I can't seem to find any bluetooth references in the documentation or interface. I can't even find any sensors.

Can you please advise on how I can start looking into getting this functionality working?

Thanks! and keep up the good work!

EverythingSmartHome, 2024-04-10:
Glad to hear you are enjoying them!

This question is about tracking bluetooth devices and is kinda out of the scope of the official firmware. However it is possible to do this with ESPHome, you will just need to compile your own firmware for that - unfortunately there is no way for me to provide pre-compiled firmware for this feature as you need to enter your own BLE mac addresses to track during compiling.

What you are looking for is the bluetooth tracking feature of ESPHome: https://esphome.io/components/esp32_ble_tracker.html

Please be aware that you may run into connectivity issues when having bluetooth tracking enabled at the same time due to increased radio usage, some people do and some don't.

Kelerion-UK, 2024-04-10:
> What you are looking for is the bluetooth tracking feature of ESPHome: https://esphome.io/components/esp32_ble_tracker.html
> 

Excellent.. I'll look into that.. seems fairly simple.
If I have any success I'll report it back here.

Thanks for the quick response!

Kelerion-UK, 2024-04-10:
> What you are looking for is the bluetooth tracking feature of ESPHome: https://esphome.io/components/esp32_ble_tracker.html
> 
Actually... this was suprising easy.

I actually used this instead: https://esphome.io/components/sensor/ble_rssi
I linked it using my watches iBeacon (its mac address changes randomly) and it returns the RSSI value.

I'll add that to both EP1's and a simple automation to decide which RSSI value is closer can set it's room location.

I'm only tracking one device at the moment and I can report (quite happily) that I didn't seem to break anything and the rest of the EP1 seems unaffected. It's connectivity and responsiveness doesn't seem to have been affected so far!

Thanks for the help! 

C.

closed 2024-04-10 by Kelerion-UK