EverythingSmartHome/everything-presence-one #177: Beta firmware for SEN0609

jorismathijssen, 2024-04-19:
Hello,

I attempted to install the beta firmware for my SEN0609 version of the EP1. Could it be that there isn't a beta version available for this mmWave chip? When I follow the guide, it doesn't seem to flash anything. Upon checking, it appears that the only beta version listed is for the SEN0395.

I may be missing something. Can anyone assist me in getting back on track?

Thank you.

EverythingSmartHome, 2024-04-19:
There is no Beta available for the SEN0609 at the moment, it is just there as a placeholder incase one is required in the future.

jorismathijssen, 2024-04-19:
Thank you for clarifying what I already expected! 

I hope that some kind of target tracking or zone functionality would be possible in a future (beta) release. I would be happy to assist with testing.

closed 2024-04-19 by jorismathijssen