EverythingSmartHome/everything-presence-one #178: Meaning of "mmWave Range Reduced" and "mmWave Sustain Sensitivity" for SEN0609 in the documentation

Mincka, 2024-04-20:
Hello,

I did not find anywhere the meaning of the settings called `mmWave Range Reduced` and `mmWave Sustain Sensitivity` that were introduced with the support of SEN0609.

The default value is "0.8 m" for `mmWave Range Reduced` and "7" for `mmWave Sustain Sensitivity`.

I could guess that "Sustain Sensitivity" is the capability to keep tracking an object more reliably, but for the other setting, I have no idea.

Would it be possible to update the tuning guide and the documentation with these new settings?

Thank you.

pablogiganti, 2024-06-22:
Hi, I would also appreciate a bit of explanation please

EverythingSmartHome, 2024-07-19:
Added now here: https://everythingsmarthome.github.io/everything-presence-one/Home%20Assistant/home-assistant-entities.html#distance-and-sensitivity-sen0609

Hope that helps!

closed 2024-07-19 by EverythingSmartHome