EverythingSmartHome/everything-presence-one #182: Enabling BLE Kills the device

Veldkornet, 2024-05-03:
If I switch from everything-presence-one.yaml to everything-presence-one-ble.yaml (and re-flash via USB.... or any other method), the device fails to boot. 
When plugged in via USB, these were the latest logs I can get (note that even the brwoser hangs at this point:

[image]

EverythingSmartHome, 2024-05-04:
What happens if you flash directly from the user guide instead?

Veldkornet, 2024-05-04:
> What happens if you flash directly from the user guide instead?

What do you mean from the user guide? Can you point me to it?

Currently, I just opened ESPHome, edited the device settings "package" section to use ble instead of the normal one.

EverythingSmartHome, 2024-05-04:
Yes that's right

Veldkornet, 2024-05-05:
> Yes that's right

Yes, it seems to work now! Not sure why though, I compared the two configs in ESP home and they look the same...

closed 2024-05-05 by Veldkornet

Kummallinen, 2024-06-22:
I had the same issue today, the trick seems to be to do a "Clean Build Files" after changing the packages config to point to the ble one, but before building & installing it