EverythingSmartHome/everything-presence-one #183: EP1 triggering once every hour

Beachviolence, 2024-05-06:
Hi!

First of all, super happy with the product! But...

I have an issue with my EP1 triggering once every hour. As you can see from the logs, it is the PIR sensor that is triggering, turning my livingroom lights on every night hourly. I am on the latest release of ESPhome, Home Assistant and on the EP1 itself. Could not manage to find any similar cases. Does anyone have any idea what the cause may be? 

[image]

EverythingSmartHome, 2024-05-06:
Hello, glad to hear it!

Does the PIR face any heat source like a radiator, AC or underfloor heating? Does it correspond with any heating/cooling device coming on perhaps?

Beachviolence, 2024-05-07:
> Hello, glad to hear it!
> 
> Does the PIR face any heat source like a radiator, AC or underfloor heating? Does it correspond with any heating/cooling device coming on perhaps?

I tried changing the direction away from the radiator and any other potential heat sources, but the issue persisted. Now i tried moving it to another location in the livingroom, and for now it seems like the issue went away! (after one night of testing) Probably not the EP1 then, and maybe some automated device in my house doing something weird. Thanks for your help!

closed 2024-05-07 by Beachviolence