EverythingSmartHome/everything-presence-one #184: Everything Presence One Home ssistant

swilbaby, 2024-05-25:
I have set up the new Presence One Kit, but Home assistant is not showing any zone controls on the configuration. I have tried, numerous times, to re flash with the beta firmware 1.2.3b, but it only installs the Stable firmware, which appears to have no zone configurations. Can anyone help?

Home Assistant is fully updated.

mcsdodo, 2024-06-28:
EP One mmWave sensor does not support this. Only EP Lite.

Katana-Ha, 2024-07-02:
does the other sensor, the SEN0395 support it. Seems to if I flash that firmware, but the the mmWave doesn't actually work because I have the wrong sensor flashed

Xyz00777, 2024-11-03:
isnt there any way... because these was the main point in buying it a few months ago... i hope it will for the price ;(

Xyz00777, 2024-11-29:
Wouldn't it be a possibility to sell it with optional the other sensor for multi zoning so a customer can chose which one he want and provide the software for both? Because that would be really aweseome! :/