EverythingSmartHome/everything-presence-one #185: EP1 PIR sensitivity?

codemunkie15, 2024-05-26:
Hi

I recently bought an EP1 hoping that I could use the PIR to turn lights on and then keep them on with the mmWave sensor as I have cats and I don't want them triggering my automations with the mmwave.

Unfortunately they are still setting off the PIR. Is there any way to reduce the PIR sensitivity please? Usually PIRs have this functionality but I couldn't find anything in the docs.

Many thanks

EverythingSmartHome, 2024-05-26:
Hello,

There is no PIR sensitivity, however we did make the PIR removable and swappable so that you can install another PIR from the Panasonic industrial PIR range, there are others with different sensitivity levels that may suit your needs better

codemunkie15, 2024-05-26:
> Hello,
> 
> There is no PIR sensitivity, however we did make the PIR removable and swappable so that you can install another PIR from the Panasonic industrial PIR range, there are others with different sensitivity levels that may suit your needs better

Thanks for the quick reply. Any idea which models would have lower sensitivity please? I've been trying to find EKMC models but I can't find much detail on the Panasonic website.

EverythingSmartHome, 2024-07-19:
Sorry I don't have any recommendations for you as I've not tried the others in the range. I'd recommend taking a look at the EKMC catalogue if you haven't seen it already: https://api.pim.na.industrial.panasonic.com/file_stream/main/fileversion/251032

There is a product matric on Page 8 showing models and differences. Again I haven't tested

closed 2024-07-19 by EverythingSmartHome