EverythingSmartHome/everything-presence-one #187: framework-espidf does not contain a CMakeLists.txt file.

bekovan, 2024-06-01:
During update of the everything-presence-one I get in the reading CMake :

- Project sdkconfig file /data/build/everything-presence-one-21508c/sdkconfig.everything-presence-one-21508c
-- Configuring incomplete, errors occurred!
See also "/data/build/everything-presence-one-21508c/.pioenvs/everything-presence-one-21508c/CMakeFiles/CMakeOutput.log".

fatal: not a git repository (or any parent up to mount point /)
Stopping at filesystem boundary (GIT_DISCOVERY_ACROSS_FILESYSTEM not set).
CMake Error at /data/cache/platformio/packages/framework-espidf/tools/cmake/build.cmake:546 (add_subdirectory):
  The source directory

    /data/cache/platformio/packages/framework-espidf

  does not contain a CMakeLists.txt file.
Call Stack (most recent call first):
  /data/cache/platformio/packages/framework-espidf/tools/cmake/project.cmake:476 (idf_build_process)
  CMakeLists.txt:3 (project)

Runnning now ESPhome 2024.5.4 (
Not sure with what version this issue started.

EverythingSmartHome, 2024-06-03:
Are you sure this isn't a cache thing? Try and press the "Clean build files" on the config, then try to install again

bekovan, 2024-06-03:
Thank you for the reaction
This was the first thing I tried, & I just repeated the test.
No difference, error stays

bekovan, 2024-06-26:
still the same error, any progress ?

EverythingSmartHome, 2024-06-26:
This isn't an EP1 issue sorry, it would need to be reported upstream to ESPHome. 

I have a feeling it's your local ESPHome addon but just a guess

closed 2024-06-26 by EverythingSmartHome