EverythingSmartHome/everything-presence-one #190: OTA Platforms - the easy way

neildsb, 2024-06-20:
Hi, is there and easy way to update with the following on ESPHome 2024.6.1?

https://next.esphome.io/changelog/2024.6.0.html#ota-platforms

neildsb, 2024-06-20:
** update **

fixed by adding 

ota:
  platform: esphome

and update again works...leaving 'open' for a while for others in the same boat

Dinth, 2024-06-20:
The workaround is super easy but it should be fixed upstream asap, as many people are just including code from github:
```
packages:
  EverythingSmartTechnology.Everything_Presence_Lite: github://everythingsmarthome/everything-presence-lite/everything-presence-lite-ha.yaml@main
```

EverythingSmartHome, 2024-06-20:
Yes I will fix this morning

EverythingSmartHome, 2024-06-20:
Fixed in: https://github.com/EverythingSmartHome/everything-presence-one/pull/189

closed 2024-06-20 by EverythingSmartHome