EverythingSmartHome/everything-presence-one #193: Temperature changing by 1c every 30 min

Ralphy140, 2024-06-27:
I have 2 EP1 in 2 diffrent locations of my house, as well as a bunch of other ESP based temp sensors. This issue is not related to the calibration or drift but that the sensors seems to be doing something every 30 min that makes the temp spike. I am not sure how to start debugging this issue and keen to know if anyone else has seen anything like this? I have included 3 sensors in the graph, the 2 EP1s and a ESP based solution that uses the BME680 as the temp sensor. 

[image]

EverythingSmartHome, 2024-07-19:
Can you let me know which firmware you are using?

Ralphy140, 2024-07-19:
It's on version 1.1.3

EverythingSmartHome, 2024-07-19:
That's quite old, can you update and see if anything changes?

I meant, can you let me know which firmware you are using as there is a few different ones (bluetooth, non bluetooth, SEN0609, SEN0395, beta, stable)

Ralphy140, 2024-07-19:
Ahh yes sure, this is one of the very first release devices and I moved over to manual firmware to try and get Bluetooth working before it was included due to issue with improv I think. but it caused wifi dropouts so removed the feature but never went back to the git firmware management. I will reflash it and let you know if any improvement

Ralphy140, 2024-07-22:
Ok have just updated to the latest firmware, Hardware: 1.2.4 Its running the SEN0395, no BLE, stable version. I am also using ESPHome 2024.7.0.  I will let you know if the behaviour continues

Ralphy140, 2024-07-25:
Ok seems like after updating to the latest firmware the issue is resolved though i find it interesting that the Temp seems to runs hotter in general after the update. 
[image]