EverythingSmartHome/everything-presence-one #199: mmwave not working?

stoupaspan, 2024-07-17:
Hello there! I just purchased the ep1 but I think that the mm wave is not working! I have made an automation where a smart light turns on when occupancy is detected, so every once and a while the light turns on even when I am next to the ep1. By looking at the logbook, I can see that only the PIR sensor detects motion, note the mmwave. Moreover, when I try to turn on the mmWave LED through the configuration page in HA, the led never turns on.
I am attaching the configuration screenshot, in case I have confirured something wrong

Thank you!
[image]

EverythingSmartHome, 2024-07-19:
Hello, sorry for the delay, I didn't notice this. Can you please send a photo of the mmWave sensor plugged into the board so I can check for any issues there?

Also, do you ever see any LEDs on the board at any point when plugging in?

stoupaspan, 2024-07-20:
Hi and thanks for the response. I am attaching the images, as you requested.
Concerning the LED on the board, I only see a red light on the left of the board when I plug it in. No LED lights turn on on thw mm wave.

Thanks!! 
[image]
[image]
[image]

EverythingSmartHome, 2024-07-21:
Hello, you have installed the mmwave in the wrong position, plugging it in like this can damage the sensor..

Please refer to the user guide which is very clear about the position to plug in to:  https://everythingsmarthome.github.io/everything-presence-one/assembly.html#single-8-pin-slot

DaveP317, 2024-07-26:
I'm having a similar issue with mmwave not working, but my daughterboard is plugged in correctly. I have the EP1 board with 6 pin headers, and the mmwave board with 6 pins. The mmwave LED does light up, but only the PIR sensor reports detection.

I have 4 EP1 boards, and they all behave the same.

Do you have any tips or tricks to debug this?

Cheers!

[image]
[image]
[image]
[image]

BradleyFord, 2024-07-26:
> ID: <EverythingSmartHome/everything-presence-one/issues/199/2252739070@
> github.com>

This happens to be every single time there is a power outage.
Current I manually turn off the mmwave, wait, and then turn back on. Rinse
and repeat for each EP1

DaveP317, 2024-07-26:
Interesting idea...I just tried that, but it didn't work for me.

DaveP317, 2024-07-26:
On a lark, I reflashed one of the units from the latest stable fw using the web flasher utility, and the mmwave sensor started working. I confirmed that it still wasn't working prior to flashing it.

Then I went to grab the second unit to reflash it, and noticed that it is now working, too.

This is a head scratcher...anyway, I'll keep an eye on things. For now, it seems to be working properly.