EverythingSmartHome/everything-presence-one #200: Breaking change after 26.4

ADeadPixel, 2024-07-17:
Whole device became unavailable after updating to latest ESPHome.
Current info from EP1:
ESPHome Firmware: 2024.7.0
Hardware Rev: 1.4

I get the following logs when attempting to connect via wifi
```txt
INFO ESPHome 2024.7.0
INFO Reading configuration /config/esphome/ep1-0.yaml...
WARNING GPIO15 is a strapping PIN and should only be used for I/O with care.
Attaching external pullup/down resistors to strapping pins can cause unexpected failures.
See https://esphome.io/guides/faq.html#why-am-i-getting-a-warning-about-strapping-pins
INFO Starting log output from 192.168.20.213 using esphome API
WARNING Can't connect to ESPHome API for ep1-0 @ 192.168.20.213: Error connecting to [AddrInfo(family=<AddressFamily.AF_INET: 2>, type=<SocketKind.SOCK_STREAM: 1>, proto=6, sockaddr=IPv4Sockaddr(address='192.168.20.213', port=6053))]: [Errno 111] Connect call failed ('192.168.20.213', 6053) (SocketAPIError)
INFO Trying to connect to ep1-0 @ 192.168.20.213 in the background
INFO Successfully connected to ep1-0 @ 192.168.20.213 in 0.121s
WARNING ep1-0 @ 192.168.20.213: Connection error occurred: ep1-0 @ 192.168.20.213: Connection lost
INFO Successfully connected to ep1-0 @ 192.168.20.213 in 0.063s
WARNING ep1-0 @ 192.168.20.213: Connection error occurred: ep1-0 @ 192.168.20.213: Connection lost
INFO Successfully connected to ep1-0 @ 192.168.20.213 in 0.194s
WARNING ep1-0 @ 192.168.20.213: Connection error occurred: ep1-0 @ 192.168.20.213: Connection lost
INFO Successfully connected to ep1-0 @ 192.168.20.213 in 0.153s
WARNING ep1-0 @ 192.168.20.213: Connection error occurred: ep1-0 @ 192.168.20.213: Connection lost
```

and plugging in directly I get the following

```txt
[14:48:57]h�����F�HF±�[E][esp-idf:000][wifi]: I (7745) wifi:
[E][esp-idf:000][wifi]: <ba-add>idx:1 (ifx:0, da:21:f9:83:3c:cf), tid:0, ssn:0, winSize:64
[E][esp-idf:000][wifi]: 

E (11600) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
[14:48:57]E (11600) task_wdt:  - loopTask (CPU 1)
[14:48:57]E (11600) task_wdt: Tasks currently running:
[14:48:57]E (11600) task_wdt: CPU 0: IDLE0
[14:48:57]E (11600) task_wdt: CPU 1: IDLE1
[14:48:57]E (11600) task_wdt: Aborting.
[14:48:57]
[14:48:57]abort() was called at PC 0x400f99c8 on core 0
[14:48:57]
[14:48:57]
[14:48:57]Backtrace: 0x40082966:0x3ffb0860 0x400896e5:0x3ffb0880 0x4008f89a:0x3ffb08a0 0x400f99c8:0x3ffb0910 0x400832e9:0x3ffb0930 0x4016ce6b:0x3ffbad90 0x400fa119:0x3ffbadb0 0x4008aabc:0x3ffbadd0
[14:48:57]
[14:48:57]
[14:48:57]
[14:48:57]
[14:48:57]ELF file SHA256: f98735db8beff4fd
[14:48:57]
[14:48:57]Rebooting...
[14:48:57]ets Jul 29 2019 12:21:46
[14:48:57]
[14:48:57]rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
[14:48:57]configsip: 0, SPIWP:0xee
[14:48:57]clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
[14:48:57]mode:DIO, clock div:2
[14:48:57]load:0x3fff0030,len:6652
[14:48:57]ho 0 tail 12 room 4
[14:48:57]load:0x40078000,len:15052
[14:48:57]load:0x40080400,len:3836
[14:48:57]entry 0x4008069c
[14:48:57]I (29) boot: ESP-IDF 4.4.7 2nd stage bootloader
[14:48:57]I (29) boot: compile time 13:03:59
[14:48:57]I (29) boot: Multicore bootloader
[14:48:57]I (32) boot: chip revision: v3.1
[14:48:57]I (36) boot.esp32: SPI Speed      : 40MHz
[14:48:57]I (41) boot.esp32: SPI Mode       : DIO
[14:48:57]I (45) boot.esp32: SPI Flash Size : 4MB
[14:48:57]I (50) boot: Enabling RNG early entropy source...
[14:48:57]I (55) boot: Partition Table:
[... 385 lines cut ...]
[E][esp-idf:000][wifi]: 

[E][esp-idf:000][wifi]: I (4982) wifi:
[E][esp-idf:000][wifi]: connected with RPFP - IoT, aid = 1, channel 6, 40D, bssid = da:21:f9:83:3c:cf
[E][esp-idf:000][wifi]: 

[E][esp-idf:000][wifi]: I (4983) wifi:
[E][esp-idf:000][wifi]: security: WPA2-PSK, phy: bgn, rssi: -34
[E][esp-idf:000][wifi]: 

[E][esp-idf:000][wifi]: I (4998) wifi:
[E][esp-idf:000][wifi]: pm start, type: 1

[E][esp-idf:000][wifi]: 

[E][esp-idf:000][wifi]: I (5010) wifi:
[E][esp-idf:000][wifi]: AP's beacon interval = 102400 us, DTIM period = 1
[E][esp-idf:000][wifi]: 

[E][esp-idf:000][wifi]: I (5055) wifi:
[E][esp-idf:000][wifi]: <ba-add>idx:0 (ifx:0, da:21:f9:83:3c:cf), tid:6, ssn:2, winSize:64
[E][esp-idf:000][wifi]: 

[E][esp-idf:000][sys_evt]: I (6013) esp_netif_handlers: sta ip: 192.168.20.213, mask: 255.255.255.0, gw: 192.168.20.1
```

thetonto, 2024-07-17:
Confirmed.  Device appears back as default and also back to older version once you adopt it.

mannerisms, 2024-07-18:
Also same here, tried re-flashing over usb but nothing seems to help, all my ESP one devices have gone dead and become unavailable to homeassistant

EverythingSmartHome, 2024-07-19:
Can you confirm if it works flashing from the user guide: https://everythingsmarthome.github.io/everything-presence-one/updating.html

This version is on 2024.6.6

ADeadPixel, 2024-07-19:
> Can you confirm if it works flashing from the user guide: https://everythingsmarthome.github.io/everything-presence-one/updating.html
> 
> This version is on 2024.6.6

That is where I attempted to re-flash from.

EverythingSmartHome, 2024-07-19:
Just so we can rule it out, is it possible for you to connect the WiFi to your regular SSID with internet access?

ADeadPixel, 2024-07-20:
Sure thing, I swapped it over to the primary network but still nothing in HA. For context the EP1 can still communicate via WiFi (I can still upload wirelessly) but nothing is in the log and nothing is in HA.
```txt
INFO ESPHome 2024.7.0
INFO Reading configuration /config/esphome/ep1-0.yaml...
WARNING GPIO15 is a strapping PIN and should only be used for I/O with care.
Attaching external pullup/down resistors to strapping pins can cause unexpected failures.
See https://esphome.io/guides/faq.html#why-am-i-getting-a-warning-about-strapping-pins
INFO Starting log output from 192.168.1.166 using esphome API
INFO Successfully connected to ep1-0 @ 192.168.1.166 in 0.040s
INFO Successful handshake with ep1-0 @ 192.168.1.166 in 0.087s
```

Here is how it looks in HA. Notice how it shows old information even though the logs show it was ESPHome version 2024.7.0 above.
[image]

TheHackmeister, 2024-07-20:
I've been having the same issue end of this week. Ended up rolling back the VM running ESPHome to try and flash with  2024.6.4 and 2024.6.6 but encountered the same. Before that I had a backup of 2024.6.2 but that version of ESPHome was incompatible with the EP package. Figured fine, I'll pin the package to an earlier version. Turns out, the issues is likely in package version v1.2.4 as  v1.2.3 works fine on ESPHome 2024.7.0. So this sort of thing at least brought my sensors back into a usable state: 
```
packages:
  athom.epo-ble: github://everythingsmarthome/presence-one/everything-presence-one-ble.yaml@v1.2.3
```

Hope it helps!

ADeadPixel, 2024-07-20:
I can confirm updating to the package listed did address the problem. So something is wrong with the v1.2.4 release and >2024.6.3 ESPHome.

[image]

Here is my config for anyone who needs a reference. If you do revert to 1.2.3 make sure to add in the OTA information.

```yaml
substitutions:
  name: ep1-0
  friendly_name: Robert's Office EP1

packages:
  # Everything_Smart_Technology.Everything_Presence_One: github://everythingsmarthome/everything-presence-one/everything-presence-one.yaml@1.2.3
  testing: github://everythingsmarthome/presence-one/everything-presence-one-ble.yaml@v1.2.3

esphome:
  name: ${name}
  name_add_mac_suffix: false
  friendly_name: ${friendly_name}

ota:
  platform: esphome

api:
  encryption:
    key: !secret encryption_key
    
wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
```

EverythingSmartHome, 2024-07-20:
Thanks for the details. The only real difference in those versions is the new update mechanism which does make me wonder if it can't access the update file and reboots constantly trying to keep checking for updates.

SasStu, 2024-07-22:
Hi I can confirm this. Just installed the EP One in my IoT network and its working, back to my NoT network and we are back to red leds...

mannerisms, 2024-07-22:
Same here, after following the discussion here, the NoT network was the issue.

SasStu, 2024-07-22:
OK I just added

update: !remove

as a workaround to my configuration and now the device is working again in my NoT network.

EverythingSmartHome, 2024-07-23:
This should be resolved in https://github.com/EverythingSmartHome/everything-presence-one/pull/206, please update to version 1.0.7. A fix is coming in ESPHome 2024.7.2 but this workaround does the same thing for now.

closed 2024-07-23 by EverythingSmartHome

taintedkernel, 2024-07-23:
@EverythingSmartHome Thank you, that update solved it! I spent a few days trying to troubleshoot what was wrong with my new device on my NoT HASS network.