EverythingSmartHome/everything-presence-one #201: Unable to Update to 2024.7.0 Failed Config Certification Verification

jwpine, 2024-07-19:
I don't know if this is related to the breaking change that is mentioned in a different issue. However, I cannot update to 2024.7.0. If I try to update I get the following error. 

> INFO ESPHome 2024.7.0
> INFO Reading configuration /config/esphome/everything-presence-one-76384c.yaml...
> Failed config
> 
> http_request: [source /data/packages/405587a8/common/everything-presence-one-base.yaml:30]
>   
>   ESPHome supports certificate verification only via ESP-IDF. Set 'verify_ssl: false' to skip certificate validation and allow less secure HTTPS connections.
>   {}

my yaml file is below for reference:

`substitutions:
  name: everything-presence-one-76384c
  friendly_name: Everything Presence One
packages:
  Everything_Smart_Technology.Everything_Presence_One: github://everythingsmarthome/everything-presence-one/everything-presence-one-sen0609-ble.yaml@main
esphome:
  name: ${name}
  name_add_mac_suffix: false
  friendly_name: ${friendly_name}
api:
  encryption:
    key: redacted
esp32:
  board: esp32dev
  framework:
    type: arduino
#ota:
#  - platform: esphome

wifi:
  ssid: redacted
  password: redacted
`

EverythingSmartHome, 2024-07-19:
Was there a particular reason you added the extra lines to change the framework to Arduino instead of the default?

jwpine, 2024-07-19:
> Was there a particular reason you added the extra lines to change the framework to Arduino instead of the default?

Without it I get a completely different error.

INFO ESPHome 2024.7.0
INFO Reading configuration /config/esphome/everything-presence-one-76384c.yaml...
INFO Generating C++ source...
INFO Compiling app...
Processing everything-presence-one-76384c (board: esp32dev; framework: espidf; platform: platformio/espressif32@5.4.0)
--------------------------------------------------------------------------------
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
 - framework-espidf @ 3.40407.0 (4.4.7) 
 - tool-cmake @ 3.16.9 
 - tool-ninja @ 1.10.2 
 - toolchain-esp32ulp @ 2.35.0-20220830 
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch5
Reading CMake configuration...
-- The C compiler identification is unknown
-- The CXX compiler identification is unknown
-- Configuring incomplete, errors occurred!
See also "/data/build/everything-presence-one-76384c/.pioenvs/everything-presence-one-76384c/CMakeFiles/CMakeOutput.log".
See also "/data/build/everything-presence-one-76384c/.pioenvs/everything-presence-one-76384c/CMakeFiles/CMakeError.log".

CMake Error at CMakeLists.txt:2 (include):
  include could not find load file:

    /data/cache/platformio/packages/framework-espidf/tools/cmake/project.cmake

CMake Error at CMakeLists.txt:3 (project):
  No CMAKE_C_COMPILER could be found.

  Tell CMake where to find the compiler by setting either the environment
  variable "CC" or the CMake cache entry CMAKE_C_COMPILER to the full path to
  the compiler, or to the compiler name if it is in the PATH.

CMake Error at CMakeLists.txt:3 (project):
  No CMAKE_CXX_COMPILER could be found.

  Tell CMake where to find the compiler by setting either the environment
  variable "CXX" or the CMake cache entry CMAKE_CXX_COMPILER to the full path
  to the compiler, or to the compiler name if it is in the PATH.

========================== [FAILED] Took 9.49 seconds ==========================

EverythingSmartHome, 2024-07-19:
Did you try pressing the clean build files button on the config and trying again?

jwpine, 2024-07-19:
> Did you try pressing the clean build files button on the config and trying again?

Same error.
 Is there a default yaml I can use as a template?

EverythingSmartHome, 2024-07-19:
Yes if you delete that config and restart the ESPHome addon, it should auto detect the config again and let you adopt it. Take note of your encryption key first before deleting and then replace it once re-adopted

jwpine, 2024-07-19:
Still getting the same compiler errors.

EverythingSmartHome, 2024-07-19:
This is an issue with your local ESPHome add-on as the builds are still compiling fine on GitHub. The only other thing I could suggest would be to uninstall and re-install your ESPHome add-on. This won't delete your configs, though might be worth deleting the EP1 config before uninstall and adopting after (copy encryption key first).

If that doesn't work, then next would be to try and uninstall the add-on, then deleting the esphome data folder (this will remove all your configs) under /config/esphome

jwpine, 2024-07-30:
This was a wider esphome issue resolved in a subsequent update. Posting this for documentation's sake.

closed 2024-07-30 by jwpine