EverythingSmartHome/everything-presence-one #202: Humidity sensor accuracy

idkpmiller, 2024-07-19:
Hi,
IAs its winter down in this hemi-sphere, I have been focusing on humidity and getting the climate controls set up correctly.
I have noticed a patteren, the EP1 device I have seemn to be measuring low. My current guess is around 20% too low.
The EPL seem to align with most other sensors. The condensation on the windows in the mornings suggests that the EP1s are the ones that are incorrectly measuring the relative humidity.
Is there any statement on the accuracy expected from the EP1 for the humidity measurements? 
Can I assue they have a humidity sensor and not calculation it off the temperature?
I will begin my deep dive into the measurements to see what gives and if they are usable to measure humidity for climate control.
I would really like to get a view on them though from you as if I am wasting my time checking them out then I wil need to replace them.

Thanks
Paul

EverythingSmartHome, 2024-07-19:
Hello! The EP1 (I assume it's the EP1 you mean, but you mentioned EPL also however that doesn't have humidity so assume EP1) uses the SHTC3 Temp and Humidity sensor from Sensiron. I believe (would need to check) from memory it does a calculation from temperature, so not really designed for high accuracy situations which it sounds like you are aiming for where a dedicated humidity sensor would be better.  There are spare GPIO pins on the EP1 if you want to add one directly.

idkpmiller, 2024-07-20:
Thanks for the response.
I can confirm that I am indeed talking about the EP1.
I have just read through the datasheet of the SHTC3.
According to its data sheet it is quite an accurate mapping in the household temps I am talking about it shoud be 2-2.5% + or - of the correct humidity. If this is the case then it is way accurate enough. I think there is something else going on here.
Before I conduct a humidty calibration test, I think I will leave the case off for 24 hours and see if that resolves the issue.
I will let you know my findings.
Thanks
Paul

idkpmiller, 2024-07-20:
Taking the front of the case off made a small improvement in the correct direction. But the temperature was still measureing too high abouth 16.5% too high. so this will translate to a too low humidity reading.
So the real issue here is the temperature sensor is not measuring accurately. As the datashet for the sensor shows a 0.2% +- accuracy and all three of my sensors are running high, I am leaning towards the PCB layout.
Has the EP1 been calibrated with the case on to be accurate?
If so, at what temperature?

I have for the time being used the offset but that is a short term solution.

I would be interested to know if others have noticed the sqame issue.

Paul

idkpmiller, 2024-07-30:
I have been monitoring the EP1 temperature as that is the source of the humidity calculation.
To get it matching a little closer to the temperature recoded by other devices such as the one I am using as the baseline (I removed it temporarily from the bathroom).

The base devices is an Aquara Temperature Humidity Pressure sensor that reports via zigbee as a lumi.weather.
I have found it to be a reasonably accurate device over the years and prefer it over other types I have tried.

[image]

In the graph I have tried to get the EP1 to be as resoably close to the aquare as possible. to achieve this I placed an offset in the temperature of -7.7C. This is a way lartger number than I would expect to need for a device that claims -/+ 0.2C

Even though I have them reasonable close I still needed an additional  whopping humidity offset of 22 to get anything of close to a reasonable value.  

There is something definitely not right here. I ask again have you calibrated or even measured these after pcb build?
 
Regards
Paul

EverythingSmartHome, 2024-07-30:
Yes of course we measured them. We've tested everything about the PCBs extensively.

This offset is mentioned in the description off the product and has been ever since it launched 2 years ago. It specifically mentions if you need a very accurate measurement then to use an external sensor:

[image]

Measuring between different sensors may not give the same results, due to variance in sensors. This article demonstrates it quite well and shows the variance in different sensors, you can see from the results graph a 4 degree difference between different sensors. Unfortunately the SHTC3 isn't here but illustrates well:

https://randomnerdtutorials.com/dht11-vs-dht22-vs-lm35-vs-ds18b20-vs-bme280-vs-bmp180/

So measuring against an unknown sensor isn't an apples to apples comparison IMO.

But yes we do acknowledge that for accurate readings you will want to use an external sensor, that's why it's in the description and always had been.