EverythingSmartHome/everything-presence-one #207: No connection after upgrade to 1.2.5 stable

atle-dahlman, 2024-07-23:
Just upgraded Everything Presence One to 1.2.5 stable.  Now it won't connect to Home Assistant.

Log shows:

`INFO Reading configuration /config/esphome/everything-presence-one-xxxxxx.yaml...
INFO Starting log output from 192.168.1.48 using esphome API
INFO Successfully connected to everything-presence-one-xxxxxx @ 192.168.1.48 in 0.107s
WARNING everything-presence-one-xxxxxx @ 192.168.1.48: Connection error occurred: everything-presence-one-xxxxxx @ 192.168.1.48: The connection dropped immediately after encrypted hello; Try enabling encryption on the device or turning off encryption on the client (ESPHome Logs 2024.7.1).`

dannytsang, 2024-07-23:
I'm also noticing the same but unable to get any logs.

EverythingSmartHome, 2024-07-23:
This isn't an issue with the update, I'm guessing you upgraded via the entity in Home Assistant? The reason this occurs is because you have an encryption key configured in the ESPHome add-on, which gets overwritten when you update directly from GitHub. Home Assistant is then trying to communicate with the device using the encryption key which it's no longer configured for. Hopefully that makes sense.

You can either flash using the ESPHome add-on again over wireless, or delete from HA and re-add it without the encryption key. Home Assistant may even prompt you to reconfigure without the key. If you do this then make sure to remove the key from your ESPHome addon

BenHolmes, 2024-07-23:
Same here! 

I removed my ota encryption line then installed yaml w/o that line. 
Once that was installed I added back the encryption line into my yaml config and installed. 

This reverted the version and got all my sensors back to how they were in home assistant before attempting the update.

BenHolmes, 2024-07-23:
> This isn't an issue with the update, I'm guessing you upgraded via the entity in Home Assistant? The reason this occurs is because you have an encryption key configured in the ESPHome add-on, which [...]

Is there no way to update without re-flashing and losing the existing device/ device data in home assistant?

BenHolmes, 2024-07-23:
The ESPHome add-on does not have any updates available at this time. 
ESPHome add-on dash 
[image]
vs Home assistant update prompt
[image]
[image]

EverythingSmartHome, 2024-07-23:
Yes but if your config is standard, installing it like this will pull down the latest info from the GitHub anyways and compile it on your local system - the end result is the same, it just takes longer to do

atle-dahlman, 2024-07-23:
Tried to reflash without the encryption key.  Still no go:
Here is my YAML code.

`substitutions:
  name: everything-presence-one-xxxxxx
  friendly_name: EPO-gangen
packages:
  Everything_Smart_Technology.Everything_Presence_One: github://everythingsmarthome/everything-presence-one/everything-presence-one-sen0609.yaml@main
esphome:
  name: ${name}
  name_add_mac_suffix: false
  friendly_name: ${friendly_name}

api:

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
`

EverythingSmartHome, 2024-07-23:
Can you provide logs of the attempt?

BenHolmes, 2024-07-23:
Sorry! I might be mis-understanding but the issue seems to be happening by installing via the ESPHome add-on as well as via Home assistant update entity.

Grabbing logs.

BenHolmes, 2024-07-23:
> Tried to reflash without the encryption key. Still no go: Here is my YAML code.
>
> `substitutions: name: everything-presence-one-xxxxxx friendly_name: EPO-gangen packages: [...]

I removed API completely in order to get it back online before installing the current version again.

atle-dahlman, 2024-07-23:
Tried again.  Here are the log:

`INFO ESPHome 2024.7.1
INFO Reading configuration /config/esphome/everything-presence-one-xxxxxx.yaml...
INFO Generating C++ source...
INFO Compiling app...
Processing everything-presence-one-xxxxxx (board: esp32dev; framework: espidf; platform: platformio/espressif32@5.4.0)
--------------------------------------------------------------------------------
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
 - framework-espidf @ 3.40407.240606 (4.4.7) 
 - tool-cmake @ 3.16.4 
 - tool-ninja @ 1.7.1 
 - toolchain-esp32ulp @ 2.35.0-20220830 
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch5
Reading CMake configuration...
Dependency Graph
|-- ArduinoJson @ 6.18.5
|-- Improv @ 1.2.3
RAM:   [=         ]   9.7% (used 31896 bytes from 327680 bytes)
Flash: [=====     ]  51.6% (used 946201 bytes from 1835008 bytes)
========================= [SUCCESS] Took 9.14 seconds =========================
INFO Successfully compiled program.
INFO Connecting to 192.168.1.48
INFO Uploading /data/build/everything-presence-one-62a3c4/.pioenvs/everything-presence-one-62a3c4/firmware.bin (952784 bytes)
Uploading: [============================================================] 100% Done...

INFO Upload took 3.60 seconds, waiting for result...
INFO OTA successful
INFO Successfully uploaded program.
INFO Starting log output from 192.168.1.48 using esphome API
INFO Successfully connected to everything-presence-one-xxxxxx @ 192.168.1.48 in 5.163s
INFO Successful handshake with everything-presence-one-xxxxxx @ 192.168.1.48 in 0.886s`

BenHolmes, 2024-07-23:
Logs on a compilation that doesn't push update but doesn't cause the encryption issue.
[logs_everything-presence-beta-708010_run.txt](https://github.com/user-attachments/files/16353808/logs_everything-presence-beta-708010_run.txt)
[logs_everything-presence-one-6e94fc_run.txt](https://github.com/user-attachments/files/16353823/logs_everything-presence-one-6e94fc_run.txt)
Just on a few devices, can grab all if necessary.

EverythingSmartHome, 2024-07-23:
@atle-dahlman everything looks great there?

EverythingSmartHome, 2024-07-23:
@BenHolmes also looks good

EverythingSmartHome, 2024-07-23:
@atle-dahlman sorry I'm not sure what your issue is, as your logs show it uploading and completing 100% successfully

EverythingSmartHome, 2024-07-23:
That's because you remove the API key BEFORE flashing. If you remove the API key and flash, that means the EP1 is now flashed with no encryption key, while home assistant is still trying to contact it using the encryption key.

The only option now would be to either add the key back if you happened to take a note of it and flash again, or delete and re-add to home assistant without the key

BenHolmes, 2024-07-23:
> @BenHolmes also looks good

Yeah device is fine now that I corrected the encryption. However the update is not applying when compiling from GitHub but perhaps I mis-understood and this is not currently on the standard config?

BenHolmes, 2024-07-23:
Ran updates on all and updates are still available.

[image]

atle-dahlman, 2024-07-23:
Ok.  Deleted the device and re-added.  Working now.

dannytsang, 2024-07-23:
> This isn't an issue with the update, I'm guessing you upgraded via the entity in Home Assistant? The reason this occurs is because you have an encryption key configured in the ESPHome add-on, which [...]

This fixed it for me. Remove the encryption key and install via USB.

BenHolmes, 2024-07-23:
> This fixed it for me. Remove the encryption key and install via USB.

When you installed via USB did you get the latest 1.2.5? trying to avoid taking them down from the wall

atle-dahlman, 2024-07-23:
I flashed the device completely without the api section.  Deleted the device and re-added.
The device came back with same name on device and all sensors, AND it says it's running on 1.2.5.
I'm happy now.

BenHolmes, 2024-07-23:
> I flashed the device completely without the api section. Deleted the device and re-added. The device came back with same name on device and all sensors, AND it says it's running on 1.2.5. I'm happy now.

When you re-added do you have the api section with encryption?

EverythingSmartHome, 2024-07-23:
It's likely your ESPHome is caching, you can try and temporarily switch to another release by changing this line (for example, I'm not sure which version you are on):

```
  EverythingSmartTechnology.Everything_Presence_Lite: github://everythingsmarthome/everything-presence-one/everything-presence-one-ble.yaml@main
```

To

```
  EverythingSmartTechnology.Everything_Presence_Lite: github://everythingsmarthome/everything-presence-one/everything-presence-one.yaml@main@main
```

Save, install and just do manual download (to force a recompile but without actually installing it to the device. Then change it back and repeat but install over wireless

atle-dahlman, 2024-07-23:
> > I flashed the device completely without the api section. Deleted the device and re-added. The device came back with same name on device and all sensors, AND it says it's running on 1.2.5. I'm happy now.
> 
> When you re-added do you have the api section with encryption?

Nope.  The entire api section is gone.

BenHolmes, 2024-07-23:
> Nope. The entire api section is gone.

So does that mean your device is now un-encrypted?

dannytsang, 2024-07-23:
> When you installed via USB did you get the latest 1.2.5? trying to avoid taking them down from the wall

Yes with 1.2.5. I got the update notice and cleaned the build file to make sure it pulled down the latest version.

EverythingSmartHome, 2024-07-23:
Yes hitting the clean build files button first is another way to clear cache as opposed to switching releases

atle-dahlman, 2024-07-23:
> So does that mean your device is now un-encrypted?

Most likely.  Don't know how to find the encryption key.  Maybe by taking it down from the wall and connecting to a computer via USB?

BenHolmes, 2024-07-23:
```
substitutions:
  name: everything-presence-one-6db96c
  friendly_name: Everything Presence One 6db96c
packages:
  Everything_Smart_Technology.Everything_Presence_One: github://everythingsmarthome/everything-presence-one/everything-presence-one-sen0609.yaml@main@main
esphome:
  name: ${name}
  name_add_mac_suffix: false
  friendly_name: ${friendly_name}

api:
  encryption:
    key: 

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
```
Attempting to resolve caching and did a clean build and still didn't get 1.2.5 perhaps I am not pulling the proper version.

BenHolmes, 2024-07-23:
> Most likely. Don't know how to find the encryption key. Maybe by taking it down from the wall and connecting to a computer via USB?

Yeah trying to avoid that, maybe I'll just run un-encrypted :/

EverythingSmartHome, 2024-07-23:
You have a double "@main" at the end of the package line. Not sure if that is causing issues

BenHolmes, 2024-07-23:
> You have a double "@main" at the end of the package line. Not sure if that is causing issues

Not a problem, just added because I saw that in your example.

> It's likely your ESPHome is caching, you can try and temporarily switch to another release by changing this line (for example, I'm not sure which version you are on):
>
> ``` [...]

EverythingSmartHome, 2024-07-23:
My bad, edited it now

BenHolmes, 2024-07-23:
> My bad, edited it now

No worries! it's the same result either way for me. I'll try a couple more restarts and clean builds.

BenHolmes, 2024-07-23:
> > My bad, edited it now
> 
> No worries! it's the same result either way for me. I'll try a couple more restarts and clean builds.

It doesn't look like a caching issue for me; I have cleared, clean build, restarted, and referenced different builds and still cannot get 1.2.5 without removing the api section. Is this by design and I simply missing the context?

EverythingSmartHome, 2024-07-23:
Can you try and change the packages section to this:

```
packages:
  EverythingSmartTechnology.Everything_Presence_Lite:
    url: https://github.com/everythingsmarthome/everything-presence-one
    files: everything-presence-one-sen0609.yaml
    ref: main
    refresh: 0s
 ```

Ocramius, 2024-07-23:
FWIW, reached similar unresponsive EP1 state today, but didn't notice it because I use it to control lights only once it's dark: 

[image]

My ESPHome config, roughly:

```yml
substitutions:
  name: "dining-room-ep1"
packages:
  Everything_Smart_Technology.Everything_Presence_One: github://everythingsmarthome/presence-one/everything-presence-one.yaml@main
esphome:
  name: ${name}
  name_add_mac_suffix: false
api:
  encryption:
    key: <SNIP>

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

logger:
```

Also used the HA upgrade UI under `/config/updates`.

I'll try re-flashing it with 1.2.5 and this config tomorrow: worked fine with 1.2.4 and previous upgrades done via ESPHome UI or HA `/config/updates`.

EverythingSmartHome, 2024-07-23:
> FWIW, reached similar unresponsive EP1 state today, but didn't notice it because I use it to control lights only once it's dark:
> 
> ![EP1 going dark](https://private-user-images.githubusercontent.com/154256/351498095-d9d00aee-1000-4f39-80d3-5184199dc1d7.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjE3NzA2MjgsIm5iZiI6MTcyMTc3MDMyOCwicGF0aCI6Ii8xNTQyNTYvMzUxNDk4MDk1LWQ5ZDAwYWVlLTEwMDAtNGYzOS04MGQzLTUxODQxOTlkYzFkNy5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwNzIzJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDcyM1QyMTMyMDhaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT0yYTRjNjM4NWFmNWIxMmEwNGEyZGU1OGIwODhhODM3NjFlMjg0MDk4ZWU2NjQyYjk1ZTY2YTBmNDg5ZTAyYmFmJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.yd30h4OwUmCdXz9-HVxFiD-1xdupw0WTfqXt96OB4UY)
> 
> My ESPHome config, roughly:
> 
> ```yaml
> substitutions:
>   name: "dining-room-ep1"
> packages:
>   Everything_Smart_Technology.Everything_Presence_One: github://everythingsmarthome/presence-one/everything-presence-one.yaml@main
> esphome:
>   name: ${name}
>   name_add_mac_suffix: false
> api:
>   encryption:
>     key: <SNIP>
> 
> wifi:
>   ssid: !secret wifi_ssid
>   password: !secret wifi_password
> 
> logger:
> ```
> 
> Also used the HA upgrade UI under `/config/updates`.
> 
> I'll try re-flashing it with 1.2.5 and this config tomorrow: worked fine with 1.2.4 and previous upgrades done via ESPHome UI or HA `/config/updates`.

In this case, I would not advise you to use the new update firmware since you have a non standard config, which will overwrite your local changes. Stick to the ESPHome method of updating

Ocramius, 2024-07-23:
> since you have a non standard config

This is what was created by ESPHome on adoption ( a year ago): I added `logger:`, but that's it.

Should I roll that back? Or is this default generated config already out of the standard approach/setup? 🤔

BenHolmes, 2024-07-23:
> Can you try and change the packages section to this:
> 
> ```
> packages:
>   EverythingSmartTechnology.Everything_Presence_One:
>     url: https://github.com/everythingsmarthome/everything-presence-one
>     files: everything-presence-one-sen0609.yaml
>     ref: main
>     refresh: 0s
> ```

Solved it! This is the best solution! Thank you so much for your help!! 😄

Carphone12, 2024-07-23:
I have always updated my EP1 when prompted by Home assistant - after today’s update it is flashing red. Reading the comments above seems removing and reflashing will resolve. 

Question - why has this update knocked it out vs all previous updates been fine?

EverythingSmartHome, 2024-07-23:
> Solved it! This is the best solution! Thank you so much for your help!! 😄

Make sure and revert that back to the way it was, as leaving it will cause a performance hit to your system - should be good to put back to standard now

BenHolmes, 2024-07-23:
> Make sure and revert that back to the way it was, as leaving it will cause a performance hit to your system - should be good to put back to standard now

Ah gotcha! Will do! 😄👍

EverythingSmartHome, 2024-07-23:
> I have always updated my EP1 when prompted by Home assistant - after today’s update it is flashing red. Reading the comments above seems removing and reflashing will resolve.
> 
> Question - why has this update knocked it out vs all previous updates been fine?

It's not the update per say. The update itself is perfectly fine.

I posted the reason above in one of the first messages here:

> This isn't an issue with the update, I'm guessing you upgraded via the entity in Home Assistant? The reason this occurs is because you have an encryption key configured in the ESPHome add-on, which [...]

>You can either flash using the ESPHome add-on again over wireless, or delete from HA and re-add it without the encryption key. Home Assistant may even prompt you to reconfigure without the key. If you do this then make sure to remove the key from your ESPHome addon

The reason is essentially if you've added it to the ESPHome add-on and added an encryption key to your setup which isn't on by default.

BenHolmes, 2024-07-23:
> Make sure and revert that back to the way it was, as leaving it will cause a performance hit to your system - should be good to put back to standard now

So interesting... switching the line from 
```
packages:
  EverythingSmartTechnology.Everything_Presence_One:
    url: https://github.com/everythingsmarthome/everything-presence-one
    files: everything-presence-one-sen0609.yaml
    ref: main
    refresh: 0s
```
to 
```
packages:
  Everything_Smart_Technology.Everything_Presence_One: github://everythingsmarthome/everything-presence-one/everything-presence-one-sen0609.yaml@main
```
reverts the version back from 1.2.5 to 1.2.4

Carphone12, 2024-07-23:
> It's not the update per say. The update itself is perfectly fine.
>
> I posted the reason above in one of the first messages here:
>
>
>
> The reason is essentially if you've added it to the ESPHome add-on [...]

Thanks - all fixed now. I just removed from HA, restarted and added again when auto detected. I did not need to reflash or do anything with my encryption key. 

Back in business - thanks very much

brutus1975cdn, 2024-07-23:
Changing my packages to:

> packages:
  EverythingSmartTechnology.Everything_Presence_One:
    url: https://github.com/everythingsmarthome/everything-presence-one
    files: everything-presence-one-sen0609.yaml
    ref: main
    refresh: 0s

Solved the issue for me.

BenHolmes, 2024-07-23:
> > > > Can you try and change the packages section to this:
> > > > ```
> > > > packages:
> > > >   EverythingSmartTechnology.Everything_Presence_One:
> > > >     url: https://github.com/everythingsmarthome/everything-presence-one
> > > >     files: everything-presence-one-sen0609.yaml
> > > >     ref: main
> > > >     refresh: 0s
> > > > ```
> > > 
> > > 
> > > Solved it! This is the best solution! Thank you so much for your help!! 😄
> > 
> > 
> > Make sure and revert that back to the way it was, as leaving it will cause a performance hit to your system - should be good to put back to standard now
> 
> So interesting... after getting updated to 1.2.5 switching the line from
> 
> ```
> packages:
>   EverythingSmartTechnology.Everything_Presence_One:
>     url: https://github.com/everythingsmarthome/everything-presence-one
>     files: everything-presence-one-sen0609.yaml
>     ref: main
>     refresh: 0s
> ```
> 
> to
> 
> ```
> packages:
>   Everything_Smart_Technology.Everything_Presence_One: github://everythingsmarthome/everything-presence-one/everything-presence-one-sen0609.yaml@main
> ```
> 
> reverts the version back from 1.2.5 to 1.2.4. Despite it supposedly being the same package.

> Make sure and revert that back to the way it was, as leaving it will cause a performance hit to your system - should be good to put back to standard now

It looks like like this line will update you without needing to remove from home assistant, remove the api line, or face performance issues with 0s refresh.

```
packages:
  Everything_Smart_Technology.Everything_Presence_One: github://everythingsmarthome/everything-presence-one/everything-presence-one-sen0609.yaml
```
curious why @main would change anything

brutus1975cdn, 2024-07-23:
> packages:
  Everything_Smart_Technology.Everything_Presence_One: github://everythingsmarthome/everything-presence-one/everything-presence-one-sen0609.yaml

Ok.  Did this.  everything is online, will advise if it changes.

Thank you.

dmshimself, 2024-07-23:
I've hit this issue too and I wonder if it might be possible to pop up a warning somewhere which suggests that you don't use the new ESPHome manufacturer provided update feature if you also have used ESPHome to update the device?  Ages back ESPHome went through highly recommending encryption be applied and so I think many people will have done this and eagerly press the button to update.

liquidbrains, 2024-07-24:
> I've hit this issue too and I wonder if it might be possible to pop up a warning somewhere which suggests that you don't use the new ESPHome manufacturer provided update feature if you also have [...]

Continuing on this, "Adopting" the EP1 device into ESPHome sets an encryption key by default.  This seems like a conflict between expected configurations that could cause repeat issues?

PHLAK, 2024-08-19:
I've run into this issue a few times including as recently as this morning. Been able to get the device working again by re-flashing manually.

However, I was wondering if there's a way to disable the Home Assistant update notification for these devices to prevent this from happening in the future.

EverythingSmartHome, 2024-08-21:
> FWIW, I've just run the software update proposed by HA for the official EP1 today (fresh off the shelf) and it's bricked **again**.
> 
> Is there an official "setting up with ESPHome" guide somewhere? The default adoption approach via ESPHome leads to a bricked device at every update, which is getting kinda tiresome...

Appreciate it's not easy from a user perspective to tell, but it's not the update thats at issue. It's that you are overwriting your encryption key as above. But I'm in agreement that it's not easy to tell what's happening from a user perspective.

Anyways, update 1.2.6 does actually disable the new GitHub entity by default. This should stop you updating this way and overwriting the key moving forward

Ocramius, 2024-08-21:
@EverythingSmartHome thanks, glad to hear of the `1.2.6` changes! Re-flashing tonight 🤞

PHLAK, 2024-08-21:
> Anyways, update 1.2.6 does actually disable the new GitHub entity by default. This should stop you updating this way and overwriting the key moving forward

Perfect, thank you!

sumitkm, 2024-08-29:
Okay, I am confused. So the solution is to remove Encryption? But if I do that ESPHome complains that device doesn't have encryption. How do I get rid of encryption requirement from ESPHome? I guess I am talking to myself, I'll go figure out.

ESPHome is bugging me out at the moment. All my presence sensors are either broken or stuck failing to upgrade. I think the encryption module is too big, so my 1MB Wemos boards, that I used to bootleg a few of these, are out of space for OTA!

EverythingSmartHome, 2024-08-30:
> The advice to remove encryption key isn't great, because it will fail to upgrade ESP Home firmware the next time. You can't have "Auto Adopt" in HA and not have encryption key. Once you have encryption key, you can't remove it and install again so you'll be stuck with your version of ESP.

Just to clarify, I was stating that removing the encryption key is one option, but I also provided another option to resolve. What you are saying isn't strictly true, you can absolutely remove the encryption key without issue and still upgrade anytime. You just need to ensure that both sides match (either both have the encryption key or both don't use it). You can absolutely auto adopt in HA and not use the encryption key. And you can absolutely remove the encryption key once using it.

Glad you got it working in the end and hope that helps clarify

closed 2024-08-30 by EverythingSmartHome