EverythingSmartHome/everything-presence-one #208: OTA Update to 2024.7.0 and 2024.7.1 failing

TomFrank19087, 2024-07-24:
This applies to both EP1 and EP Lite. When updating to 2024.6.6 I had to change my config files ota:, platform: esphome to ota:, platform: http_request, and the update was successful. Both the 2024.7.0 and 2024.7.1 updates fail, claiming platform: esphome is missing (see logs below). If I revert back to platform: esphome, it fails because http_request is missing, just like iit did before the 2024.6.6 update. If I include both I get a duplicate parameter error. The complies seem to go fine, so I think there's an issue with my yaml files. Below are my yaml files and the update all logs from ESPHome. Other than the ota platform changes, I've made no other edits to the yaml files since adding the devices. Thank you.

everything-presence-one-e011a8.yaml

substitutions:
  name: everything-presence-one-e011a8
packages:
  Everything_Smart_Technologies.Everything_Presence_One: github://everythingsmarthome/presence-one/everything-presence-one.yaml
esphome:
  name: ${name}
  name_add_mac_suffix: false

ota:
  platform: http_request

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

everything-presence-lite-890a04.yaml

substitutions:
  name: everything-presence-lite-890a04
  friendly_name: Everything Presence Lite 890a04
packages:
  EverythingSmartTechnology.Everything_Presence_Lite: github://everythingsmarthome/everything-presence-lite/everything-presence-lite-ha.yaml@main
esphome:
  name: ${name}
  name_add_mac_suffix: false
  friendly_name: ${friendly_name}
api:
  encryption:
    key: XXX Redacted XXX

ota:
  platform: http_request

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

Logs

Updating /config/esphome/everything-presence-lite-890a04.yaml
------------------------------------------------------------

INFO ESPHome 2024.7.1
INFO Reading configuration /config/esphome/everything-presence-lite-890a04.yaml...
INFO Generating C++ source...
INFO Compiling app...
Processing everything-presence-lite-890a04 (board: esp32dev; framework: espidf; platform: platformio/espressif32@5.4.0)
--------------------------------------------------------------------------------
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
 - framework-espidf @ 3.40407.0 (4.4.7) 
 - tool-cmake @ 3.16.9 
 - tool-ninja @ 1.10.2 
 - toolchain-esp32ulp @ 2.35.0-20220830 
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch5
Reading CMake configuration...
Dependency Graph
|-- noise-c @ 0.1.4
|-- ArduinoJson @ 6.18.5
|-- Improv @ 1.2.3
Compiling .pioenvs/everything-presence-lite-890a04/src/esphome/components/api/api_connection.o
Compiling .pioenvs/everything-presence-lite-890a04/src/esphome/components/api/api_frame_helper.o
Compiling .pioenvs/everything-presence-lite-890a04/src/esphome/components/api/api_pb2.o
Compiling .pioenvs/everything-presence-lite-890a04/src/esphome/components/api/api_pb2_service.o
Compiling .pioenvs/everything-presence-lite-890a04/src/esphome/components/api/api_server.o
Compiling .pioenvs/everything-presence-lite-890a04/src/esphome/components/api/list_entities.o
Compiling .pioenvs/everything-presence-lite-890a04/src/esphome/components/api/proto.o
[... 256 lines cut ...]
Compiling .pioenvs/everything-presence-one-e011a8/src/esphome/core/entity_base.o
Compiling .pioenvs/everything-presence-one-e011a8/src/esphome/core/helpers.o
Compiling .pioenvs/everything-presence-one-e011a8/src/esphome/core/log.o
Compiling .pioenvs/everything-presence-one-e011a8/src/esphome/core/ring_buffer.o
Compiling .pioenvs/everything-presence-one-e011a8/src/esphome/core/scheduler.o
Compiling .pioenvs/everything-presence-one-e011a8/src/esphome/core/string_ref.o
Compiling .pioenvs/everything-presence-one-e011a8/src/esphome/core/util.o
Compiling .pioenvs/everything-presence-one-e011a8/src/main.o
Linking .pioenvs/everything-presence-one-e011a8/firmware.elf
RAM:   [=         ]   9.7% (used 31880 bytes from 327680 bytes)
Flash: [=====     ]  51.4% (used 944073 bytes from 1835008 bytes)
Building .pioenvs/everything-presence-one-e011a8/firmware.bin
Creating esp32 image...
Successfully created esp32 image.
esp32_create_combined_bin([".pioenvs/everything-presence-one-e011a8/firmware.bin"], [".pioenvs/everything-presence-one-e011a8/firmware.elf"])
Wrote 0xf8180 bytes to file /data/build/everything-presence-one-e011a8/.pioenvs/everything-presence-one-e011a8/firmware.factory.bin, ready to flash to offset 0x0
esp32_copy_ota_bin([".pioenvs/everything-presence-one-e011a8/firmware.bin"], [".pioenvs/everything-presence-one-e011a8/firmware.elf"])
======================== [SUCCESS] Took 180.13 seconds ========================
INFO Successfully compiled program.
ERROR Cannot upload Over the Air as the ota configuration is not present or does not include platform: esphome
 [ERROR] /config/esphome/everything-presence-one-e011a8.yaml 

======================== [SUMMARY] ========================
  - /config/esphome/everything-presence-lite-890a04.yaml: FAILED
  - /config/esphome/everything-presence-one-e011a8.yaml: FAILED

EverythingSmartHome, 2024-07-25:
This isn't need it at all, it's because the config is modified from standard. To revert back to original, completely remove the OTA block as that gets pulled down from the Github as part of the package.

TomFrank19087, 2024-07-25:
That did the trick! Both EP1 and EPLite successfully updated to the just posted 2024.7.2 update. Many thanks, Lewis!

closed 2024-07-25 by EverythingSmartHome