EverythingSmartHome/everything-presence-one #210: Everything-presence-one in a boot loop after recent update

pawilli, 2024-07-27:
My Ep1 is in a boot loop after update.   I have attached my conf and the boot loop log below.

Please assist.  Thank you.

```
----ESPHOME` YAML----

substitutions:
  name: ep1-up
  friendly_name: Motion Living room
packages:
  Everything_Smart_Technology.Everything_Presence_One: github://everythingsmarthome/everything-presence-one/everything-presence-one.yaml@main

esphome:
  name: ${name}
  name_add_mac_suffix: false
  friendly_name: ${friendly_name}

api:
  encryption:
    key: <<redacted>>

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

#----END ESPHOME YAML----
```
-----LOG FILE BELOW from boot to boot---

[09:18:46]Rebooting...
[09:18:46]ets Jul 29 2019 12:21:46
[09:18:46]
[09:18:46]rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
[09:18:46]configsip: 0, SPIWP:0xee
[09:18:46]clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
[09:18:46]mode:DIO, clock div:2
[09:18:46]load:0x3fff0030,len:6652
[09:18:46]ho 0 tail 12 room 4
[09:18:46]load:0x40078000,len:15052
[09:18:46]load:0x40080400,len:3836
[09:18:46]entry 0x4008069c
[09:18:46]I (29) boot: ESP-IDF 4.4.7 2nd stage bootloader
[09:18:46]I (29) boot: compile time 17:10:53
[09:18:46]I (29) boot: Multicore bootloader
[09:18:46]I (32) boot: chip revision: v3.0
[09:18:46]I (36) boot.esp32: SPI Speed      : 40MHz
[09:18:46]I (41) boot.esp32: SPI Mode       : DIO
[09:18:46]I (45) boot.esp32: SPI Flash Size : 4MB
[09:18:46]I (50) boot: Enabling RNG early entropy source...
[09:18:46]I (55) boot: Partition Table:
[09:18:46]I (59) boot: ## Label            Usage          Type ST Offset   Length
[09:18:46]I (66) boot:  0 otadata          OTA data         01 00 00009000 00002000
[09:18:46]I (73) boot:  1 phy_init         RF data          01 01 0000b000 00001000
[09:18:46]I (81) boot:  2 app0             OTA app          00 10 00010000 001c0000
[09:18:46]I (88) boot:  3 app1             OTA app          00 11 001d0000 001c0000
[09:18:46]I (96) boot:  4 nvs              WiFi data        01 02 00390000 0006d000
[09:18:46]I (103) boot: End of partition table
[09:18:46]I (108) esp_image: segment 0: paddr=001d0020 vaddr=3f400020 size=37f5ch (229212) map
[09:18:46]I (199) esp_image: segment 1: paddr=00207f84 vaddr=3ffbdb60 size=03380h ( 13184) load
[09:18:46]I (205) esp_image: segment 2: paddr=0020b30c vaddr=40080000 size=04d0ch ( 19724) load
[09:18:46]I (213) esp_image: segment 3: paddr=00210020 vaddr=400d0020 size=a93c4h (693188) map
[09:18:46]I (464) esp_image: segment 4: paddr=002b93ec vaddr=40084d0c size=10224h ( 66084) load
[09:18:46]I (502) boot: Loaded app from partition at offset 0x1d0000
[09:18:46]I (502) boot: Disabling RNG early entropy source...
[09:18:46]I (513) cpu_start: Multicore app
[09:18:46]I (514) cpu_start: Pro cpu up.
[09:18:46]I (514) cpu_start: Starting app cpu, entry point is 0x40081f94
[09:18:46]I (502) cpu_start: App cpu up.
[09:18:46]I (532) cpu_start: Pro cpu start user code
[... 164 lines cut ...]
[E][esp-idf:000][wifi]: I (6573) wifi:
[E][esp-idf:000][wifi]: <ba-add>idx:1 (ifx:0, 70:0f:6a:7c:b1:e2), tid:0, ssn:0, winSize:64
[E][esp-idf:000][wifi]: 

E (11566) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
[09:18:46]E (11566) task_wdt:  - loopTask (CPU 1)
[09:18:46]E (11566) task_wdt: Tasks currently running:
[09:18:46]E (11566) task_wdt: CPU 0: IDLE0
[09:18:46]E (11566) task_wdt: CPU 1: IDLE1
[09:18:46]E (11566) task_wdt: Aborting.
[09:18:46]
[09:18:46]abort() was called at PC 0x40101e40 on core 0
[09:18:46]
[09:18:46]
[09:18:46]Backtrace: 0x40082966:0x3ffbe3f0 0x400896e5:0x3ffbe410 0x4008f89a:0x3ffbe430 0x40101e40:0x3ffbe4a0 0x400832e9:0x3ffbe4c0 0x40175e87:0x3ffb2f00 0x40102591:0x3ffb2f20 0x4008aabc:0x3ffb2f40
[09:18:46]
[09:18:46]
[09:18:46]
[09:18:46]
[09:18:46]ELF file SHA256: 14b136596c557818
[09:18:46]
[09:18:46]Rebooting...
.....boot loop continues.....

----end log----

EverythingSmartHome, 2024-07-27:
Hello, do you happen to have it blocked from internet access?

ThreeFN, 2024-08-02:
Can confirm I'm having a similar issue, and yes, in my case my 'esphome net' is default blocked from most networks including WAN.  I assume there's a blocking internet call that causes a reboot on failure?

EverythingSmartHome, 2024-08-03:
Can you share config? Also if you can share the firewall rule you have that would be useful as I am unable to reproduce this on my side currently

ThreeFN, 2024-08-06:
I'll start with this screen shot from pfsense but I can provide more detail if needed:
[image]

brantje, 2024-10-28:
Same here, why is this happening to a device that supposed to be local?

EverythingSmartHome, 2024-10-28:
> Same here, why is this happening to a device that supposed to be local?

It is a bug in ESPHome that causes this, related to the new manufacturer mechanism which lets it check the GitHub directly for updates. I will repeat again, it is a bug and certainly not intentional. It also doesn't happen under every circumstances

brantje, 2024-10-28:
Moved it to a wifi with internet access, but still getting the error. 
Downside of this i lost all EP sensors in my house.

Would merging all yamls into a single file solve this?

EverythingSmartHome, 2024-10-28:
Post the startup logs if you can please

What would that change?

brantje, 2024-10-28:
```

[11:02:49][E][esp-idf:000]: I (241) gpio: GPIO[25]| InputEn: 1| OutputEn: 1| OpenDrain: 1| Pullup: 1| Pulldown: 0| Intr:0 

[E][esp-idf:000]: I (242) gpio: GPIO[26]| InputEn: 1| OutputEn: 1| OpenDrain: 1| Pullup: 1| Pulldown: 0| Intr:0 

[E][esp-idf:000]: I (276) uart: queue free spaces: 20

[E][esp-idf:000]: I (277) gpio: GPIO[32]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 

[E][esp-idf:000]: I (278) gpio: GPIO[15]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pulldown: 1| Intr:0 

[E][shtcx:080]: sensor polling failed
[E][esp-idf:000]: I (297) gpio: GPIO[33]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pulldown: 1| Intr:0 

[E][esp-idf:000]: I (314) wifi:
[E][esp-idf:000]: wifi driver task: 3ffc7da4, prio:23, stack:6656, core=0
[E][esp-idf:000]: 

[E][esp-idf:000][wifi]: I (1262) system_api: Base MAC address is not set

[E][esp-idf:000][wifi]: I (1269) system_api: read default base MAC address from EFUSE

[E][esp-idf:000][wifi]: I (1283) wifi:
[E][esp-idf:000][wifi]: wifi firmware version: ff661c3
[E][esp-idf:000][wifi]: 

[E][esp-idf:000][wifi]: I (1300) wifi:
[E][esp-idf:000][wifi]: wifi certification version: v7.0
[E][esp-idf:000][wifi]: 

[E][esp-idf:000][wifi]: I (1310) wifi:
[E][esp-idf:000][wifi]: config NVS flash: enabled
[E][esp-idf:000][wifi]: 

[E][esp-idf:000][wifi]: I (1331) wifi:
[E][esp-idf:000][wifi]: config nano formating: disabled
[E][esp-idf:000][wifi]: 

[E][esp-idf:000][wifi]: I (1341) wifi:
[E][esp-idf:000][wifi]: Init data frame dynamic rx buffer num: 32
[E][esp-idf:000][wifi]: 

[E][esp-idf:000][wifi]: I (1362) wifi:
[E][esp-idf:000][wifi]: Init static rx mgmt buffer num: 5
[E][esp-idf:000][wifi]: 

[E][esp-idf:000][wifi]: I (1382) wifi:
[E][esp-idf:000][wifi]: Init management short buffer num: 32
[E][esp-idf:000][wifi]: 

[E][esp-idf:000][wifi]: I (1392) wifi:
[E][esp-idf:000][wifi]: Init dynamic tx buffer num: 32
[E][esp-idf:000][wifi]: 

[E][esp-idf:000][wifi]: I (1413) wifi:
[E][esp-idf:000][wifi]: Init static rx buffer size: 1600
[E][esp-idf:000][wifi]: 

[E][esp-idf:000][wifi]: I (1433) wifi:
[E][esp-idf:000][wifi]: Init static rx buffer num: 10
[E][esp-idf:000][wifi]: 

[E][esp-idf:000][wifi]: I (1444) wifi:
[E][esp-idf:000][wifi]: Init dynamic rx buffer num: 32
[E][esp-idf:000][wifi]: 

[E][esp-idf:000]: I (1465) wifi_init: rx ba win: 6

[E][esp-idf:000]: I (1474) wifi_init: tcpip mbox: 32

[E][esp-idf:000]: I (1484) wifi_init: udp mbox: 6

[E][esp-idf:000]: I (1485) wifi_init: tcp mbox: 6

[E][esp-idf:000]: I (1495) wifi_init: tcp tx win: 5760

[E][esp-idf:000]: I (1495) wifi_init: tcp rx win: 5760

[E][esp-idf:000]: I (1505) wifi_init: tcp mss: 1440

[E][esp-idf:000]: I (1515) wifi_init: WiFi IRAM OP enabled

[E][esp-idf:000]: I (1515) wifi_init: WiFi RX IRAM OP enabled

[E][esp-idf:000][wifi]: I (1527) phy_init: phy_version 4791,2c4672b,Dec 20 2023,16:06:06

[E][esp-idf:000][wifi]: I (1611) wifi:
[E][esp-idf:000][wifi]: mode : sta (d4:d4:da:29:40:18)
[E][esp-idf:000][wifi]: 

[E][esp-idf:000][wifi]: I (1613) wifi:
[E][esp-idf:000][wifi]: enable tsf
[E][esp-idf:000][wifi]: 

[E][esp-idf:000][wifi]: I (1617) wifi:
[E][esp-idf:000][wifi]: Set ps type: 1

[E][esp-idf:000][wifi]: 

[E][shtcx:094]: sensor read failed
[E][esp-idf:000][wifi]: I (1667) wifi:
[E][esp-idf:000][wifi]: Set ps type: 1

[E][esp-idf:000][wifi]: 

[E][shtcx:094]: sensor read failed
[E][esp-idf:000][wifi]: I (4446) wifi:
[E][esp-idf:000][wifi]: new:<11,0>, old:<1,0>, ap:<255,255>, sta:<11,0>, prof:1
[E][esp-idf:000][wifi]: 

[E][esp-idf:000][wifi]: I (4659) wifi:
[E][esp-idf:000][wifi]: state: init -> auth (b0)
[E][esp-idf:000][wifi]: 

[E][esp-idf:000][wifi]: I (4712) wifi:
[E][esp-idf:000][wifi]: state: auth -> assoc (0)
[E][esp-idf:000][wifi]: 

[E][esp-idf:000][wifi]: I (4787) wifi:
[E][esp-idf:000][wifi]: state: assoc -> run (10)
[E][esp-idf:000][wifi]: 

[E][esp-idf:000][wifi]: I (4832) wifi:
[E][esp-idf:000][wifi]: connected with SecureHaven, aid = 4, channel 11, BW20, bssid = e4:38:83:8d:0b:2a
[E][esp-idf:000][wifi]: 

[E][esp-idf:000][wifi]: I (4834) wifi:
[E][esp-idf:000][wifi]: security: WPA2-PSK, phy: bgn, rssi: -34
[E][esp-idf:000][wifi]: 

[E][esp-idf:000][wifi]: I (4849) wifi:
[E][esp-idf:000][wifi]: pm start, type: 1

[E][esp-idf:000][wifi]: 

[E][esp-idf:000][wifi]: I (4859) wifi:
[E][esp-idf:000][wifi]: AP's beacon interval = 102400 us, DTIM period = 1
[E][esp-idf:000][wifi]: 

[E][esp-idf:000][wifi]: I (4888) wifi:
[E][esp-idf:000][wifi]: <ba-add>idx:0 (ifx:0, e4:38:83:8d:0b:2a), tid:6, ssn:2, winSize:64
[E][esp-idf:000][wifi]: 

[E][esp-idf:000][sys_evt]: I (5884) esp_netif_handlers: sta ip: 192.168.10.243, mask: 255.255.255.0, gw: 192.168.10.1

E (10901) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
[11:02:49]E (10901) task_wdt:  - loopTask (CPU 1)
[11:02:49]E (10901) task_wdt: Tasks currently running:
[11:02:49]E (10901) task_wdt: CPU 0: IDLE0
[11:02:49]E (10901) task_wdt: CPU 1: IDLE1
[11:02:49]E (10901) task_wdt: Aborting.
[11:02:49]
[11:02:49]abort() was called at PC 0x40101de4 on core 0
[11:02:49]
[11:02:49]
[11:02:49]Backtrace: 0x40082966:0x3ffb08e0 0x400896e5:0x3ffb0900 0x4008f89e:0x3ffb0920 0x40101de4:0x3ffb0990 0x400832e9:0x3ffb09b0 0x40176247:0x3ffbaa20 0x40102535:0x3ffbaa40 0x4008aabc:0x3ffbaa60
[11:02:49]
[11:02:49]
[11:02:49]
[11:02:49]
[11:02:49]ELF file SHA256: 166a3091a88fa130
[11:02:49]
[11:02:49]Rebooting...
```

Now i think of it, won't change anything indeed.

EverythingSmartHome, 2024-10-28:
Yeah, this log looks exactly like the bug I'm describing above. One thing I've helped a couple users with recently is that they think internet is accessible, however turns out that their primary DNS server does not work and so the device is unable to reach GitHub. ESPHome doesn't seem to use secondary DNS servers so make sure that primary is OK

brantje, 2024-10-28:
I'll setup a completely open vlan with 9.9.9.9 as DNS and try again.

brantje, 2024-10-28:
Using 9.9.9.9 on a separate vlan worked. 
Ill check pihole logs, maybe its blocking a domain it shouldn't.

brantje, 2024-10-29:
I do see everythingsmarthome.github.io coming by in the requests, it is allowed, yet it crashes.
[image]

Added `github.io` to the whitelist and now it seems to work.

ThreeFN, 2024-10-30:
Can confirm that after running ntopng and watching traffic on the EP1 and similar 'manufacturer esphome' devices that they typically try to connect at bootup and don't fail gracefully if they're unable.  I've implemented the rules below.  I had a similar experience with the UltimateSensor from smarthomeshop and that one required a different domain for the download of a sound file.  Essentially I look through the device configuration yaml for domains and add an appropriate rule.

[image]

fnord123, 2024-11-03:
Is there an esphome github bug open for this?  I've search around and found the following, but they all seem to be closed:

https://github.com/esphome/issues/issues/6074
https://github.com/esphome/issues/issues/6187

Both were resolved as of August.

therealeldaria, 2024-12-21:
I also have this problem, bootloop if on isolated LAN.

pawilli, 2025-03-26:
> Can confirm that after running ntopng and watching traffic on the EP1 and similar 'manufacturer esphome' devices that they typically try to connect at bootup and don't fail gracefully if they're [...]

I just installed the most recent update and guess what? -> boot loop again with multiple Wi-Fi errors.  Why does this continue to happen with every update?   It is making me apprehensive about adding updates now.

EP1 is beginning to look extremely unreliable and I am surprised this issue has not been resolved since this thread was open in [Jul 27, 2024]

pawilli, 2025-03-26:
> I just installed the most recent update and guess what? -> boot loop again with multiple Wi-Fi errors. Why does this continue to happen with every update? It is making me apprehensive about adding [...]

Ok - I applied the suggestions from @therealeldaria, @brantje and @ThreeFN as a pass rule and it is working now.  

For your convenience I am typing them out so you can just copy and paste - getting them from the images was a bit difficult.

`github.io, everythingsmarthome.github.io, githubusercontent.com, githubusercontent.com, raw.githubusercontent.com`

Thanks guys!! 

@EverythingSmartHome please get this resolved otherwise it will continue to be an issue for most users who might not know about this page.

EverythingSmartHome, 2025-03-27:
There is no need, this was resolved quite a few updates ago now in ESPHome itself. Should no longer be a problem and it should fail gracefully if internet is unavailable

EverythingSmartHome, 2025-03-27:
> EP1 is beginning to look extremely unreliable and I am surprised this issue has not been resolved since this thread was open in [Jul 27, 2024]

I understand it's a frustrating issue, but this comment doesn't really fit. The issue originally existed in upstream ESPHome, not something I can fix or resolve myself unfortunately.

If you are still having this issue (no other reports), then please open a new issue and we can take a look

closed 2025-03-27 by EverythingSmartHome