EverythingSmartHome/everything-presence-one #211: Needs reconfigure every day

cowboysdude, 2024-08-02:
I've updated, reinstalled and it still every single day has to be reconfigured.

Not sure what's going on but it's getting annoying.

EverythingSmartHome, 2024-08-03:
I'm not sure what you are referring to here, any chance of more information, details, context?

taitum1234, 2024-10-19:
Im having this issue as well. it seems like every day (or every other day) the sensor gives me a notification saying it needs to be reconfigured. when i press on it it says it needs an api encryption key. i go into the config and get it and sometimes it works. sometimes it doesnt, saying api key isnt valid. not sure where im supposed to go to get a valid one if the one in hte config isnt correct? when it doesnt work its jsut down for days until it finally works again somehow?

EverythingSmartHome, 2024-10-19:
If your having an issue with encryption keys, I recommend removing the config from the ESPHome add-on (since that's where the confusion comes from as it generated the encryption keys)

taitum1234, 2024-12-05:
im back commenting on this thread as the issue fixed itself somehow but has now returned again months later. based on your comment of removing the config, i dont understand what to do after. i remove it from esphome but then how do i bring it back into home assistant? why is it the encryption in the device config isnt correct? just not understanding