EverythingSmartHome/everything-presence-one #219: No sensor information

dannytsang, 2024-08-19:
I upgraded an EP1 to 1.2.6 from Home Assistant and the sensors are all showing as unavailable.

I have cleaned the build files and reinstalled the firmware from ESPHome and still the same.

The logs are showing the following the below and stays for for a long time (5+ minutes).
```
[21:16:27]I (13) boot: ESP-IDF 4.4.7 2nd stage bootloader
[21:16:27]I (13) boot: compile time 13:42:52
[21:16:27]I (13) boot: Multicore bootloader
[21:16:27]I (15) boot: chip revision: v3.0
[21:16:27]I (19) boot.esp32: SPI Speed      : 40MHz
[21:16:27]I (24) boot.esp32: SPI Mode       : DIO
[21:16:27]I (28) boot.esp32: SPI Flash Size : 4MB
[21:16:27]I (33) boot: Enabling RNG early entropy source...
[21:16:27]I (38) boot: Partition Table:
[21:16:27]I (42) boot: ## Label            Usage          Type ST Offset   Length
[21:16:27]I (49) boot:  0 otadata          OTA data         01 00 00009000 00002000
[21:16:27]I (56) boot:  1 phy_init         RF data          01 01 0000b000 00001000
[21:16:27]I (64) boot:  2 app0             OTA app          00 10 00010000 001c0000
[21:16:27]I (71) boot:  3 app1             OTA app          00 11 001d0000 001c0000
[21:16:27]I (79) boot:  4 nvs              WiFi data        01 02 00390000 0006d000
[21:16:27]I (86) boot: End of partition table
[21:16:27]I (91) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=44f2ch (282412) map
[21:16:27]I (201) esp_image: segment 1: paddr=00054f54 vaddr=3ffbdb60 size=050b4h ( 20660) load
[21:16:27]I (209) esp_image: segment 2: paddr=0005a010 vaddr=40080000 size=06008h ( 24584) load
[21:16:27]I (220) esp_image: segment 3: paddr=00060020 vaddr=400d0020 size=ff48ch (1045644) map
[21:16:27]I (598) esp_image: segment 4: paddr=0015f4b4 vaddr=40086008 size=18328h ( 99112) load
[21:16:27]I (654) boot: Loaded app from partition at offset 0x10000
[21:16:27]I (654) boot: Disabling RNG early entropy source...
[21:16:27]I (666) cpu_start: Multicore app
[21:16:27]I (666) cpu_start: Pro cpu up.
[21:16:27]I (666) cpu_start: Starting app cpu, entry point is 0x400826b0
[21:16:27]I (0) cpu_start: App cpu up.
[21:16:27]I (686) cpu_start: Pro cpu start user code
[21:16:27]I (686) cpu_start: cpu freq: 160000000
[21:16:27]I (686) cpu_start: Application information:
[21:16:27]I (691) cpu_start: Project name:     living_room-motion
[21:16:27]I (697) cpu_start: App version:      2024.7.3
[21:16:27]I (702) cpu_start: Compile time:     Aug  1 2024 13:42:20
[21:16:27]I (708) cpu_start: ELF file SHA256:  69d4142d6b067e08...
[21:16:27]I (714) cpu_start: ESP-IDF:          4.4.7
[21:16:27]I (718) cpu_start: Min chip rev:     v0.0
[21:16:27]I (723) cpu_start: Max chip rev:     v3.99 
[21:16:27]I (728) cpu_start: Chip rev:         v3.0
[21:16:27]I (733) heap_init: Initializing. RAM available for dynamic allocation:
[21:16:27]I (740) heap_init: At 3FFAFF10 len 000000F0 (0 KiB): DRAM
[21:16:27]I (746) heap_init: At 3FFB6388 len 00001C78 (7 KiB): DRAM
[21:16:27]I (752) heap_init: At 3FFB9A20 len 00004108 (16 KiB): DRAM
[21:16:27]I (758) heap_init: At 3FFCB938 len 000146C8 (81 KiB): DRAM
[21:16:27]I (764) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
[21:16:27]I (771) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
[21:16:27]I (777) heap_init: At 4009E330 len 00001CD0 (7 KiB): IRAM
[21:16:27]I (785) spi_flash: detected chip: generic
[21:16:27]I (788) spi_flash: flash io: dio
[21:16:27]I (795) cpu_start: Starting scheduler on PRO CPU.
[21:16:27]I (0) cpu_start: Starting scheduler on APP CPU.
```

Pressing Reset Device goes through the same sequence as below. Is it hanging and not booting up properly?
```
[21:20:41][E][esp-idf:000]: I (94) gpio: GPIO[25]| InputEn: 1| OutputEn: 1| OpenDrain: 1| Pullup: 1| Pulldown: 0| Intr:0 

[E][esp-idf:000]: I (95) gpio: GPIO[26]| InputEn: 1| OutputEn: 1| OpenDrain: 1| Pullup: 1| Pulldown: 0| Intr:0 

[E][esp-idf:000]: I (132) uart: queue free spaces: 20

[E][esp-idf:000]: I (133) gpio: GPIO[32]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 

[E][esp-idf:000]: I (133) gpio: GPIO[15]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pulldown: 1| Intr:0 

[... 195 lines cut ...]
[21:20:41]I (704) cpu_start: cpu freq: 160000000
[21:20:41]I (704) cpu_start: Application information:
[21:20:41]I (708) cpu_start: Project name:     living_room-motion
[21:20:41]I (714) cpu_start: App version:      2024.7.3
[21:20:41]I (719) cpu_start: Compile time:     Aug  1 2024 13:42:20
[21:20:41]I (725) cpu_start: ELF file SHA256:  69d4142d6b067e08...
[21:20:41]I (731) cpu_start: ESP-IDF:          4.4.7
[21:20:41]I (736) cpu_start: Min chip rev:     v0.0
[21:20:41]I (740) cpu_start: Max chip rev:     v3.99 
[21:20:41]I (745) cpu_start: Chip rev:         v3.0
[21:20:41]I (750) heap_init: Initializing. RAM available for dynamic allocation:
[21:20:41]I (757) heap_init: At 3FFAFF10 len 000000F0 (0 KiB): DRAM
[21:20:41]I (763) heap_init: At 3FFB6388 len 00001C78 (7 KiB): DRAM
[21:20:41]I (769) heap_init: At 3FFB9A20 len 00004108 (16 KiB): DRAM
[21:20:41]I (776) heap_init: At 3FFCB938 len 000146C8 (81 KiB): DRAM
[21:20:41]I (782) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
[21:20:41]I (788) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
[21:20:41]I (795) heap_init: At 4009E330 len 00001CD0 (7 KiB): IRAM
[21:20:41]I (802) spi_flash: detected chip: generic
[21:20:41]I (805) spi_flash: flash io: dio
[21:20:41]I (812) cpu_start: Starting scheduler on PRO CPU.
[21:20:41]I (0) cpu_start: Starting scheduler on APP CPU.
```

I cleaned the build files and reinstalled again but still the same issue. Any ideas? The sensors that are on 1.2.5 are up and running.

dannytsang, 2024-08-19:
I forgot, this is the YAML config:
```YAML
substitutions:
  name: "living-room-motion"
packages:
  Everything_Smart_Technology.Everything_Presence_One: github://everythingsmarthome/presence-one/everything-presence-one.yaml@main
esphome:
  name: ${name}
  name_add_mac_suffix: false
esp32:
  board: esp32dev
  framework:
    type: esp-idf

esp32_ble_tracker:
  scan_parameters:
    active: true  # Whether to send scan-request packets to devices to gather more info (like devicename)
    interval: 211ms  # suggested 211ms # default 320ms
    window: 120ms  # suggested 120ms # default 30ms

bluetooth_proxy:
  active: true

switch:
  - platform: restart
    name: "Living Room Motion Restart"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
```

dannytsang, 2024-08-19:
ok maybe not 1.2.6 issue because I changed the below line and still get the same output stopping at Starting CPU scheduler as per above.
```YAML
packages:
  Everything_Smart_Technology.Everything_Presence_One: github://everythingsmarthome/presence-one/everything-presence-one.yaml@1.2.5
```
Is my EP1 dying 😢

dannytsang, 2024-08-19:
I tried again with @main and HA is reporting data now so maybe the output is as expected. I'll put it down my own wifi issue but all is good again.

closed 2024-08-19 by dannytsang

EverythingSmartHome, 2024-08-19:
The log output is normal, no issues there. I'm guessing the issue you had with the update is the same as before, where you applied the GitHub update which overwrote your encryption key.

The 1.2.6 update disables the GitHub updates again as many users were applying it without realising this would happen if they had it added to ESPHome addon. Hopefully it can be re-enabled again once ESPHome handles this better

dannytsang, 2024-08-19:
I thought I sorted the encryption key out last time. If you do enable it again, stick in the release notes.

On another note, how has it picked up 1.2.6 in Home Assistant when GitHub release is at 1.2.5? I didn't see a release note link to go with 1.2.6 either.