EverythingSmartHome/everything-presence-one #222: Issue, after tweaking esphome in HA.

qwerc, 2024-10-03:
Ik have a problem with my sensor EP1. The red LED is flasing. 
The sensor was ok for allmost a year. I did something is esphome, my fault and now the EP1 is offline.
Tryed everything reset sensot unplug usb, etc. No result.
I ma not a specialist , the question is what can I do to fix my problem.
Iám running the EP1 in ESPHome with in HA.

Is there somebody who can help me??

EverythingSmartHome, 2024-10-03:
Hello, the flashing light either means that it is not connected to WiFi, or it is connected to WiFi but Home Assistant has not connected to the API.

You can re-install the firmware from the user guide here which will help you to setup WiFi and connect to Home Assistant: https://everythingsmarthome.github.io/everything-presence-one/updating.html

qwerc, 2024-10-03:
Ok thanks, I will try this, later this evening. Iám going first for a diner.

qwerc, 2024-10-05:
Sorry guys, no result. I managed to load esphome again... But now the origanal code. Where can i feed it???
Iám lost. Please I need need some help.

EverythingSmartHome, 2024-10-05:
No code required, just follow the steps in the link above

qwerc, 2024-10-05:
I'm very sorry, perhaps someone can explain the link for me. I do not understand it. My sensor is now in contact with esphome , but not functioning as a detctor.

EverythingSmartHome, 2024-10-05:
Can you show an image of what you are seeing?

qwerc, 2024-10-06:
This is what I get in edit-mode of esphome

substitutions:
  name: esphome-web-701814
  friendly_name: Everything Presence One

esphome:
  name: ${name}
  friendly_name: ${friendly_name}
  min_version: 2024.6.0
  name_add_mac_suffix: false
  project:
    name: esphome.web
    version: dev

esp32:
  board: esp32dev
  framework:
    type: arduino

# Enable logging
logger:

# Enable Home Assistant API
api:

# Allow Over-The-Air updates
ota:
- platform: esphome

# Allow provisioning Wi-Fi via serial
improv_serial:

wifi:
  # Set up a wifi access point
  ap: {}

# In combination with the `ap` this allows the user
# to provision wifi credentials to the device via WiFi AP.
captive_portal:

dashboard_import:
  package_import_url: github://esphome/example-configs/esphome-web/esp32.yaml@main
  import_full_config: true

# Sets up Bluetooth LE (Only on ESP32) to allow the user
# to provision wifi credentials to the device.
esp32_improv:
  authorizer: none

# To have a "next url" for improv serial
web_server:

EverythingSmartHome, 2024-10-06:
It is not in the instructions to do that? Please follow the instructions, you do not need to do anything in the ESPHome addon. Everything is done from the link I sent you

qwerc, 2024-10-07:
I understand you, but my problem I do not understand the instructions of your link. Thats my problem. I tought realy that this was what the instructions of your link are. So, I am NOT a netif englishman. just a simpel Dutchman. So if you make the instructions in Dutch.... That wil help me much. You see my problem?? What i like the best in a instruction is a stap-plan . That is easy to follow. A lot of text is hard (sometimes) to understand. Atleast for me. I see in the instructions a link to this and than to that... That make it not easy for me.

EverythingSmartHome, 2024-10-07:
You only need to click the big picture of the Home Assistant logo and follow the step by step guide. Do not click any other links

qwerc, 2024-10-07:
I will try it, Tankx

qwerc, 2024-10-07:
It is finally succes. It was not clear for me to click on the HA picture. And I struggeld to find out wich mmWave sensor . But I know it now.
Thanks for your help and patience with me. 
Honestly I love the EP1 sensor. 
Again thanks for the help.

qwerc, 2024-10-09:
Sorry i'm back again.
Basicly with the same problem, I updated the EP1 sensor. And it looks ok.
After aprox 30 min I reasized that the mmwav sensor was not functioning.
So again upload and again no result.
Now the situation is worse, the red LED flasing and no contact withe the sensor.
I question my self, what do i wrong?

EverythingSmartHome, 2024-10-10:
How did you upload?

qwerc, 2024-10-11:
First the good news. The sensor works now. TOP.
I do not know what the realy problem was, heb I removed the sensor (not functioning) Outof esphome, also from HA everywhere. 
Then reboot my HA. After that i connected tthe sensor EP1 again to my desktop. and upload again as writen in your manual (the link) Then... connecting with my HA was een issue. I hat this to do manual (ip adress ETC.) and afterthat reboor and.... it was working.  No issue anymore. Again thanks for your advise .

closed 2024-10-11 by qwerc