EverythingSmartHome/everything-presence-one #223: encryption, cant connect to wifi

CoolhappyDE, 2024-10-07:
`INFO Starting log output from 192.168.178.183 using esphome API
INFO Successfully connected to everything-presence-one-dacf54 @ 192.168.178.183 in 11.309s
WARNING Can't connect to ESPHome API for everything-presence-one-dacf54 @ 192.168.178.183: Handshake timed out after 30.0s (TimeoutAPIError)
INFO Trying to connect to everything-presence-one-dacf54 @ 192.168.178.183 in the background
WARNING everything-presence-one-dacf54 @ 192.168.178.183: Connection error occurred: everything-presence-one-dacf54 @ 192.168.178.183: Connection lost
INFO Successfully connected to everything-presence-one-dacf54 @ 192.168.178.183 in 0.078s
WARNING everything-presence-one-dacf54 @ 192.168.178.183: Connection error occurred: everything-presence-one-dacf54 @ 192.168.178.183: The connection dropped immediately after encrypted hello; Try enabling encryption on the device or turning off encryption on the client (ESPHome Logs 2024.9.2).
INFO Successfully connected to everything-presence-one-dacf54 @ 192.168.178.183 in 0.078s
WARNING everything-presence-one-dacf54 @ 192.168.178.183: Connection error occurred: everything-presence-one-dacf54 @ 192.168.178.183: The connection dropped immediately after encrypted hello; Try enabling encryption on the device or turning off encryption on the client (ESPHome Logs 2024.9.2).
INFO Successfully connected to everything-presence-one-dacf54 @ 192.168.178.183 in 0.109s
WARNING everything-presence-one-dacf54 @ 192.168.178.183: Connection error occurred: everything-presence-one-dacf54 @ 192.168.178.183: The connection dropped immediately after encrypted hello; Try enabling encryption on the device or turning off encryption on the client (ESPHome Logs 2024.9.2).
INFO Successfully connected to everything-presence-one-dacf54 @ 192.168.178.183 in 0.095s
WARNING everything-presence-one-dacf54 @ 192.168.178.183: Connection error occurred: everything-presence-one-dacf54 @ 192.168.178.183: The connection dropped immediately after encrypted hello; Try enabling encryption on the device or turning off encryption on the client (ESPHome Logs 2024.9.2).
INFO Successfully connected to everything-presence-one-dacf54 @ 192.168.178.183 in 0.042s
WARNING everything-presence-one-dacf54 @ 192.168.178.183: Connection error occurred: everything-presence-one-dacf54 @ 192.168.178.183: The connection dropped immediately after encrypted hello; Try enabling encryption on the device or turning off encryption on the client (ESPHome Logs 2024.9.2).
INFO Successfully connected to everything-presence-one-dacf54 @ 192.168.178.183 in 0.097s
WARNING everything-presence-one-dacf54 @ 192.168.178.183: Connection error occurred: everything-presence-one-dacf54 @ 192.168.178.183: The connection dropped immediately after encrypted hello; Try enabling encryption on the device or turning off encryption on the client (ESPHome Logs 2024.9.2).
INFO Successfully connected to everything-presence-one-dacf54 @ 192.168.178.183 in 0.042s
WARNING everything-presence-one-dacf54 @ 192.168.178.183: Connection error occurred: everything-presence-one-dacf54 @ 192.168.178.183: The connection dropped immediately after encrypted hello; Try enabling encryption on the device or turning off encryption on the client (ESPHome Logs 2024.9.2).

Hey i did got the problem that my EP1 is connected to the wifi, but is unavalible in HA.

this is my code:

`substitutions:
  name: everything-presence-one-dacf54
  friendly_name: Küche Bewegungsmelder
packages:
  Everything_Smart_Technology.Everything_Presence_One: github://everythingsmarthome/everything-presence-one/everything-presence-one.yaml@main
esphome:
  name: ${name}
  name_add_mac_suffix: false
  friendly_name: ${friendly_name}`
  
 this is the log on the ESP32

`9:02:15	[D]	[api:102]	
Accepted 192.168.178.111
19:02:15	[W]	[api.connection:107]	
: Reading failed: BAD_INDICATOR errno=11
19:02:15	[W]	[component:170]	
Component api cleared Warning flag
19:02:15	[W]	[component:157]	
Component api set Warning flag: unspecified
19:03:15	[D]	[api:102]	
Accepted 192.168.178.111
19:03:15	[W]	[api.connection:107]	
: Reading failed: BAD_INDICATOR errno=11
19:03:15	[W]	[component:170]	
Component api cleared Warning flag
19:03:15	[W]	[component:157]	
Component api set Warning flag: unspecified`

HA 24.10.1`
ESPHome is the latest from the [https://web.esphome.io/?dashboard_install](url)

do you need more information?

EverythingSmartHome, 2024-10-07:
The error usually means you are trying to connect to a device using an encryption key but no encryption key exists on the device (or the other way around). Check the encryption key you have set on the device vs what you have in the config.

CoolhappyDE, 2024-10-09:
Hey thanks for the fast answer. 
there is a encryption key in my ESPHome YAML, where do i have to set the other key?

EverythingSmartHome, 2024-10-09:
Do you happen to have internet blocked on the device? There is currently an unintentional bug in ESPHome that will cause a loop like that when internet is blocked (it's trying to check for updates from this GitHub)

CoolhappyDE, 2024-10-09:
the internt is not blocked. on first install from your site ([](https://everythingsmarthome.github.io/everything-presence-one/updating.html)) it is working like it should. but after a powercycle it is showing the error code. ESPHome is running on version 2024.7.3 after install. i wil try to install your image again and update to 2024.9.2. i will let you know, if that is working then.

EverythingSmartHome, 2024-10-09:
The log your posting is the exact same as when the internet is unreachable - either from firewall rules or DNS misconfiguration. Please ensure network settings provided over DHCP are correct

CoolhappyDE, 2024-10-10:
Thanks for the DHCP reminder. now i found the problem, there was another local DNS server set that was offline.