EverythingSmartHome/everything-presence-one #228: EP1 ble roome presence

Sivivatu, 2024-11-27:
I'm trying to configure my EP1 for ble tracking, but after updating the firmware with home assistant + ble version, the device doesn't appear as a proxy in my Bermuda BLE configuration.

I suspect I'm just missing something but I cant find any mention on the right way to configure ep1 as a proxy aside from configuring the firmware. I also don't know how I would confirm if the Bluetooth settings have applied correctly, 

If anyone can point me in the right direction I would be most grateful.

EverythingSmartHome, 2024-11-27:
I think you are perhaps misunderstanding what Bluetooth Proxy is? Bluetooth Proxy is for extending your Home Assistant servers bluetooth, not for BLE tracking: https://esphome.io/components/bluetooth_proxy.html

Sivivatu, 2024-11-27:
That is possible, what I'm wanting to do is detect when my phone is in my office with the EP1. My HA server is in the loft. 

What i thought the Proxy would do would detect if my phone was in the room for the tracking?
I also thought the proxy would be detected by [bermuda ](https://github.com/agittins/bermuda) to do the actual locating of the phone (it its in the office). 

Have I misunderstood something here?

EverythingSmartHome, 2024-11-27:
That's called Bluetooth tracking and not Bluetooth proxy. Bluetooth tracking is a feature on ESPHome but you'd need to make your own firmware to add that (be aware that you can get WiFi issues when having this on): https://esphome.io/components/esp32_ble_tracker.html

Sivivatu, 2024-11-27:
Oh ok I have misunderstood the process. Can I use the ep1 for ble tracking or is that not a capability?

EverythingSmartHome, 2024-11-27:
Yes I provided the link to the docs you need above

Sivivatu, 2024-11-27:
Great thank you very much I'll look into that. Appreciate all the help and great work in the sensor and YT channel

closed 2024-11-28 by EverythingSmartHome