EverythingSmartHome/everything-presence-one #230: PIR false positives

golles, 2024-11-29:
After yesterday's video, I wanted to give my EPOs another try, so I reinstalled 1 of them, and I'm still having the PIR sensor giving false positives.

This issue has been discussed before in #64 but that got closed without an actual fix.

Config:
```yaml
substitutions:
  name: epo-logeerkamer
  friendly_name: EPO Logeerkamer
packages:
  Everything Smart Technology.Everything Presence One: github://everythingsmarthome/everything-presence-one/everything-presence-one.yaml@main
esphome:
  name: ${name}
  name_add_mac_suffix: false
  friendly_name: ${friendly_name}
api:
  encryption:
    key: !secret epo_logeerkamer_encryption_key

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  manual_ip:
    static_ip: !secret epo_logeerkamer_static_ip
    gateway: !secret gateway
    subnet: !secret subnet

```

HA history:
[image]

PIR false detections:
- 21:47
- 22:04
- 22:32
- 22:34
- 22:48
- 22:55
- 23:02

When both sensors detected occupancy, it was correct.

golles, 2024-11-30:
I had put the device pointing to the wall (~10 cm) during the night, many false positives.

[image]

EverythingSmartHome, 2024-11-30:
Can you let me know how you power it?

golles, 2024-11-30:
I'm powering it with a `Xiaomi MDY-11-EP`, the 22,5 W max version

golles, 2024-11-30:
I've replaced the PIR from another sensor, for about an hour it was fine, but then it started to report occupancy again.
[image]

I've also set up the other unit, also facing the wall on ~10cm, that keeps on giving false positives on mmWave (first on the small iPhone charger, now on Baseus 65W GaN, CCGP040101)
[image]

EverythingSmartHome, 2024-12-03:
OK if it's facing a wall and still getting detections then please drop us an email through the shop and we can get you sorted

schneuwlym, 2024-12-19:
Hi

I've an EP1 since about a week and I also have "semi" random triggers of the PIR. I write "semi" as it often happens during the same time. As an example last night:
* on	2024-12-19T00:29:48.569Z
* off	2024-12-19T00:30:07.300Z
* on	2024-12-19T02:29:49.972Z
* off	2024-12-19T02:30:16.157Z
* on	2024-12-19T03:29:50.569Z
* off	2024-12-19T03:30:17.255Z
* on	2024-12-19T04:29:51.696Z
* off	2024-12-19T04:30:19.574Z
* on	2024-12-19T05:29:48.631Z
* off	2024-12-19T05:30:02.034Z
* on	2024-12-19T05:30:04.308Z
* off	2024-12-19T05:30:14.418Z

At the same time, there is no event from the mmWave.

I know that our power plant overlays some signals to trigger the boilers to start heating. This is also visible as the light sometimes flickers. Could it be related to this?

phaeton82, 2024-12-19:
It quite funny. I stumbled upon this when looking into an issue with my Panasonic PIR sensors. Which have the same behavior. But they are not in a EP1. I do have an EP1, so I can test if the same happens in that device. 

Some thing I looked into:
Pull down resistor added to esphome yaml. (Didn't work)

A few things on my list to check:

• Use a capacitor for more steady power
• External resistor as a replacement of the internal resistor of the esp board 
• Esphome as the cause, is this behaviour since the last my update of esphome? I'm on 2024.11.2

EverythingSmartHome, 2024-12-19:
Do you have a USB battery bank? If so, could you try and run it on a battery for a while to see if it happens there? Hopefully it should be apparent quite quickly if it's resolved

schneuwlym, 2024-12-19:
> Do you have a USB battery bank? If so, could you try and run it on a battery for a while to see if it happens there? Hopefully it should be apparent quite quickly if it's resolved

Interesting idea! I'll give it a try the next days...

BradleyFord, 2024-12-19:
> ID: <EverythingSmartHome/everything-presence-one/issues/230/2553324519@
> github.com>

my solution is quite simple... I only use the mmWave and ignore the PIR.

But I'm also in the process of switching out my EP1s to an alternative

phaeton82, 2024-12-19:
> Do you have a USB battery bank? If so, could you try and run it on a battery for a while to see if it happens there? Hopefully it should be apparent quite quickly if it's resolved

I'll try!

schneuwlym, 2024-12-20:
> Do you have a USB battery bank? If so, could you try and run it on a battery for a while to see if it happens there? Hopefully it should be apparent quite quickly if it's resolved

:see_no_evil: 
Well, at least for my case, I can forget about the theory with the noisy power supply.  I've powered the sensor with a power bank and it still regularly triggered at half past...

What bothers me is, that it always triggers at half past... Is there anything else I can try?

nivekmai, 2024-12-23:
I had the same issue, tried swapping chargers and reseating the sensor as suggested in #64 , none of it worked, so I just taped over the PIR sensor.

My issue may have been that my sensor is actually looking at another PIR sensor (garage door opener that I can't move), so I expect that could be the cause.

Alexbeard, 2025-01-04:
I have the same issue and mostly at night with some constant interval, only PIR is affected 

[image]
[image]

Pingfragger, 2025-01-04:
Same issue here.
For me at night the PIR detects motion every hour at 1:14, 2:14, 3:14 and so on...

golles, 2025-01-04:
Interesting that that most of you see a constant interval, mine look pretty random on both devices.

schwaboy, 2025-01-14:
I'm also seeing this issue. It's at random, but very frequent intervals. I've run it off a battery. I've moved it around. I've reseated the mmwave, co2, and PIR modules. Nothing seems to change it.

mtf169, 2025-01-23:
Adding to this thread.  I'm seeing the same exact thing on two separate EPOs.  It's baffling and infuriating.  It's hard to tell during the day all the false detections because it may have been valid (they're in my work space).  But throughout the night and times I know nobody is around it's picking up a momentary detection.  One of them last night at 2:32am for 10 seconds.  The other at 9:32pm for 9 seconds, 1:32am for 22 seconds, and 3:33am for 11 seconds.  There's no rhyme or reason for it.

blackgold9, 2025-03-29:
Same issue PIR fires all day, whether the room is occupied or not. MMWave is pretty well behaved

mikensan, 2025-05-01:
Wanted to state I have this issue too (~30m-60m) but I believe my issue is that I have radiant heating. My EPO is in my basement where all my radiant piping is as well - my hot water is stored in an indirect tank and so my educated guess is every hour it's kicking on to bring the temp up (I have my delta set to 5 degrees because my temp is set to 115).

I assume gradual heat rise won't trigger the PIR but sudden rise would, hence my guess. The intervals suggest something that's regularly getting warm rapidly. I wish we could set a "max" heat / delta to activate the PIR. So I could say make it ignore everything over 100 degrees.

popy2k14, 2025-05-11:
Guys, thx for this thread and this https://github.com/EverythingSmartHome/everything-presence-one/issues/64
Was about to order ~10x EPO's for a replacement of Hue PIR + Wemos D1 (ESP8266) + SEN0395.
And searched for PIR false positives!

Sadly they seems to be a thing on the EPO.
Years ago i also experimented with PIR's (cheap ones from aliexpress but also Panasonic ones + ESP) and also had false positive's.
Gave up on making my own and used HUE PIR's since than, which are really stable and has NO false positives at all.

Wanted the EPO's just for BT proxy functionality and the PIR was a nice touch to get rid of my HUE PIR's.
Sadly i will not go that way now.

Rather thinking of staying with my HUE PIR's and replacing my Wemos D1 + SEN0395 with EPL's and replacing the LD2450 with my SEN0395 (which was reliable for me, well tuned and positioned around my appartment).
In my case in some rooms it's neccesary to place the PIR sensor in another position than the mmwave (just because the shape of the room).

Must think about that, if i'll go that route, just to get BT proxy support for my rooms with an hefty price.

So it seems, there is currently no "one sensor to rule them all".

Will keep an eye on this and hope there will be an one sensor to rule them all.

P.S.: Had also Aqara FP1's before the SEN0395 and they was way worse!

golles, 2025-05-11:
@popy2k14 Have you tried the EPL? With those I'm really satisfied, they don't have a PIR, but the mmWave is really good and not slow in detecting occupancy.

popy2k14, 2025-05-11:
@golles not yet, but I will do.

How they are regarding false positives?
Really want to minimize my sensors and get rid of the battery powered hue sensors.

golles, 2025-05-11:
I don't get false positives with my EPL. However they are sensitive, so they can activate on plants, curtains, etc. You might need to configure (masking) zones, which is easy with the everything-presence-mmwave-configurator that can be found here https://github.com/EverythingSmartHome/everything-presence-addons.

The only thing I don't like is that they are not battery-powered, so you are limited on where to put them, or you need to hide cables (on the other hand I also never need to worry about replacing the batteries).

EverythingSmartHome, 2025-05-12:
> [@golles](https://github.com/golles) not yet, but I will do.
> 
> How they are regarding false positives? The SEN0395 and SEN0609 also have some, so that's why I am using the hue PIR's to turn and on and mmwave (SEN0395) to turn the lights off. Really want to minimize my sensors and get rid of the battery powered hue sensors.

To jump in, the SEN0395 and SEN0609 don't have false positives, if you are getting some then it's a tuning or placement issue and the thing they are detecting is legit movement and not false (at least in 99/100 cases).

The PIR false positives we believed to be resolved and we've got replacements sent out to customers to help

popy2k14, 2025-05-12:
@EverythingSmartHome 

Thanks for stepping in.
Will analyse my logs of my SEN0395 further regarding false positives.

What was the technical reason for the PIR false positives (hw engineer here :-) )?
HW defect, other PIR sensor, ...
If i order EPO's now, will they have the false positive PIR issue?

thx

popy2k14, 2025-05-17:
> To jump in, the SEN0395 and SEN0609 don't have false positives, if you are getting some then it's a tuning or placement issue and the thing they are detecting is legit movement and not false (at [...]

Sure, but if a curtain moves i dont want to switch on lights -> hence from a user perspective it's a false positive.
The hue PIR doesnt trigger on curtain movement :-)

@EverythingSmartHome 
Are those EPO PIR false positives fixed?
This holding me back from buying some.

P.S.: Any chance to get the EPO board alone or other EPO part's like PIR?

JT114881, 2025-05-20:
> To jump in, the SEN0395 and SEN0609 don't have false positives, if you are getting some then it's a tuning or placement issue and the thing they are detecting is legit movement and not false (at [...]

I bought 8 of them and I have installed 4 so far and all of them have false positives on the PIR sensor. Is the fix really a replacement?

EverythingSmartHome, 2025-05-20:
When did you purchase them and what is the pattern you are seeing?

JT114881, 2025-05-20:
> When did you purchase them and what is the pattern you are seeing?

A month or so ago. 

[image]

JT114881, 2025-05-20:
I should specify, the activity from 7:56-7:58 is legitimate, 8:09 and 9:09 are false positives

popy2k14, 2025-05-20:
I dont have an EP1 here to meassure the schematic.
Here are a few ideas/points which can leads to false positives:

- power source should be clean (adding 100nF & 100µF) caps near the PIR should do it
- Output (of the Panasonic ones) is an open-drain FET which can just handle 100µF, so an external pull-down is needed, which does not exceed 100µA when the PIR pulls the OUT high.

  Link: https://industrial.panasonic.com/cdbs/www-data/pdf/EWA0000/bltn_eng_ekmc160711_ast-ind-247328.pdf

- Placement of other components which emitting RF noise

   See here is a guy which fixed the false positive by moving the PIR away from the ESP32 Wifi Antenna: https://community.home-assistant.io/t/esphome-pir-false-positives-cheap-esp32/536783/9

- The PIR socket has contact issues to the PIR

@EverythingSmartHome 
Which of these do you have implemented or considered?

@all
Any of you with false positives and electronic skills can check/try those points mentioned?

golles, 2025-06-24:
I'm using a TP-Link deco M9 set of 4 units. This system has no configuration options for channel width/number, these are the current values.

2.4GHz Wireless:
Enable: On
Channel: 11

5GHz-1 Wireless:
Enable: On
Channel: 48

5GHz-2 Wireless:
Enable: On
Channel: 0 (Wireless backhaul)

EverythingSmartHome, 2025-06-30:
Just an update, one customer (not sure if he's in this thread) has been able eliminate all of the hourly detections that were occurring previously after we worked through some WiFi changes. We haven't figured out exactly which setting fixed it, however we think it might have been setting the 2.4GHz power to medium (which most people should generally do anyways) instead of leaving it on Auto/high.

So far since the change, he has has 5 days without false triggers where previously they were occurring every hour. If someone else wants to give this a test that would be great. I'm also going to try to replicate.

EverythingSmartHome, 2025-06-30:
> Which of these settings do we think is the most likely culprit? I am one of the customers that replied to you via email, I am happy to tweak any of these settings and monitor to see if the issue is fixed.
> 
> You are currently using Unifi as well, are you able to reproduce with how you have your network setup?

I have 4 different Unifi sites that I test at regularly and none show this issue, however all are configured with manual settings and none are left on auto which is perhaps the reason I don't see it.

As above, if you want to give the 2.4GHz transmit power a shot that would be great. Will drop you an email just now too

EverythingSmartHome, 2025-07-01:
> Thanks for this update [@EverythingSmartHome](https://github.com/EverythingSmartHome). I've changed my 2.4Ghz transmit power to medium. Will report back with results.

How did you get on so far?

popy2k14, 2025-07-02:
@S-n-d are the false positives just a short time?

@EverythingSmartHome Is there some debouncing in software?
Like just accept pulses longer than "1s" (could be adapted and has to be tested) as motion and ignore all pulses shorter than this time.

popy2k14, 2025-07-02:
@S-n-d i ok, did know that there is an "PIR off latency". Can you set it to "0"?
If you look at the HA logs, are the false positive detection time frames equal to the set "PIR off latency"?

My theory is that there is maybe a missing "debouncing" inj the ESPHome firmware.

What does that mean:
 - short impulse (high and low) from PIR
 - measure time how long it's high
 - is this time too short maybe it's an false positive?

This is a typical "debouncing" of an I/O pin in the world of hardware.
I font know how the signals look like from the PIR to the ESP but i guess when someone detected high, otherwise low.

popy2k14, 2025-07-02:
> We tried this with another customer but some of them were around the 100ms mark which is fine but occasionally you'd get some in excess of 600ms which would add quite a lot of latency to the PIR. Maybe it's still a good option to have in the meantime?

Thanks for the info that you already considered this and the actual delays (pulse durations) we have.
I am hardware engineer for over 20 years now and developed all from RFID readers, mainboards, PCB's with capactive sensors, ...
I think ~500ms is the sweetspot in RFID regarding delay, but for an PIR in my home (and i have a lot's of them - currently HUE) 1000ms would not bother me.

What would bother me are the false positives.

So i think adding an debounce delay of ~700ms for tests with @S-n-d would be good.
If that isn't enough you could increase it in 100ms steps and @S-n-d should test again.

popy2k14, 2025-08-06:
@EverythingSmartHome nice that you included such an debounce delay.

@S-n-d Could you please try this in your setup where the false positives happened?
But when i read you last post (with 16 seconds false positives where 10 seconds are off delay = 6 seconds false positives?), i think it could happen again!?

I think we would know it if you try it.

If it happens again and it's longer than this debounce delay (minus the off delay) than the trigger for the false positive comes from another source (we maybe did'nt have considered yet).

dr0nn1, 2026-09-19:
This started for me also after the latest esp home update.

The first row is the PIR. The second is the mmWave
[image]

EverythingSmartHome, 2026-09-22:
Going to close this one in favour of https://github.com/EverythingSmartHome/everything-presence-one/issues/234 to keep things central and in one thread.

Please see the most recent comment posted there.

closed 2026-09-22 by EverythingSmartHome