EverythingSmartHome/everything-presence-one #231: Bluetooth Proxy - configuration

JulianBastel, 2024-12-03:
What is the yaml config if want to use the everything-presence-one also as Bluetooth Proxy?

Current config is:

`substitutions:
  name: everything-presence-one-707fe8
packages:
  Everything_Smart_Technology.Everything_Presence_One: github://everythingsmarthome/presence-one/everything-presence-one.yaml@main
esphome:
  name: ${name}
  name_add_mac_suffix: false

  
api:
  encryption:
    key: ...

ota:
  - platform: esphome

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password`

EverythingSmartHome, 2024-12-03:
You can just flash the BLE firmware from the user guide here to remain on the official firmware. This contains bluetooth proxy: https://everythingsmarthome.github.io/everything-presence-one/updating.html

Just select the bluetooth option from the menu

JulianBastel, 2024-12-03:
I would like to have a local ESPHome configuration for this device. Is this possible?

EverythingSmartHome, 2024-12-03:
Sure, just adopted it the same as you did with this config after you flash