EverythingSmartHome/everything-presence-one #234: Everyting sensor detects occupancy at 24 minutes after EVERY hour

bkinstle, 2024-12-18:
I received my everything sensor yesterday and installed it in my workshop which is a small building in my backyard.  I set home addistant to turn on the lights when I enter but I noticed the lights on in the morning and looked at the logs.  The sensor detects occupancy at 24 minutes after the hour, every hour.  9:24, 10:24, 11:24 etc.  It can't be reacting to something natural and the building was unoccupied all night.  

I set the sensitivity to zero and set the max range to be the size of the building, Min is 0.5m and max is 6.1m.  Also set the letency to 0.5

HA log shows the yellow bars of detection.  I have a full day of these detections now.
[image]
[image]

EverythingSmartHome, 2024-12-18:
Is it coming from the PIR or the mmwave?

bkinstle, 2024-12-18:
The hourly blips are on
binary_sensor.everything_presence_one_04230c_pir
binary_sensor.everything_presence_one_04230c_occupancy

but binary_sensor.everything_presence_one_04230c_mmwave looks clear
[image]

bkinstle, 2024-12-19:
The only thing with power in the shop was the Mysa thermostat that uses IR to control the mini split and it was right next to the everything sensor.  I removed that and the HVAC remote just to see if it was the cause but the hourly blip still happens.  The main reason for the everythign sensor is actually to turn on the heater if it gets close to the dewpoint so my machines won't rust.  I also tried running the mini split which has some moving vanes on it, and it did no trigger either PIR or mm wave.

bkinstle, 2024-12-19:
Normally the HVAC is off when I'm not using the shop and it only turns on at night when the temperature aproaches the dewpoint.  Last night I did an experiment and ran the heater all night long with the motorized louvers going and the mm sensor didn't see that movement, nor did PIR change it's hourly detection.

[image]

Ultimately that's a good thing in my view because the HVAC won't trigger the sensor, and it eliminates it as a possibility.

I do have a USB battery and will try that later today.

bkinstle, 2024-12-19:
I swapped the sensor to a battery pack and that had no effect on the problem.  But interstingly the problem still happened 24 minutes after the hour, so unplugging the power from the sensor didn't reset any clock (IE if the error happened 1 hour after powering on).  Later I put the battery and the sensor in a cardboard box and sealed it up under my table saw and still got the blip at 3:24pm

[image]

phaeton82, 2024-12-20:
Mine is now 11 hrs without a detection. Since connected to a battery pack and pointed towards a wall.

schneuwlym, 2024-12-20:
I have the same issue, see my comment here: https://github.com/EverythingSmartHome/everything-presence-one/issues/230#issuecomment-2553261106

My PIR triggers always around half past. I also tried with a battery pack, but that didn't change anything.

EverythingSmartHome, 2024-12-20:
Thanks everyone, we are investigating and will feedback any updates soon

EverythingSmartHome, 2024-12-20:
Is anyone here in the ESH Discord? Would be great if we can live troubleshoot this, I would love to customise the firmware a little bit so we can see how long the event lasts for in total

phaeton82, 2024-12-20:
Today will be a bit of an issue. But I'll add the discord. If nobody is available, I'll let you know when I'm online. Our time zones are not far from each other :)

schneuwlym, 2024-12-20:
I've just created an account on discord with the same name. Let me know if I can help.

bkinstle, 2024-12-20:
I'm on the discord server now.  Which channel?  I made a post in the intro board

EverythingSmartHome, 2024-12-22:
I set up 4 brand new EP1s out the box to see if I can reproduce, they've been on test for around 40 hours now but so far haven't had a single false detection and have not yet been able to reproduce unfortuantely. There was one detection since we started but was a legitimate trigger. You can see from the screenshot its had 24 hours plus now with no trigger
[image]

This is making it difficult for me to troubleshoot without being able to reproduce it. I see some of you are in the Discord now so I'll reach out to you soon and we can try and figure out what's going on.

Do we have anyone in the UK with this problem? I could arrange to ship your unit back so we can test it and see if we get the same results here (and ship you another that we've verified doesn't have the issue in our studio)

bkinstle, 2024-12-23:
OK great.  I'm in California.

Are there any sensitivity settings you'd like us to try out in the meantime?

EverythingSmartHome, 2024-12-23:
I can think of one potential way to resolve it but would need to have a unit to try it on or try some live troubleshooting with someone who has one

AppelBoomHD, 2024-12-25:
I have the same problem on the 50th minute of every hour. I'm from the Netherlands though.

kgleeson, 2025-01-04:
I have this at the :54 minute every hour on two units and :33 on another.
I took the co2 sensor out of the first one today and the issues have stopped. I'm still using the same config too.
I have a 4th that doesn't have a co2 sensor and that's always worked perfect.
Mine are from November 2023.

Voteax, 2025-01-24:
Same Problem here. PIR Sensor Detects Motion almost every :06
Only on one EP1 from 5.

FHTheron, 2025-02-23:
I have concluded that these specific PIR sensors are just not good in this specific setup and need a serious amount of filtering to make them usable. Maybe a bad batch, maybe somehow mmwave interference.

I have tried the following over at least a day for each test. Everything was done using a battery power bank.
* Removing the PIR sensor and wiring the sense pin directly to the Vdd pin. No glitches, so it's not the ESP power supply to the sensor that is glitching.
* Without the sensor. No glitches.
* Following https://industrial.panasonic.com/cdbs/www-data/pdf/EWA0000/ast-ind-250615.pdf
  * Q2-16: Soldering some capacitors across the power supply pins
* For fun, putting those same capacitors across the sense pin and GND. This results in an RC style filter curve, but does not really fix anything, so this one was not run for more than an hour.
* I had the PIR sense pin, while plugged into the PCB, on a handheld oscilloscope for half a day to try to capture one of these glitches, but couldn't. The scope showed random massive spikes that should not be possible and these seemed to happen on certain scope settings only, so I wrote them off as "cheap" oscilloscope issues.

I then opted to load a custom ESPhome config where the PIR input is reported immediately. With this I could capture the glitches on their odd mostly hourly schedule. The pattern of each glitch looked the same as a momentary movement detection, so if you were to try filtering this out in software you would need to ignore at lest 3 seconds of detection, which makes it too slow. This was also done using a big fat battery power bank.

Events over about a day, with real movement towards the end, after turning the EP1 around to stop facing the corner of the small bathroom that had zero movement:
[image]

Zooming in on two of the glitches:
[image]
[image]

I have not yet tried to power and monitor the PIR sensor outside of the EP1. I disabled them and for one of the EP1s I used that input pin for a door magnet sensor instead.

EverythingSmartHome, 2025-02-24:
To give an update, the PIR itself is not the issue here (as we've been using 2.5 years without this issue). We did investigating using the PIR on delay several weeks ago but found it was unusable for the reasons mentioned.

We had a couple of users units shipped back to us with the issue a few weeks back but under the oscilloscope we weren't able to really re-produce it (when not connected to HA) unfortunately and results weren't really conclusive. Inside a box, there were no false detections.

We think this may be something in the manufacturing chain for this batch, since they all seem to stem from this new batch. We think we may have found the problem (not 100% confirmed yet) and we hope to receive a new batch soon and we can test the fix and then get in touch with affected users if it is indeed something in the batch

golles, 2025-02-24:
@EverythingSmartHome do you think this is the same issue as #230? In that case, should we close it and just keep this one? as it has more info

KentuckyMC, 2025-03-15:
Any news on this issue? Is it because of the false batch?

[image]

EverythingSmartHome, 2025-03-27:
> Hi [@EverythingSmartHome](https://github.com/EverythingSmartHome) is this also for users with random time pattern triggers, eg. [#230](https://github.com/EverythingSmartHome/everything-presence-one/issues/230) ?

Drop us a message, we can take a look!

EverythingSmartHome, 2026-09-22:
Hi all, we've made some real progress on this one so wanted to share where we've got to.

Over the last few days we set up a proper test rig for it: an EP1 running a diagnostic firmware that reports the raw PIR pin with no latency filters, logs every pulse with its duration, and has the Wi-Fi driver's debug output turned on so rekeys, beacon loss and reconnects all land in the same log as the PIR. We put it on its own Unifi SSID on a single AP so we could change settings without worrying about anything else, sat it facing a wall, and scored everything automatically from the log. Partway through we also dug out one of the units that had come back to us with the hourly PIR problem, and ran the same tests on that.

### **What we tried first**

We started with the theories from these threads. Group key rekey was the big one, so the test SSID had rekey set to every 60 seconds instead of the default hour, and we watched every handshake in the driver log while counting PIR pulses. Then we went through the rest of the WLAN settings one at a time, four minutes each: rekey off, rekey at 3600, PMF required, WPA3 transition, 802.11r fast roaming, 802.11v off, UAPSD, DTIM 3, minimum data rate control off, multicast enhancement, hidden SSID, client isolation and proxy ARP. After that we did the AP radio itself: transmit power high, auto and low, channel auto, channel auto with power auto (the combination a few of you mentioned), and 40 MHz width.

Every single one of those came back with zero PIR pulses. Dozens of rekeys, dozens of clean reconnects, nothing.

### **What actually did it**

The only time the PIR fired during that whole sweep was during two of the AP radio changes, where the AP's radio went quiet for a few seconds while it reconfigured, and both times the PIR gave a burst of short pulses with nobody in the room. So we made that repeatable: flipping the AP's 2.4 GHz channel back and forth forces the EP1 to lose beacons for about 8 seconds each time, and we ran that 30 times per configuration.

That turned out to be a very reliable trigger. On the bench unit it fired 15 times out of 15 when the EP1 was on channel 1, and on the unit that came back from a customer 9 out of 15. Interestingly it never fired once when the EP1 was on channel 6 or 11, on either unit. The pulse arrives 3 to 6 seconds after the last beacon, before the Wi-Fi driver has even given up on the link.

What's going on is that when the AP stops beaconing, the ESP32 keeps retrying frames at full power with nothing acknowledging them. That's a much bigger RF burst than anything in normal operation, and it's coming from a few centimetres away from a very sensitive PIR amplifier. On some units, and on some channels, that's enough to register as motion. The channel dependence points at a resonance somewhere around the PIR header that will vary from unit to unit, which would explain why one unit in a house does it and the one next to it doesn't.

We also checked that it wasn't power related by running the same 30 flips with Wi-Fi power save turned off, and got exactly the same result. And for anyone wondering why you see it in Home Assistant when the raw pulse is under two seconds: the 10 second PIR off latency stretches it, so it comfortably outlasts the outage.

### **What looks like it helps**

We then tried turning the transmit power down. On the returned unit that took it from 9 hits out of 15 down to 1 at a moderate reduction and 0 at a slightly larger one; on the bench unit from 15 out of 15 to 0. Nothing else we tried moved the numbers at all. It's only the EP1's uplink that gets weaker and the EP1 sends very little, so in most installs it shouldn't be noticeable, but it's not something we'd push to everyone. If it holds up it would be offered as a separate firmware for people who are affected.

### **Testers wanted**

So, if you're seeing the hourly PIR pattern and would be willing to run a modified firmware for a few days to see if it goes away on your setup, please reply here or drop us an email and we'll get a build to you. Two other things would help a lot: which 2.4 GHz channel your access point is on, and whether your hourly PIR event lines up with a brief gap in the EP1's other sensor updates in your HA history (illuminance updates every 2 seconds, so a gap is easy to spot).

Thanks to everyone who's posted logs, timings and hardware notes in these threads, it genuinely made this a lot quicker to narrow down.

dvanp, 2026-09-22:
I am up for testing the modified firmware! 
my AP is currently on channel 1 and one of my units is having the issue more than hourly somehow.

EverythingSmartHome, 2026-09-22:
> I am up for testing the modified firmware! my AP is currently on channel 1 and one of my units is having the issue more than hourly somehow.

Thanks and interesting to know it's also channel 1. Can you let me know your board version and sensor (SEN0395 or SEN0609)? Easiest way is if you go to Developer tools > Actions and search for and run the "get build flags" service and paste the output here.

dvanp, 2026-09-22:
> Thanks and interesting to know it's also channel 1. Can you let me know your board version and sensor (SEN0395 or SEN0609)? Easiest way is if you go to Developer tools > Actions and search for and run the "get build flags" service and paste the output here.

```
ethernet_enabled: false
co2_enabled: false
bluetooth_enabled: true
board_revision: '1.5'
sensor_variant: sen0609
firmware_channel: stable
model: everything-presence-one
```

EverythingSmartHome, 2026-09-22:
@dvanp Do you have Discord? I am trying to attach the firmware here but won't let me due to file type prohibited. Alternatively drop me an email and I can send there

Yekimek, 2026-09-22:
> requests on the go with GitHub Mobile for iOS
> and Android
> Download it today!

Hi,

I’d be very happy to help test the modified firmware and work with you on
tracking this down.

I actually have two EP1 units affected by the hourly false PIR issue, and
interestingly they are different board/sensor versions and are connected to
different UniFi APs.

EP1.2 Hallway:

   - Board revision: 1.5
   - Sensor variant: SEN0609
   - CO2: disabled
   - UniFi AC Pro South
   - 2.4 GHz channel 11 / 20 MHz
   - False PIR normally occurs around XX:08

Build flags:

ethernet_enabled: false
co2_enabled: false
bluetooth_enabled: false
board_revision: ‘1.5’
sensor_variant: sen0609
firmware_channel: stable
model: everything-presence-one

EP1.3 Closet:

   - Board revision: 1.3
   - Sensor variant: SEN0395
   - CO2: enabled
   - UniFi AC Pro North
   - 2.4 GHz channel 6 / 20 MHz
   - False PIR normally occurs around XX:24/25

Build flags:

ethernet_enabled: false
co2_enabled: true
bluetooth_enabled: false
board_revision: ‘1.3’
sensor_variant: sen0395
firmware_channel: stable
model: everything-presence-one

Some recent EP1.3 false PIR events occurred at:

14:25:17
15:25:09
23:25:08

They lasted approximately 11–12 seconds in Home Assistant, with PIR off
latency configured to 10 seconds.

I also found something that may be relevant. UniFi shows both affected EP1
units disconnecting and reconnecting roughly every 15 minutes.

EP1.2 reconnects around XX:07 and the false PIR generally occurs around
XX:08.

EP1.3 reconnects around XX:24 and the false PIR generally occurs around
XX:25.

EP1.3 reconnects to the same AP rather than roaming, while EP1.2 has also
moved between the North and South APs.

I have now set up additional Home Assistant logging to record the PIR start
time, duration, illuminance value and time since the last illuminance
report for every event.

I’d be very happy to test the modified firmware on either or both units and
send you the results and logs.

Please feel free to send the firmware by email.

Thanks again for investigating this.

bkinstle, 2026-09-22:
> Hi,
>
> I’d be very happy to help test the modified firmware and work with you on
> tracking this down.
>
> I actually have two EP1 units affected by the hourly false PIR issue, and
> interestingly they are [...]

Are you by chance running any U7 AP's from unifi?  If so there's a nasty bug that causes devices to hop on and off the networkd due an issue with how they forward DHCP.  If you are running a U7, try updating to the early access firmware 8.8.8 or 8.8.9 and see if that helps.

Yekimek, 2026-09-22:
Thanks for checking. I’m not using any U7 APs. Both of my access points are UniFi UAP-AC-PRO units running firmware 6.8.2, so that particular U7 DHCP issue shouldn’t apply in my case.

jdiviney, 2026-09-22:
Happy to help test the diagnostic firmware. My setup:
Firmware: 1.2.17 (ESPHome 2026.9.0),
AP: UniFi U7 Pro, 2.4 GHz, channel 11, 20 MHz width, signal -47/-48 dBm (strong).
Pulled 3 days of history on this unit (in the kitchen): 108 isolated PIR "on" events with 5+ min of quiet either side — 107 of those 108 had the mmWave sensor showing zero presence at the same time, so it's not real motion.
During quiet overnight hours these cluster at a rolling ~57–65 min interval rather than a fixed clock minute (e.g. not always :XX24) — the interval drifts night to night. 
Noting for the channel data point: matches @Yekimek's ch. 11 unit, differs from @dvanp's ch. 1.

EverythingSmartHome, 2026-09-23:
Thanks everyone, if you can send me an email or message on discord and I can get firmware over to test

Yekimek, 2026-09-24:
Hi Lewis, I sent you a DM on Discord. I’d be very happy to test the diagnostic firmware on my two affected EP1 units if you’re still looking for testers. Thanks!

frodehf, 2026-09-28:
I also experience this issue:

[image]

Let me know if i can help with the troubleshooting.

marcuscps, 2026-11-04:
I have a similar issue. My sensor keeps flipping between detected and clear non stop.
Did you get to a conclusion about it?

[image]

[image]

keirans, 2026-11-05:
Same problem here, happy to help test.

**Setup**

Build flags from get_build_flags:

```
ethernet_enabled: false
co2_enabled: false
bluetooth_enabled: false
board_revision: "1.6"
sensor_variant: sen0609
firmware_channel: stable
model: everything-presence-one
```

It was on 1.2.10 when I noticed this, and I've just updated to 1.2.17 (ESPHome 2026.10.0). 

 Network is UniFi, AP channel:  1

**What I'm seeing**

The PIR fires once an hour, 24/7, at roughly xx:07:20. I only noticed because it was switching my living room lights on.

Things I've ruled out

- Nothing in HA runs at :07: no time_pattern automations, and nothing else in the logbook changes just before the trigger.
- The on-board temperature and illuminance don't move through the event 
- I have a camera covering the same room. Nothing visibly happens at :07, and the camera's own motion detection only lined up with 1 of 6 overnight events, which is just its normal random blips.
- The event kept its phase across the brief API reconnects HA logged (30 Oct, 3 Nov, 4 Nov), but I can't say for.

**Testing**

I'd be happy to run the diagnostic or reduced-TX-power firmware on this unit. 

Let me know the best way to get the firmware (email or Discord) and I'll sort it out.

I have 2 more unopened units in boxes that I bought at the same time that i'll guess will be impacted, so keen to help get this resolved.

K

pfmacedo, 2026-11-05:
I was seeing this same issue, although not on a fixed time interval (for what is worth, my wifi setup is using TPLink Omada kit). I've tried a few things based on this issue report and what fully fixed for me was limiting the wifi power to 10dB via the build config yaml - my three EP1s that had random activations have had no random events for the last three days since I enabled this change.

marcuscps, 2026-11-14:
> I have a similar issue. My sensor keeps flipping between detected and clear non stop. Did you get to a conclusion about it?
> 
> <img alt="Image" width="851" height="76" src="https://private-user-images.githubusercontent.com/5632324/695320370-57ca12bd-f8ea-449f-bdd6-89c11fd0cff6.png?jwt=eyJ0eXAiOiJKV1QiLCJhbGciOiJIUzI1NiJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3OTQ2OTczNTQsIm5iZiI6MTc5NDY5NzA1NCwicGF0aCI6Ii81NjMyMzI0LzY5NTMyMDM3MC01N2NhMTJiZC1mOGVhLTQ0OWYtYmRkNi04OWMxMWZkMGNmZjYucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI2MTExNCUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNjExMTRUMjI1NzM0WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9NjczNWI5ZTFlYjU4OGNlZDRkMjIwYTc0NDhjNmQwOTlkZTU0N2Y4ODdjZWNhYThkMmJhMDE1ZGZmZTAwNGY2OSZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmcmVzcG9uc2UtY29udGVudC10eXBlPWltYWdlJTJGcG5nIn0.NNQnuyxrRg_Ongr38iyRWlBy8eUj11QHW_GeAOMNUm4"> <img alt="Image" width="848" height="74" src="https://private-user-images.githubusercontent.com/5632324/695319567-ba0c1d22-13a9-4b81-88b3-2e8e499cbe22.png?jwt=eyJ0eXAiOiJKV1QiLCJhbGciOiJIUzI1NiJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3OTQ2OTczNTQsIm5iZiI6MTc5NDY5NzA1NCwicGF0aCI6Ii81NjMyMzI0LzY5NTMxOTU2Ny1iYTBjMWQyMi0xM2E5LTRiODEtODhiMy0yZThlNDk5Y2JlMjIucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI2MTExNCUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNjExMTRUMjI1NzM0WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9NWMyYzY2NjlhODc0NTVlNWZhMTFiNmViNzA3YTkwZDRhMWE2MTkyZmRhODE0NTc2NTVmYzVmMWI1ZGIzNzg2NCZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmcmVzcG9uc2UtY29udGVudC10eXBlPWltYWdlJTJGcG5nIn0.WC0Yt5KupEAx-6RaW7yH8UoVLZ4LG3nIPyCMwPs-pTU">

@EverythingSmartHome 
Do you have any conclusions about this issue?