EverythingSmartHome/everything-presence-one #235: Occupancy not updating state

Snuggle, 2024-12-20:
Hello there!

I am having this issue with my brand new Everything Presence One where the mmWave sensor and/or the PIR sensor are 'Detected' but the occupancy is 'Clear'.

This shouldn't be possible as far as I understand? What's the cause of this?

I'm running on a Unifi Wi-Fi network with a Unifi AP 7 Pro using 'Enhanced-IoT Mode'.

Clicking entities such as enabling 'ESP32 status LED' are immediately responsive, so the sensor hasn't locked up or anything. The mmWave and PIR are still updating their state properly as far as I can tell, it's just the Occupancy that isn't updating.

<details>

<summary>Debug Information</summary>

```
 <h1> Current entity states </h1> <ha-expansion-panel outlined="">    </ha-expansion-panel> <div class="table-wrapper"> 
Entity | State | Attributes
-- | -- | --

binary_sensor.everything_presence_one_0441cc_mmwave     Living Room EPO-0441cc mmWave | on | device_class: occupancy friendly_name: Living Room EPO-0441cc mmWave
binary_sensor.everything_presence_one_0441cc_occupancy     Living Room EPO-0441cc Occupancy | off | device_class: occupancy friendly_name: Living Room EPO-0441cc Occupancy
binary_sensor.everything_presence_one_0441cc_pir     Living Room EPO-0441cc PIR | off | device_class: motion friendly_name: Living Room EPO-0441cc PIR
button.everything_presence_one_0441cc_set_distance     Living Room EPO-0441cc Set Distance | unknown | friendly_name: Living Room EPO-0441cc Set Distance
button.everything_presence_one_0441cc_set_sensitivity     Living Room EPO-0441cc Set Sensitivity | unknown | friendly_name: Living Room EPO-0441cc Set Sensitivity
light.everything_presence_one_0441cc_esp32_status_led     Living Room EPO-0441cc ESP32 status LED | off | supported_color_modes: onoff color_mode: null friendly_name: Living Room EPO-0441cc ESP32 status LED supported_features: 8
light.everything_presence_one_0441cc_mmwave_led     Living Room EPO-0441cc mmWave LED | off | supported_color_modes: onoff color_mode: null friendly_name: Living Room EPO-0441cc mmWave LED supported_features: 8
number.everything_presence_one_0441cc_humidity_offset     Living Room EPO-0441cc Humidity Offset | 0.0 | min: -50 max: 50 step: 0.1 mode: box unit_of_measurement: % icon: mdi:water-percent friendly_name: Living Room EPO-0441cc Humidity Offset
number.everything_presence_one_0441cc_illuminance_offset     Living Room EPO-0441cc Illuminance Offset | 0.0 | min: -50 max: 50 step: 1 mode: box unit_of_measurement: lx icon: mdi:brightness-5 friendly_name: Living Room EPO-0441cc Illuminance Offset
number.everything_presence_one_0441cc_mmwave_max_distance     Living Room EPO-0441cc mmWave Max Distance | 12.0 | min: 0 max: 25 step: 0.1 mode: slider unit_of_measurement: m icon: mdi:arrow-left-right friendly_name: Living Room EPO-0441cc mmWave Max Distance
number.everything_presence_one_0441cc_mmwave_minimum_distance     Living Room EPO-0441cc mmWave Minimum Distance | 1.0 | min: 0 max: 25 step: 0.1 mode: slider unit_of_measurement: m icon: mdi:arrow-left-right friendly_name: Living Room EPO-0441cc mmWave Minimum Distance
number.everything_presence_one_0441cc_mmwave_off_latency     Living Room EPO-0441cc mmWave off latency | 15.0 | min: 1 max: 600 step: 5 mode: slider unit_of_measurement: seconds icon: mdi:clock-end friendly_name: Living Room EPO-0441cc mmWave off latency
number.everything_presence_one_0441cc_mmwave_on_latency     Living Room EPO-0441cc mmWave on latency | 0.0 | min: 0 max: 2 step: 0.25 mode: slider unit_of_measurement: seconds icon: mdi:clock-start friendly_name: Living Room EPO-0441cc mmWave on latency
number.everything_presence_one_0441cc_mmwave_range_reduced     Living Room EPO-0441cc mmWave Range Reduced | 0.8 | min: 0 max: 25 step: 0.1 mode: slider unit_of_measurement: m icon: mdi:arrow-left-right friendly_name: Living Room EPO-0441cc mmWave Range Reduced
number.everything_presence_one_0441cc_mmwave_sustain_sensitivity     Living Room EPO-0441cc mmWave Sustain Sensitivity | 7.0 | min: 0 max: 9 step: 1 mode: auto icon: mdi:target-variant friendly_name: Living Room EPO-0441cc mmWave Sustain Sensitivity
number.everything_presence_one_0441cc_mmwave_trigger_sensitivity     Living Room EPO-0441cc mmWave Trigger Sensitivity | 5.0 | min: 0 max: 9 step: 1 mode: auto icon: mdi:target-variant friendly_name: Living Room EPO-0441cc mmWave Trigger Sensitivity
number.everything_presence_one_0441cc_occupancy_off_latency     Living Room EPO-0441cc Occupancy off latency | 15.0 | min: 1 max: 600 step: 5 mode: slider unit_of_measurement: seconds icon: mdi:clock-end friendly_name: Living Room EPO-0441cc Occupancy off latency
number.everything_presence_one_0441cc_pir_off_latency     Living Room EPO-0441cc PIR off latency | 10.0 | min: 1 max: 120 step: 1 mode: slider unit_of_measurement: seconds icon: mdi:clock-end friendly_name: Living Room EPO-0441cc PIR off latency
number.everything_presence_one_0441cc_temperature_offset     Living Room EPO-0441cc Temperature Offset | 0.0 | min: -20 max: 20 step: 0.1 mode: box unit_of_measurement: °C icon: mdi:thermometer friendly_name: Living Room EPO-0441cc Temperature Offset
sensor.everything_presence_one_0441cc_humidity     Living Room EPO-0441cc Humidity | 36.8 | state_class: measurement unit_of_measurement: % device_class: humidity friendly_name: Living Room EPO-0441cc Humidity
sensor.everything_presence_one_0441cc_illuminance     Living Room EPO-0441cc Illuminance | 31.1 | state_class: measurement unit_of_measurement: lx device_class: illuminance friendly_name: Living Room EPO-0441cc Illuminance
sensor.everything_presence_one_0441cc_temperature     Living Room EPO-0441cc Temperature | 21.8 | state_class: measurement unit_of_measurement: °C device_class: temperature friendly_name: Living Room EPO-0441cc Temperature

 </div> Current entity states
Entity 	State 	Attributes
	
	
[binary_sensor.everything_presence_one_0441cc_mmwave](https://home.snugg.ie/developer-tools/state#)
Living Room EPO-0441cc mmWave
	on 	device_class: occupancy
friendly_name: Living Room EPO-0441cc mmWave
[binary_sensor.everything_presence_one_0441cc_occupancy](https://home.snugg.ie/developer-tools/state#)
Living Room EPO-0441cc Occupancy
	off 	device_class: occupancy
friendly_name: Living Room EPO-0441cc Occupancy
[binary_sensor.everything_presence_one_0441cc_pir](https://home.snugg.ie/developer-tools/state#)
Living Room EPO-0441cc PIR
	off 	device_class: motion
friendly_name: Living Room EPO-0441cc PIR
[button.everything_presence_one_0441cc_set_distance](https://home.snugg.ie/developer-tools/state#)
Living Room EPO-0441cc Set Distance
	unknown 	friendly_name: Living Room EPO-0441cc Set Distance
[button.everything_presence_one_0441cc_set_sensitivity](https://home.snugg.ie/developer-tools/state#)
Living Room EPO-0441cc Set Sensitivity
	unknown 	friendly_name: Living Room EPO-0441cc Set Sensitivity
[light.everything_presence_one_0441cc_esp32_status_led](https://home.snugg.ie/developer-tools/state#)
Living Room EPO-0441cc ESP32 status LED
	off 	supported_color_modes: onoff
color_mode: null
friendly_name: Living Room EPO-0441cc ESP32 status LED
supported_features: 8
[... 112 lines cut ...]
[sensor.everything_presence_one_0441cc_humidity](https://home.snugg.ie/developer-tools/state#)
Living Room EPO-0441cc Humidity
	36.8 	state_class: measurement
unit_of_measurement: %
device_class: humidity
friendly_name: Living Room EPO-0441cc Humidity
[sensor.everything_presence_one_0441cc_illuminance](https://home.snugg.ie/developer-tools/state#)
Living Room EPO-0441cc Illuminance
	31.1 	state_class: measurement
unit_of_measurement: lx
device_class: illuminance
friendly_name: Living Room EPO-0441cc Illuminance
[sensor.everything_presence_one_0441cc_temperature](https://home.snugg.ie/developer-tools/state#)
Living Room EPO-0441cc Temperature
	21.8 	state_class: measurement
unit_of_measurement: °C
device_class: temperature
friendly_name: Living Room EPO-0441cc Temperature
``` 
</details>

[image]
[image]
[image]
[image]

Snuggle, 2024-12-20:
Restarting the unit by unplugging it & replugging it back in seems to fix the issue, but I'd like to prevent it happening again.

EverythingSmartHome, 2024-12-20:
Hey, yeah that shouldn't really happen and I've not seen it reported before. Let me take a look at the code and see if I can spot anything.

(p.s I edited your comment to remove mention of your HA URL)

Dinth, 2024-12-27:
Hiya. Im not sure if that's related by my EP Lite stop detecting occupancy randomly and only restarting them brings the occupancy detection back.

When i connect to the sensor while its not detecting occupancy im seeing this in logs:
```
INFO ESPHome 2024.12.2
INFO Reading configuration /config/esphome/everything-presence-lite-2117a4.yaml...
INFO Updating https://github.com/everythingsmarthome/everything-presence-lite.git@main
INFO Starting log output from 10.10.10.141 using esphome API
INFO Successfully connected to everything-presence-lite-2117a4 @ 10.10.10.141 in 0.093s
INFO Successful handshake with everything-presence-lite-2117a4 @ 10.10.10.141 in 0.112s
[18:56:20][I][app:100]: ESPHome version 2024.12.2 compiled on Dec 22 2024, 08:44:39
[18:56:20][I][app:102]: Project EverythingSmartTechnology.Everything Presence Lite version 1.2.0
[18:56:20][C][status_led:065]: Status Led Light:
[18:56:20][C][status_led:066]:   Pin: GPIO14
[18:56:20][C][wifi:600]: WiFi:
[18:56:20][C][wifi:428]:   Local MAC: 08:D1:F9:21:17:A4
[18:56:20][C][wifi:433]:   SSID: [redacted]
[18:56:20][C][wifi:436]:   IP Address: 10.10.10.141
[18:56:20][C][wifi:439]:   BSSID: [redacted]
[18:56:20][C][wifi:441]:   Hostname: 'everything-presence-lite-2117a4'
[18:56:20][C][wifi:443]:   Signal strength: -50 dB ▂▄▆█
[18:56:20][C][wifi:447]:   Channel: 1
[18:56:20][C][wifi:448]:   Subnet: 255.255.0.0
[18:56:20][C][wifi:449]:   Gateway: 10.10.0.1
[18:56:20][C][wifi:450]:   DNS1: 10.10.0.1
[18:56:20][C][wifi:451]:   DNS2: 0.0.0.0
[18:56:20][C][logger:185]: Logger:
[18:56:20][C][logger:186]:   Level: DEBUG
[18:56:20][C][logger:188]:   Log Baud Rate: 115200
[18:56:20][C][logger:189]:   Hardware UART: UART0
[18:56:20][C][i2c.idf:075]: I2C Bus:
[18:56:20][C][i2c.idf:076]:   SDA Pin: GPIO21
[18:56:20][C][i2c.idf:077]:   SCL Pin: GPIO22
[18:56:20][C][i2c.idf:078]:   Frequency: 50000 Hz
[18:56:20][C][i2c.idf:084]:   Recovery: bus successfully recovered
[18:56:20][I][i2c.idf:094]: Results from i2c bus scan:
[18:56:20][I][i2c.idf:100]: Found i2c device at address 0x23
[18:56:20][C][uart.idf:159]: UART Bus 1:
[18:56:20][C][uart.idf:160]:   TX Pin: GPIO17
[18:56:20][C][uart.idf:161]:   RX Pin: GPIO16
[18:56:20][C][uart.idf:163]:   RX Buffer Size: 256
[18:56:20][C][uart.idf:165]:   Baud Rate: 256000 baud
[18:56:20][C][uart.idf:166]:   Data Bits: 8
[18:56:20][C][uart.idf:167]:   Parity: NONE
[18:56:20][C][uart.idf:168]:   Stop bits: 1
[18:56:20][C][template.number:050]: Template Number 'Illuminance Offset'
[18:56:20][C][template.number:050]:   Icon: 'mdi:brightness-5'
[18:56:20][C][template.number:050]:   Unit of Measurement: 'lx'
[18:56:20][C][template.number:051]:   Optimistic: YES
[18:56:20][C][template.number:052]:   Update Interval: never
[18:56:20][C][template.number:050]: Template Number 'Occupancy Off Delay'
[18:56:20][C][template.number:050]:   Unit of Measurement: 's'
[18:56:20][C][template.number:051]:   Optimistic: YES
[18:56:20][C][template.number:052]:   Update Interval: 60.0s
[18:56:20][C][template.number:050]: Template Number 'Max Distance'
[18:56:20][C][template.number:050]:   Unit of Measurement: 'cm'
[18:56:20][C][template.number:051]:   Optimistic: YES
[18:56:20][C][template.number:052]:   Update Interval: 60.0s
[18:56:20][C][template.number:050]: Template Number 'Installation Angle'
[18:56:20][C][template.number:050]:   Icon: 'mdi:angle-acute'
[18:56:20][C][template.number:050]:   Unit of Measurement: 'º'
[18:56:20][C][template.number:051]:   Optimistic: YES
[18:56:20][C][template.number:052]:   Update Interval: never
[18:56:20][C][template.number:050]: Template Number 'Zone 1 Begin X'
[18:56:20][C][template.number:050]:   Unit of Measurement: 'mm'
[18:56:20][C][template.number:051]:   Optimistic: YES
[18:56:20][C][template.number:052]:   Update Interval: 60.0s
[18:56:20][C][template.number:050]: Template Number 'Zone 1 End X'
[18:56:20][C][template.number:050]:   Unit of Measurement: 'mm'
[18:56:20][C][template.number:051]:   Optimistic: YES
[... 847 lines cut ...]
[19:04:25][D][sensor:093]: 'Illuminance': Sending state 1.35827 lx with 1 decimals of accuracy
[19:04:26][I][esp32_ble_client:084]: [0] [C6:12:2D:4C:FA:9B] Disconnecting.
[19:04:27][D][bh1750.sensor:159]: 'Illuminance': Got illuminance=1.4lx
[19:04:27][D][sensor:093]: 'Illuminance': Sending state 1.35827 lx with 1 decimals of accuracy
[19:04:29][D][bh1750.sensor:159]: 'Illuminance': Got illuminance=1.4lx
[19:04:29][D][sensor:093]: 'Illuminance': Sending state 1.35827 lx with 1 decimals of accuracy
[19:04:31][D][bh1750.sensor:159]: 'Illuminance': Got illuminance=1.4lx
[19:04:31][D][sensor:093]: 'Illuminance': Sending state 1.35827 lx with 1 decimals of accuracy
[19:04:33][D][bh1750.sensor:159]: 'Illuminance': Got illuminance=1.4lx
[19:04:33][D][sensor:093]: 'Illuminance': Sending state 1.35827 lx with 1 decimals of accuracy
[19:04:35][D][bh1750.sensor:159]: 'Illuminance': Got illuminance=1.4lx
[19:04:35][D][sensor:093]: 'Illuminance': Sending state 1.35827 lx with 1 decimals of accuracy
[19:04:36][D][esp-idf:000][BTU_TASK]: W (35736411) BT_APPL: gattc_conn_cb: if=4 st=0 id=4 rsn=0x100

[19:04:36][D][esp-idf:000][BTU_TASK]: W (35736419) BT_APPL: gattc_conn_cb: if=5 st=0 id=5 rsn=0x100

[19:04:36][D][esp-idf:000][BTU_TASK]: W (35736423) BT_APPL: gattc_conn_cb: if=6 st=0 id=6 rsn=0x100

[19:04:36][D][esp32_ble_client:171]: [0] [C6:12:2D:4C:FA:9B] ESP_GATTC_DISCONNECT_EVT, reason 256
[19:04:36][D][esp32_ble_tracker:273]: Starting scan...
[19:04:37][D][bh1750.sensor:159]: 'Illuminance': Got illuminance=1.4lx
[19:04:37][D][sensor:093]: 'Illuminance': Sending state 1.35827 lx with 1 decimals of accuracy
[19:04:39][D][bh1750.sensor:159]: 'Illuminance': Got illuminance=1.4lx
[19:04:39][D][sensor:093]: 'Illuminance': Sending state 1.35827 lx with 1 decimals of accuracy
```

timota, 2024-12-29:
Similar issue here, occupancy does not change(reflect actual state) its state, eg: mmWave and PIR states are clear but occupancy still detected.

EverythingSmartHome, 2024-12-29:
@Dinth This is the EP1 repo, please post over in the EPL repo: https://github.com/EverythingSmartHome/everything-presence-lite

EverythingSmartHome, 2024-12-29:
@timota @Snuggle I actually wonder if this is not a code issue but rather caused by Bluetooth Proxy. Can you try flashing the non-ble firmware and see if that resolves?

timota, 2025-01-03:
I have updated firmware, as I updated ESPhome compiler version to the latest one. Due to compiler issue, had to change esp-idf version:
```yaml
esp32:  
  board: esp32dev
  framework:
    type: esp-idf
    version: 5.2.2
    platform_version: 6.7.0
```

Since that time, I dont see this issue anymore. Will monitor and let you know.

Gigoo25, 2025-02-18:
Also seeing similar issues with both of my EP1s. Both have BLE variant of latest stable firmware.

EverythingSmartHome, 2025-02-24:
We had a similar thing on the Lite (but worse as more sensors and more frequent updating) which is because of this bug in ESPHome: https://github.com/esphome/esphome/pull/7547

Including this PR which hasn't been merged yet into the build seems to resolve it. I can apply this same fix to the EP1 builds for now until it gets merged into ESPHome.

FDHop, 2025-02-24:
Thanks Lewis, thanks for the link and great news it's being seen to by the ESPHome team. Would you be applying the temp fix into the next beta EP1 build or straight into the next stable one? 
(for reference to anyone also experiencing this issue, my EP1's are running Stable build 1.2.6 and ESPHome 2025.2.0)

EverythingSmartHome, 2025-02-24:
https://github.com/EverythingSmartHome/everything-presence-one/pull/243

Update to 1.2.7 and hopefully should fix

Just to clarify that the PR is for a bug/improvement in ESPHome itself and not specific to the EP1/Lite.

FDHop, 2025-02-25:
Cheers Lewis, appreciate how quick you got that fix out! So far the new update hasn't run the automation to reload the integration on HA [previous post: " I now have an automation to check when Occupancy is off then check if mmWave or PIR are still on, if so, it restarts the entire integration in HA"]. I obviously I need more time to double check but so far so good! Nice!

However...Thats just the SEN0609 EP1's. I've got 10 in total and I've found upgrading the 3 x original EP1's (with the SEN0395 sensors and that 3D printed case) has made them go offline with HA 😕. Everything is showing as unavailable for these three. I haven't had time to reflash them yet or do anymore debugging, but just thought you should know.