EverythingSmartHome/everything-presence-one #238: Stop working

An76n, 2024-12-26:
My Everything Presence One stop working few days ago in the middle of the night. I update de software and it comes back to live. But when I unplugged from the computer and reconnect to the original place it becomes offline again. I made this procedure few times with same result. Any idea of what is happening?

Detalles de registro (ADVERTENCIA)

Registrador: aioesphomeapi.reconnect_logic
Fuente: runner.py:189
Ocurrió por primera vez: 25 de diciembre de 2024, 13:07:56 (3 ocurrencias)
Última vez registrado: 25 de diciembre de 2024, 15:03:04

Can't connect to ESPHome API for everything-presence-one-759ec0 @ 192.168.1.141: Error connecting to [AddrInfo(family=<AddressFamily.AF_INET: 2>, type=<SocketKind.SOCK_STREAM: 1>, proto=6, sockaddr=IPv4Sockaddr(address='192.168.1.141', port=6053))]: [Errno 113] Connect call failed ('192.168.1.141', 6053) (SocketAPIError)
Can't connect to ESPHome API for everything-presence-one-759ec0 @ 192.168.1.141: Timeout waiting for HelloResponse, ConnectResponse after 30.0s (TimeoutAPIError)

Detalles de registro (ADVERTENCIA)

Registrador: aioesphomeapi.connection
Fuente: runner.py:189
Ocurrió por primera vez: 25 de diciembre de 2024, 13:10:17 (245 ocurrencias)
Última vez registrado: 25 de diciembre de 2024, 20:14:54

everything-presence-one-759ec0 @ 192.168.1.141: Connection error occurred: everything-presence-one-759ec0 @ 192.168.1.141: Connection lost
192.168.1.141: Connection error occurred: 192.168.1.141: Connection lost
everything-presence-one-759ec0 @ 192.168.1.141: Connection error occurred: [Errno 104] Connection reset by peer

EverythingSmartHome, 2024-12-26:
Hello, it sounds like you are running into an unintentional bug in ESPHome that causes a boot loop when internet is blocked or inaccessible and the device tries to check GitHub for updates

An76n, 2024-12-27:
Thanks for your comment, it helped me find the problem, it did not connect to the internet because I had just installed Adguard. But I don't know why it only affects this device. With the Everything Presence Lite I have had no problem. Any idea how to have Adguard and keep Everything Presence One working? Thanks :)

EverythingSmartHome, 2024-12-27:
Is your Lite on an older firmware? If so that will be why.

You can permit access to everythingsmarthome.github.io only which will permit the device to check for updates only and you won't run into the bug

An76n, 2024-12-27:
The site everythingsmarthome.github.io send an error404

HeWhooIsLate2things, 2025-01-08:
Hi, I'm having the same issue, the thing is that I'm not running any ad blockers and if I try to ping everythingsmarthome.github.io I can. I can configure the sensor only if I update the firmware and it would work that first time, the moment I move it, it will permanently disconnect from HAS. is there anything else I could check? thanks in advance

EverythingSmartHome, 2025-01-08:
Can you post the full startup logs?

HeWhooIsLate2things, 2025-01-08:
Sorry kind of a newbie to the whole thing is this what you need?

[14:44:09]ets Jul 29 2019 12:21:46
[14:44:09]
[14:44:09]rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
[14:44:09]configsip: 0, SPIWP:0xee
[14:44:09]clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
[14:44:09]mode:DIO, clock div:2
[14:44:09]load:0x3fff0030,len:6652
[14:44:09]ho 0 tail 12 room 4
[14:44:09]load:0x40078000,len:15052
[14:44:09]load:0x40080400,len:3836
[14:44:09]entry 0x4008069c
[14:44:09]I (29) boot: ESP-IDF 4.4.8 2nd stage bootloader
[14:44:09]I (29) boot: compile time 10:58:32
[14:44:09]I (29) boot: Multicore bootloader
[14:44:09]I (32) boot: chip revision: v3.1
[14:44:09]I (36) boot.esp32: SPI Speed      : 40MHz
[14:44:09]I (41) boot.esp32: SPI Mode       : DIO
[14:44:09]I (45) boot.esp32: SPI Flash Size : 4MB
[14:44:09]I (50) boot: Enabling RNG early entropy source...
[14:44:09]I (55) boot: Partition Table:
[14:44:09]I (59) boot: ## Label            Usage          Type ST Offset   Length
[14:44:09]I (66) boot:  0 otadata          OTA data         01 00 00009000 00002000
[14:44:09]I (74) boot:  1 phy_init         RF data          01 01 0000b000 00001000
[14:44:09]I (81) boot:  2 app0             OTA app          00 10 00010000 001c0000
[14:44:09]I (89) boot:  3 app1             OTA app          00 11 001d0000 001c0000
[14:44:09]I (96) boot:  4 nvs              WiFi data        01 02 00390000 0006d000
[14:44:09]I (104) boot: End of partition table
[14:44:09]I (108) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=45704h (284420) map
[14:44:09]I (219) esp_image: segment 1: paddr=0005572c vaddr=3ffbdb60 size=04fd0h ( 20432) load
[14:44:09]I (227) esp_image: segment 2: paddr=0005a704 vaddr=40080000 size=05914h ( 22804) load
[14:44:09]I (237) esp_image: segment 3: paddr=00060020 vaddr=400d0020 size=fbad8h (1030872) map
[14:44:09]I (610) esp_image: segment 4: paddr=0015bb00 vaddr=40085914 size=1666ch ( 91756) load
[14:44:09]I (662) boot: Loaded app from partition at offset 0x10000
[14:44:09]I (662) boot: Disabling RNG early entropy source...
[14:44:09]I (673) cpu_start: Multicore app
[14:44:09]I (674) cpu_start: Pro cpu up.
[14:44:09]I (674) cpu_start: Starting app cpu, entry point is 0x400826b0
[14:44:09]I (0) cpu_start: App cpu up.
[14:44:09]I (694) cpu_start: Pro cpu start user code
[14:44:09]I (694) cpu_start: cpu freq: 160000000
[14:44:09]I (694) cpu_start: Application information:
[14:44:09]I (698) cpu_start: Project name:     everything-presence-one
[14:44:09]I (705) cpu_start: App version:      2024.10.0
[14:44:09]I (710) cpu_start: Compile time:     Oct 19 2024 10:57:32
[14:44:09]I (716) cpu_start: ELF file SHA256:  c01ab1952a997d8f...
[14:44:09]I (722) cpu_start: ESP-IDF:          4.4.8
[14:44:09]I (727) cpu_start: Min chip rev:     v0.0
[14:44:09]I (731) cpu_start: Max chip rev:     v3.99 
[14:44:09]I (736) cpu_start: Chip rev:         v3.1
[14:44:09]I (741) heap_init: Initializing. RAM available for dynamic allocation:
[14:44:09]I (748) heap_init: At 3FFAFF10 len 000000F0 (0 KiB): DRAM
[14:44:09]I (754) heap_init: At 3FFB6388 len 00001C78 (7 KiB): DRAM
[14:44:09]I (760) heap_init: At 3FFB9A20 len 00004108 (16 KiB): DRAM
[14:44:09]I (767) heap_init: At 3FFCB710 len 000148F0 (82 KiB): DRAM
[14:44:09]I (773) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
[14:44:10]I (779) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
[14:44:10]I (785) heap_init: At 4009BF80 len 00004080 (16 KiB): IRAM
[14:44:10]I (793) spi_flash: detected chip: generic
[14:44:10]I (796) spi_flash: flash io: dio
[14:44:10]I (802) cpu_start: Starting scheduler on PRO CPU.
[14:44:10]I (0) cpu_start: Starting scheduler on APP CPU.
[14:44:29][E][esp-idf:000]: I (1324) gpio: GPIO[25]| InputEn: 1| OutputEn: 1| OpenDrain: 1| Pullup: 1| Pulldown: 0| Intr:0 

[E][esp-idf:000]: I (1325) gpio: GPIO[26]| InputEn: 1| OutputEn: 1| OpenDrain: 1| Pullup: 1| Pulldown: 0| Intr:0 

[E][esp-idf:000]: I (1361) uart: queue free spaces: 20

[E][esp-idf:000]: I (1362) gpio: GPIO[32]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
[... 489 lines cut ...]
[14:44:50]I (674) cpu_start: Starting app cpu, entry point is 0x400826b0
[14:44:50]I (664) cpu_start: App cpu up.
[14:44:50]I (694) cpu_start: Pro cpu start user code
[14:44:50]I (694) cpu_start: cpu freq: 160000000
[14:44:50]I (694) cpu_start: Application information:
[14:44:50]I (698) cpu_start: Project name:     everything-presence-one
[14:44:50]I (705) cpu_start: App version:      2024.10.0
[14:44:50]I (710) cpu_start: Compile time:     Oct 19 2024 10:57:32
[14:44:50]I (716) cpu_start: ELF file SHA256:  c01ab1952a997d8f...
[14:44:50]I (722) cpu_start: ESP-IDF:          4.4.8
[14:44:50]I (727) cpu_start: Min chip rev:     v0.0
[14:44:50]I (731) cpu_start: Max chip rev:     v3.99 
[14:44:50]I (736) cpu_start: Chip rev:         v3.1
[14:44:50]I (741) heap_init: Initializing. RAM available for dynamic allocation:
[14:44:50]I (748) heap_init: At 3FFAFF10 len 000000F0 (0 KiB): DRAM
[14:44:50]I (754) heap_init: At 3FFB6388 len 00001C78 (7 KiB): DRAM
[14:44:50]I (760) heap_init: At 3FFB9A20 len 00004108 (16 KiB): DRAM
[14:44:50]I (767) heap_init: At 3FFCB710 len 000148F0 (82 KiB): DRAM
[14:44:50]I (773) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
[14:44:50]I (779) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
[14:44:50]I (786) heap_init: At 4009BF80 len 00004080 (16 KiB): IRAM
[14:44:50]I (793) spi_flash: detected chip: generic
[14:44:50]I (796) spi_flash: flash io: dio
[14:44:50]I (802) cpu_start: Starting scheduler on PRO CPU.
[14:44:50]I (0) cpu_start: Starting scheduler on APP CPU.

ModestTG, 2025-01-10:
I believe I'm running into these same issues. I'm getting a boot-loop when I add
```yaml
update:
  - platform: http_request
    id: update_http_request
    name: Everything Presence One Firmware
    source: https://everythingsmarthEverything Presence Oneome.github.io/everything-presence-one/everything-presence-one-manifest.json
    disabled_by_default: true
```
from `everything-presence-one-ble.yaml`. This may be my own fault as I'm not sure if my device supports BLE (it's one of the very first ones shipped). I commented out the above section and everything seems to be working now. My EP1 device is on a VLAN without internet access, but it sounds like I might need to allow access to `everythingsmarthome.github.io` so that the update top level attribute can work properly?

EverythingSmartHome, 2025-01-10:
Yes that's the component that causes boot loop and has the bug in ESPHome.

Btw you can flash the update from the user guide to enable BLE instead of building it manually

wirtulka, 2025-01-10:
I'm experiencing the same issue - no internet restrictions, but the device still gets stuck in a boot loop when powered off and on again (e.g. when moving the sensor to a different location). Is there anything we can do besides waiting for an ESPHome fix?

EverythingSmartHome, 2025-01-10:
You can try flashing the non BLE firmware. If you post the logs I can advise if it looks like the internet bug (many aren't aware they are restricting it in some way)

ModestTG, 2025-01-10:
> Btw you can flash the update from the user guide to enable BLE instead of building it manually

My ESPHome / HA are on a separate VLAN from my EP1 device. I've tried everything to have my EP1 devices be automatically picked up by ESPHome when building from the user guide but nothing has worked. I'm struggling to get the mDNS between the VLANs (I'm running pfSense). The only thing that seems to be working for me is manually creating the YAML and using the `manual-ip` component from `wifi`.

HeWhooIsLate2things, 2025-01-10:
OK, I have an update that might not be of much help since nothing was done to it but here it is: I stopped playing with it for a while and when I checked a few hours later it was connected to HAS again, so to test I unplugged it and re-plugged again and surely it went back to the boot loop, I keep checking on it and after few hours it was connected and working, tried a third time to confirm and the same happened. Right now, is connected and working, again, nothing was done on my side.