EverythingSmartHome/everything-presence-one #239: EP1 becomes unavailable

mturnwall, 2024-12-27:
I recently received 2 EP1 devices. I've been having some trouble with them though. They will randomly become unavailable for hours. The same thing happens if I unplug it to move to a different location as I try to get the placement right. It will remain unavailable after I plug it back in and will remain like that till it randomly will just come back. For example, I just moved the EP1 in my daughter's room and it's been unavailable for 20 minutes so far after I plugged it back in.

I know the device has power because the leds come on. At first I was on firmware 1.2.0 but recently upgraded to 1.2.6.

I've attached the downloaded diagnostics. Let me know if you need anymore information. Thank you.

[config_entry-esphome-01JEZ7EP0MC6B3PR0QVQN1F9HT.json](https://github.com/user-attachments/files/18263153/config_entry-esphome-01JEZ7EP0MC6B3PR0QVQN1F9HT.json)

[image]

mturnwall, 2024-12-27:
Here's an example of what happens when I unplug the EP1. It's been unavailable for 24 minutes since I plugged it back in 
[image]

EverythingSmartHome, 2024-12-27:
Do you have the device blocked from internet by any chance?

mturnwall, 2024-12-27:
Not that I know. I checked my firewall and I can see the device is connected to the network

[image]

And I as I was checking I see the device just became available.

[image]

jvrijn, 2025-03-05:
Hi, same problem here. Device becomes unavailable for a second or two and comes back. It has access to the Internet. I did notice on one occasion that the red status LED was flashing (I normally have it set to off) before it became unavailable.

EverythingSmartHome, 2025-03-05:
Sounds more like a Bluetooth proxy issue. Try flashing the non Bluetooth firmware

jvrijn, 2025-03-05:
I have the non-Bluetooth Proxy version of the firmware flashed.  I got the device a couple of weeks ago and installed it last week, following the instructions.

jvrijn, 2025-03-19:
Just wanted to give you and the community (for future reference) an update. My problem with the EP1 going unavailable has been resolved. I re-flashed the unit with firmware 1.2.7 and changed out the USB charger and the device is working flawlessly now.

nh2, 2025-08-21:
I have this problem, too. Yesterday it was unavailable for many hours:

[image]

`Firmware: 1.2.11 (ESPHome 2025.7.4)`