EverythingSmartHome/everything-presence-one #242: EP1's dead after update to ESP Home 2025.2

hesspoint, 2025-02-20:
Updated to ESP Home 2025.2 via the HA interface. 

After that all my EP1s are non responsive and the LED glows red.

I tried to manually flash the firmware via the ESPHome Builder in HA which had helped in the past.

It did flash it all right but still no response

`WARNING Can't connect to ESPHome API for everything-presence-one-d4d3e0.local: Timeout while connecting to [AddrInfo(family=<AddressFamily.AF_INET: 2>, type=<SocketKind.SOCK_STREAM: 1>, proto=6, sockaddr=IPv4Sockaddr(address='192.168.178.185', port=6053))] (TimeoutAPIError)
INFO Trying to connect to everything-presence-one-d4d3e0.local in the background`

hesspoint, 2025-02-21:
Ok I reinstalled the firmware. First I get the error message as you can see above. After waiting for a little bit it starts connecting but keeps showing warnings about BLE

[13:56:32][W][component:237]: Component esp32_ble took a long time for an operation (51 ms).
[13:56:32][W][component:238]: Components should block for at most 30 ms.
[13:56:32][V][esp32_ble:348]: (BLE) gap_event_handler - 3
[13:56:32][V][esp32_ble_tracker:411]: gap_scan_result - event 0
[13:56:32][V][esp32_ble:348]: (BLE) gap_event_handler - 3
[13:56:32][V][esp32_ble_tracker:411]: gap_scan_result - event 0
[13:56:32][V][esp32_ble:348]: (BLE) gap_event_handler - 3
[13:56:32][V][esp32_ble_tracker:411]: gap_scan_result - event 0
[13:56:32][V][esp32_ble:348]: (BLE) gap_event_handler - 3
[13:56:32][W][component:237]: Component esp32_ble took a long time for an operation (167 ms).
[13:56:32][W][component:238]: Components should block for at most 30 ms.
[13:56:32][W][esp32_ble_tracker:125]: Too many BLE events to process. Some devices may not show up.

It is connected but later randomly disconnects again

falzm, 2025-02-24:
I'm experiencing the same behavior after upgrading the firmware version from 1.2.6 to 1.2.7 OTA via Home Assistant.

EverythingSmartHome, 2025-02-24:
Do you have it added to the ESPHome dashboard?

isewise, 2025-02-24:
My EP1 was added to the ESPHome builder dashboard.

EverythingSmartHome, 2025-02-24:
This is not an issue in this case but a a misunderstanding of the update system.

I'm assuming you updated via the Everything Presence One firmware entity? If so, this entity is disabled on purpose to avoid this.

When you update via this entity, it pulls the update directly from GitHub. However if you have it already added to the ESPHome addon, it means you have an encryption key in your config, which the update from GitHub overwrites. So HA is trying to communicate with the device using a password, meanwhile the device no longer has a password set. That's the reason we don't really recommend the ESPHome addon unless you need to build a custom firmware

isewise, 2025-02-24:
I only updated ESPhome to 2025.2. I did not update via the update entity.  

I have since tried to start from scratch via the [documentation website](https://everythingsmarthome.github.io/everything-presence-one/updating.html). The EP1 works initially then become unavailable.

EverythingSmartHome, 2025-02-24:
It's possible, there were some BLE changes to 2025.2 so certainly could be

hesspoint, 2025-02-26:
@EverythingSmartHome btw I cannot install the co2-rev1.3 FW anymore. Neither the BLE nor the non-BLE version.

When I try to flash I get this error:

INFO ESPHome 2025.2.1
INFO Reading configuration /config/esphome/everything-presence-one-de9824.yaml...
INFO Updating https://github.com/esphome/esphome.git@pull/7547/head
ERROR Unexpected exception while reading configuration:
Traceback (most recent call last):
  File "/usr/local/bin/esphome", line 8, in <module>
    sys.exit(main())
             ^^^^^^
  File "/esphome/esphome/__main__.py", line 1049, in main
    return run_esphome(sys.argv)
           ^^^^^^^^^^^^^^^^^^^^^
  File "/esphome/esphome/__main__.py", line 1027, in run_esphome
    config = read_config(dict(args.substitution) if args.substitution else {})
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/esphome/esphome/config.py", line 1095, in read_config
    res = load_config(command_line_substitutions)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/esphome/esphome/config.py", line 949, in load_config
    return _load_config(command_line_substitutions)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/esphome/esphome/config.py", line 939, in _load_config
    return validate_config(config, command_line_substitutions)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/esphome/esphome/config.py", line 835, in validate_config
    target_platform = core_config.preload_core_config(config, result)
                      ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/esphome/esphome/core/config.py", line 222, in preload_core_config
    if _is_target_platform(domain):
       ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/esphome/esphome/core/config.py", line 184, in _is_target_platform
    if get_component(name, True).is_target_platform:
       ^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/esphome/esphome/loader.py", line 205, in get_component
    return _lookup_module(domain, exception)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/esphome/esphome/loader.py", line 181, in _lookup_module
    module = importlib.import_module(f"esphome.components.{domain}")
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/importlib/__init__.py", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "<frozen importlib._bootstrap>", line 1206, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1178, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1142, in _find_and_load_unlocked
ModuleNotFoundError: No module named 'esphome.components.bsubstitutions'

isewise, 2025-02-26:
Using the [latest version of ESPHome](https://esphome.io/changelog/2025.2.0.html#release-2025-2-1-february-26) has not fixed the issue for me.

hesspoint, 2025-02-26:
> Using the [latest version of ESPHome](https://esphome.io/changelog/2025.2.0.html#release-2025-2-1-february-26) has not fixed the issue for me.

Yeah I also had my hopes and installed the BLE firmware one one of my ep1s but same issues as before. Reverted back to non BLE to at at least be able to use my sensor. 
Super frustrating especially since I need the BT Proxy  to keep my plant sensors online.

hesspoint, 2025-03-02:
Any update on the non-working co2-rev1.3 FW? A fix would be highly appreciated. Thanks @EverythingSmartHome

isewise, 2025-03-03:
> Using the [latest version of ESPHome](https://esphome.io/changelog/2025.2.0.html#release-2025-2-1-february-26) has not fixed the issue for me.

This time using the latest, [as of today](https://esphome.io/changelog/2025.2.0.html#release-2025-2-2-march-3), and still not working.

EverythingSmartHome, 2025-03-03:
> Any update on the non-working co2-rev1.3 FW? A fix would be highly appreciated. Thanks [@EverythingSmartHome](https://github.com/EverythingSmartHome)

Do you mean the BLE firmware? If so, this would need to be fixed in ESPHome if it was a problem that was introduced from the previous update. We haven't changed any config on the EP1 side specifically and so not something I can resolve (at least from what I can tell)